OpenKinect/libfreenect2 #533: Kinectv2/viewer.cpp:47: error: undefined reference to `flextInit'

whatisor, 2016-01-19:
Hello,
I got this message when setup a new project using built libfreenect2.
Do you have any idea? I add -lglfw -lGL -lfreenect2
Thanks

whatisor, 2016-01-19:
Issue fixed with adding flex source.
I think it is not good, please add this into libfreenect2 because you place it in src folder.
Thank you

xlz, 2016-01-19:
`viewer.cpp` is not supposed to be included for a new project. How are you using it?

If you want to create a new project using the example, `examples/CMakeLists.txt` says "The example build system [...] will work out-of-tree with these files copied". The viewer is a standalone example, not part of libfreenect2.

whatisor, 2016-01-20:
However, flextGL.cpp and flextGL.h is inside src folder, it will make developer confusing. IT should be in example folder, right ? :+1:

xlz, 2016-01-20:
flextGL.cpp is mainly used by libfreenect2 itself. Using it in the example is a by product. You should copy it if you want to use it in your code.

whatisor, 2016-01-20:
OK, I see.
Just in my opinion, it is abnormal if an example needs to be compiled with a file of library source.
But I have no idea to improve for now  :-1:

xlz, 2016-01-24:
The example uses that file and references it without duplication. There is nothing wrong about that. Follow the CMake comment if you need it.

closed 2016-01-24 by xlz