OpenKinect/libfreenect2 #534: Why do we need Sudo permission while using libfreenect2?

whatisor, 2016-01-20:
Hi,
Run ./bin/Protonect requires sudo.
Using libfreenect2 for my project also requires sudo.
It is abnormal. Is it possible to make it same with libfreenect (kinect 1)?
Thank you

xlz, 2016-01-20:
You are never asked to use sudo. It is likely you did not follow the instructions https://github.com/OpenKinect/libfreenect2#protonect-complains-about-no-device-connected-or-failure-opening-device

whatisor, 2016-01-20:
Perfect!
Thank you.
Can you add it into install script?

de-vri-es, 2016-01-20:
Installing cross-platform udev rules might be somewhat tricky. For example, on some systems TAG+="uaccess" could be preferred over MODE="0666". I'm also not certain if udev rules go to the same directory on all platforms and distros. If the install script would installs udev rules, it would be nice to be able atleast add an option to disable it so that packagers can use the appropriate rules for their platform.

xlz, 2016-01-21:
The provided udev rules are clearly not intended for all distros. Debian usually has `GROUP="plugdev" MODE="0664"`. Distro packagers should add their own udev rules as they see fit.

closed 2016-01-21 by xlz