OpenKinect/libfreenect2 #535: temperature changes depth measurements

ahundt, 2016-01-22:
It seems depth measurements change over time with the temperature, according to [FIRST EXPERIENCES WITH KINECT V2 SENSOR FOR CLOSE RANGE 3D MODELLING](http://www.int-arch-photogramm-remote-sens-spatial-inf-sci.net/XL-5-W4/93/2015/isprsarchives-XL-5-W4-93-2015.pdf). This seems consistent with what I've heard from others' anecdotal experiences as well.

At a minimum do we have a way of getting any temperature measurements if there is one built into the sensor from libfreenect2? 

Beyond that, is there any value in dealing with the problem in libfreenect2 or is that totally out of scope?

This issue may have some tie in to calibration https://github.com/OpenKinect/libfreenect2/issues/524

xlz, 2016-01-22:
The paper you linked does not say the warm-up is associated with temperature.

ahundt, 2016-01-23:
ah sorry that was also based on conversations with others... It is definitely just an assumption, are there other possibilities?

xlz, 2016-01-23:
I'm not aware of any way to detect this change. I think the report you linked or another report simply make the recommendations that for accurate measurement, wait 10 minutes for warm-up.

bellekci, 2016-01-24:
Well, I have read the paper you have pointed like 6 months ago. As far as I remember, it proposes opposite of what you are saying. It shares that you should wait 20 mins until warm-up is complete. In other words, depth values change during the warm-up time and afterwards they virtually do not change.

Temperature does affect the measurement values in ToF cams; nevertheless, there is a huge heat sink and a fan inside the kinect which keeps the temperature at desired value.

aurelw, 2016-01-29:
Just to confirm your problem. I can also observe depth accuracy changes over time and depending on the device temperature.

xlz, 2016-01-29:
So you tried cooling down the device temperature and observed a decrease of accuracy?

aurelw, 2016-02-01:
I have a written report on this topic here, which I am not able to fully disclose. However, it tells me, that the mean distance to a fixed surface varies with temperature (plotted curve). The actual accuracy wasn't mesured and it is hard to tell to which temperature the kinect is calibrated to. But I assume it measures most accurately at room temperature after a 20 minutes warm phase.

xlz, 2016-02-02:
As a resolution, this may be included in some kind of Kinect best practice document.

I did see strings related to temperature along with voltage and power management things. But unless a reverse engineering guru helps us, there is indeed no way to extract that information.