OpenKinect/libfreenect2 #536: About converting depth->color space.

kidapu, 2016-01-22:
I want to convert depth-> color space.
So I tried to test with following codes.

```
#include "ofProtonect2.h"
#include "registration.h"

class RegistrationWrapper
{

public:
    libfreenect2::Freenect2Device::IrCameraParams depthParam;
    libfreenect2::Freenect2Device::ColorCameraParams colorParam
    libfreenect2::Registration * registration;
    
    void setup(ofProtonect2* protonect)
    {
        depthParam = protonect->getIrCameraParams();
        colorParam = protonect->getColorCameraParams();
        registration = new libfreenect2::Registration(&depthParam, &colorParam);
    }
    
    void depthToColor(int dx, int dy, int dz, float &cx, float &cy)
    {
        registration->apply(dx, dy, dz, cx, cy);
    }
};
```
And the result is following movie.
https://www.youtube.com/watch?v=s7UPJgLO1oM&feature=youtu.be

I think this is not correct, and the result has noise.
Do anyone know this is why? 

My enviroment is OSX 10.11.2, Mid 2015.

xlz, 2016-01-22:
`int dz`

`dz` should be a `float`. Are you sure with using an `int`?

Your code provided here is a thin wrapper over the API - not much can be inferred. Can you show how you are using the returned `cx cy`?

kidapu, 2016-01-22:
Thanks xlz!

> dz should be a float. Are you sure with using an int?

I change `dz` from int to float. The convert result becomes continuously, but I think the position is still wrong. 

[image]

> Can you show how you are using the returned cx cy?

I uploaded my codes on my repository.

Drawing depth recangle on [this codes](https://github.com/kidapu/libfreenect2-test/blob/master/libfreenectTest/src/ofApp.h#L160-L171), and Drawing color rectangle on [this codes](https://github.com/kidapu/libfreenect2-test/blob/master/libfreenectTest/src/ofApp.h#L192-L226).

>  It looks like the returned cx cy are "distorted color coordinates (pixel)"

Do you know the way to "undistorted color coordinates" ?

xlz, 2016-01-22:
I think you are getting 0 distance by using `_kinect.getDistanceAt()`. Black pixels in depth are zero distance I think? You should determine the 3-D centroid and use that single depth value for all four vertices of the rectangle.

kidapu, 2016-01-23:
> Black pixels in depth are zero distance I think?

I tried to make sure, but black pixels in depth is not zero.

> You should determine the 3-D centroid and use that single depth value for all four vertices of the rectangle.

I fixed my code accoring to your advice, and add little exception handling.
https://github.com/kidapu/libfreenect2-test/commit/80d8b9c441ab29c4d19388b0b5b8a3c3bd3d15d5

The result is following movie. I think it's getting better but still wrong... hmm...
https://www.youtube.com/watch?v=CpQlmod3-dI&feature=youtu.be

xlz, 2016-01-23:
They may be not zero in your processed image, but I think in `apply()` they are zero. Try to use the depth value from the unprocessed depth image.

kidapu, 2016-01-25:
hmm...

I output log of `depth value` before `apply()` by this commit.
https://github.com/kidapu/libfreenect2-test/commit/b4a7d04a3423f0aa5f661b756df1f520d379954c

And following is the part of logs.
```
 - - - - 
dx:162/ dy:102/ dz:151.667
dx:209/ dy:102/ dz:151.667
dx:209/ dy:151/ dz:151.667
dx:162/ dy:151/ dz:151.667
 - - - - 
dx:149/ dy:100/ dz:92.7086
dx:201/ dy:100/ dz:92.7086
dx:201/ dy:151/ dz:92.7086
dx:149/ dy:151/ dz:92.7086
 - - - - 
dx:149/ dy:100/ dz:92.7086
dx:201/ dy:100/ dz:92.7086
dx:201/ dy:151/ dz:92.7086
dx:149/ dy:151/ dz:92.7086
 - - - - 
dx:137/ dy:99/ dz:90.039
dx:194/ dy:99/ dz:90.039
dx:194/ dy:151/ dz:90.039
dx:137/ dy:151/ dz:90.039
 - - - - 
dx:137/ dy:99/ dz:90.039
dx:194/ dy:99/ dz:90.039
dx:194/ dy:151/ dz:90.039
dx:137/ dy:151/ dz:90.039
 - - - - 
dx:127/ dy:100/ dz:89.457
dx:189/ dy:100/ dz:89.457
dx:189/ dy:152/ dz:89.457
dx:127/ dy:152/ dz:89.457
 - - - - 
dx:122/ dy:150/ dz:354.792
dx:175/ dy:150/ dz:354.792
dx:175/ dy:202/ dz:354.792
dx:122/ dy:202/ dz:354.792
```

So I think [this code](https://github.com/kidapu/libfreenect2-test/blob/master/libfreenectTest/src/ofApp.h#L207-L210) get depth value of  unprocessed depth image.

xlz, 2016-01-25:
> dx:137/ dy:99/ dz:90.039

So the point is 9cm in front of the Kinect. Is this a reasonable measurement?

kidapu, 2016-01-26:
Thank you for your great advice! Finally I was success!
https://www.youtube.com/watch?v=TdeG0hgKzYA&feature=youtu.be

The cause is the unit of depth was `cm` rather than `mm`. I fixed the bug with following commit.
https://github.com/kidapu/libfreenect2-test/commit/93c1a5be2bdf93bcf51b58e18e3b200bba39761e

I will close this issue.

closed 2016-01-26 by kidapu

xlz, 2016-01-26:
ofxMultiKinectV2's `getDistanceAt()` convert mm to cm. NOTOURBUG.

edku2, 2017-03-28:
Hey I'm having troubles getting the ofxMultiKinectV2 example to compile and I think it's because of the libfreenect2 library. 

the two errors coming up says that" non-void function 'getirCameraCarams' should return a value" and non-void function 'getirColorCameraParams' should return a value "

 libfreenect2::Freenect2Device::ColorCameraParams getColorCameraParams() {
        if (!dev){return;} //error
        return dev->getColorCameraParams();
    }
    libfreenect2::Freenect2Device::IrCameraParams getIrCameraParams() {
        if (!dev){return;} // error
        return dev->getIrCameraParams();
    }

I would love if you can help.

Eddie