OpenKinect/libfreenect2 #537: CONTRIB file isn't updated for v0.1.0 snapshot release

JoshBlake, 2016-01-22:
Should use a script like https://github.com/OpenKinect/libfreenect/blob/master/mkcontrib.sh to generate e.g.:
```
Albert Hofkamp <a.t.hofkamp@tue.nl>
Alistair <alistair.wk@gmail.com>
Christian Kerl <christian-kerl@web.de>
Christian Kerl <christian.kerl@in.tum.de>
Dave Coleman <davetcoleman@gmail.com>
Dorian Galvez-Lopez <dorian@email.gwu.edu>
Federico Spinelli <fspinelli@gmail.com>
Florian Echtler <floe@butterbrot.org>
Francisco Facioni <fran6co@gmail.com>
Gabor Papp <gabor.papp@gmail.com>
Giacomo Dabisias <g.dabisias@sssup.it>
Henning Jungkurth <henning.jungkurth@gmail.com>
HenningJ <henning.jungkurth@gmail.com>
James Billingham <james@jamesbillingham.com>
Joshua Blake <joshblake@gmail.com>
Lars Glud <larshg@gmail.com>
Lingzhu Xiang <lingzhu@lingzhuxiang.com>
Lingzhu Xiang <xlz@users.noreply.github.com>
Ludique <contacto@ludique.cl>
Mario Wündsch <dev@wuendsch.net>
Matthias Goldhoorn <Matthias@Goldhoorn.eu>
Matthias Goldhoorn <matthias.goldhoorn@dfki.de>
Matthias Goldhoorn <matthias.goldhoorn@uni-bremen.de>
MrTatsch <tatsch@gmail.com>
P.E. Viau <viau.pe@gmail.com>
Paul Reynolds <paul@blastoffgames.net>
Rich Wareham <rjw57@cam.ac.uk>
Rich Wareham <rjw57@cantab.net>
Ryan Gordon <ryan@quizlet.com>
Sergey Gusarov <laborer2008@gmail.com>
Steffen Fuchs <st.fuchs.tr@gmail.com>
Thiemo Wiedemeyer <wiedemeyer@cs.uni-bremen.de>
Thiemo Wiedemeyer <wiedemeyer@informatik.uni-bremen.de>
Zou Hanya <hanyazou@gmail.com>
augmenta <augmenta@augmenta.home>
christiankerl <christian.kerl@in.tum.de>
hanyazou <hanyazou@gmail.com>
larshg <larshg@gmail.com>
rahulraw <rahulrawat.can@gmail.com>
vinouz <vinouz@yahoo.fr>
yuanmingze <3054502461@sina.com>
```
Should obvious duplicates be manually cleaned up?

floe, 2016-01-22:
Good point, sorry I missed that :-/ I think a manual cleanup is best, first issue for 0.1.1 I guess ;-)

JoshBlake, 2016-01-22:
Since this is an administrative detail, not a git commit, perhaps it would be acceptable to re-upload the archive artifacts with the fixed CONTRIB under 0.1.0?

But adding the script and maybe a release checklist would be good for 0.1.1. :)

xlz, 2016-01-22:
The CONTRIB is just for the Windows binary release, right? The source tarballs are automatically generated from the git repo and it's not practical to put the generated CONTRIB file in the git repo.

JoshBlake, 2016-01-23:
The way we have the source header and licenses setup, contributor information is necessary to complete  the copyright statement. Within the context of a git repo, the contributor information is available in the commit metadata and can be easily accessed with a script or git commands. For snapshot releases, it should really be included in all non-git downloadable artifacts, source included.

I'd recommend updating the CONTRIB file with a commit just prior to each release, and perhaps update the mkcontrib.sh script to include the git commit that generated the contributor list. Between releases the file will be stale, but that's fine.

xlz, 2016-01-23:
OK, so the release procedures are: (1) update CONTRIB; (2) change the version number in CMakeLists.txt; (3) `git tag`; (4) GitHub "draft a new release".

de-vri-es, 2016-01-23:
Keep in mind that if a release tarball is changed, packagers may need to update checksums as well. That's not necessarily a problem, but a heads up would be nice (a comment here maybe if/when it'd one?)

xlz, 2016-01-23:
We dont' change released tarballs.

floe, 2016-01-23:
Even if we wanted to, we couldn't. They are immutable on Github, but as mentioned, I'll make a 0.1.1 anyway (and I can just put the generated CONTRIB file in the v0.1 branch, not master).

xlz, 2016-01-23:
This mkcontrib.py removes less common duplicate names and emails.
```python
from subprocess import Popen, PIPE
from collections import defaultdict, Counter

p = Popen(["git","log","--no-merges","--format=%aN <%aE>"], stdout=PIPE).stdout
common_email = defaultdict(Counter)
common_name = defaultdict(Counter)
for line in p:
    author, email = line.rstrip().split('<')
    common_email[author].update([email])
    common_name[email].update([author])

for email in common_name:
    names = common_name[email]
    names = sorted(names, key=lambda x: (names[x], len(x)), reverse=True)
    for name in names[1:]:
        common_email[names[0]] += common_email[name]
        del common_email[name]
        print '# Less common or shorter name', name, 'is replaced by', names[0]

for name in sorted(common_email):
    for k in common_email[name].most_common()[1:]:
        print '# Less common email <' + k[0] + ' is removed.'
    print name + '<' + common_email[name].most_common(1)[0][0]
```

```
# Less common or shorter name HenningJ  is replaced by Henning Jungkurth 
# Less common or shorter name hanyazou  is replaced by Zou Hanya 
# Less common or shorter name larshg  is replaced by Lars Glud 
# Less common or shorter name christiankerl  is replaced by Christian Kerl 
Albert Hofkamp <a.t.hofkamp@tue.nl>
Alistair <alistair.wk@gmail.com>
# Less common email <christian-kerl@web.de> is removed.
Christian Kerl <christian.kerl@in.tum.de>
Dave Coleman <davetcoleman@gmail.com>
Dorian Galvez-Lopez <dorian@email.gwu.edu>
Federico Spinelli <fspinelli@gmail.com>
Florian Echtler <floe@butterbrot.org>
Francisco Facioni <fran6co@gmail.com>
Gabor Papp <gabor.papp@gmail.com>
Giacomo Dabisias <g.dabisias@sssup.it>
Henning Jungkurth <henning.jungkurth@gmail.com>
James Billingham <james@jamesbillingham.com>
Joshua Blake <joshblake@gmail.com>
Lars Glud <larshg@gmail.com>
# Less common email <xlz@users.noreply.github.com> is removed.
Lingzhu Xiang <lingzhu@lingzhuxiang.com>
Ludique <contacto@ludique.cl>
Mario Wündsch <dev@wuendsch.net>
# Less common email <matthias.goldhoorn@uni-bremen.de> is removed.
# Less common email <Matthias@Goldhoorn.eu> is removed.
Matthias Goldhoorn <matthias.goldhoorn@dfki.de>
MrTatsch <tatsch@gmail.com>
P.E. Viau <viau.pe@gmail.com>
Paul Reynolds <paul@blastoffgames.net>
# Less common email <rjw57@cantab.net> is removed.
Rich Wareham <rjw57@cam.ac.uk>
Ryan Gordon <ryan@quizlet.com>
Sergey Gusarov <laborer2008@gmail.com>
Steffen Fuchs <st.fuchs.tr@gmail.com>
# Less common email <wiedemeyer@informatik.uni-bremen.de> is removed.
Thiemo Wiedemeyer <wiedemeyer@cs.uni-bremen.de>
Zou Hanya <hanyazou@gmail.com>
augmenta <augmenta@augmenta.home>
rahulraw <rahulrawat.can@gmail.com>
vinouz <vinouz@yahoo.fr>
yuanmingze <3054502461@sina.com>
```

floe, 2016-01-25:
Great, thanks, can you put that in `tools/` (to preserve authorship)?

floe, 2016-01-25:
@xlz was there some minor documentation update left for v0.1.1 regarding wait after `Freenect2Device::close()`?

xlz, 2016-01-25:
Added to master in fa464d288a6d43ad810264b1f0a71fc85eaa50f4, you can cherry-pick it to v0.1 branch.

closed 2016-01-26 by xlz