OpenKinect/libfreenect2 #538: Failed to load nodelet [/kinect2_bridge] of type [kinect2_bridge/kinect2_bridge_nodelet]

17Skye17, 2016-01-22:
I tried: 
 roslaunch kinect2_bridge kinect2_bridge.launch

but I came aross an error：
   [ERROR] [1453460840.163354866]: Failed to load nodelet [/kinect2_bridge] of type       [kinect2_bridge/kinect2_bridge_nodelet]: Could not find library corresponding to plugin kinect2_bridge/kinect2_bridge_nodelet. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[FATAL] [1453460840.163649816]: Service call failed!

I followed the steps on github,but still failed to open kinect camera .Does anyone know why this happens？
Thanks in advance

floe, 2016-01-22:
This looks like a ROS issue. Does the Kinect work with the `Protonect` tool that's bundled with libfreenect2? If yes, then please continue discussion at https://github.com/code-iai/iai_kinect2.

17Skye17, 2016-01-22:
OK,thanks a lot

closed 2016-01-22 by floe