OpenKinect/libfreenect2 #539: The device disappear after calling close() for a few seconds.

hanyazou, 2016-01-24:
While investigating #517, I've noticed that it takes a few seconds to restart the Kinect device after calling close(). And the USB deice disappear for a second. This means that you must wait for a while before you call enumerateDevices() and openDevice() after calling close(). This issue might depend Mac OS X.

You can reproduce this issue with stock Protonect.

case 1 (hungup)
-------
> $ ./bin/Protonect; sleep 0; ./bin/Protonect

1. Stop with ESC key after the first Protonect shows the image.
2. Second Protonect does not show any images.

> [Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 005697142447
device firmware: 4.0.3912.0
[Debug] [DepthPacketStreamParser] not all subsequences received 1022
// no messages. it seems that no data come from the device.

In this case Ctrl-C and ESC can not stop the second Protonect. It requires kill -KILL.

case 2 (error)
-------
> $ ./bin/Protonect; sleep 1; ./bin/Protonect

1. Stop with ESC key after the first Protonect shows the image.
2. Second Protonect can't detect the device.

>[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 12 usb devices connected
[Info] [Freenect2Impl] found 0 devices
no device connected!

case 3 (this works fine)
-------
> $ ./bin/Protonect; sleep 3; ./bin/Protonect

1. Stop with ESC key after the first Protonect shows the image.
2. Second Protonect shows the image also.

xlz, 2016-01-24:
> The device disappear after calling close() for a few seconds.

What is wrong with this?

hanyazou, 2016-01-24:
@xlz Please refer case 1 and case 2. It fails randomly depending on the timing. This type of behavior debase stability of system which uses this library.

For example, this test script fails randomly because of this issue.

>#!/bin/bash
test()
{
  $* &
  pid=$!
  sleep 5
  kill $pid || exit
  sleep 1  # longer wait required
}
test UserViewer
test Protonect
test NiViewer

xlz, 2016-01-24:
The Kinect reboots and it takes 3 seconds. What can you do about it?

Case 1 fails because you try to use the Kinect when it is shutting down. Case 2 fails because the Kinect is still booting.

How to not shut down the Kinect when the program exits?

hanyazou, 2016-01-24:
Thank you for the information. I understand that two fixes are required.

1. wait until the shutdown sequence starts and the device disappears in close().
  If only close() knows the device will reboot, Close() should take care of this.

2a. wait until the device comes back in close().
  same reason as 1. but this looks too much.

2b. wait (retry) in enumeration if enumeration detects adaptor only like 71519cf
  While the reboot sequence takes 3 seconds, it takes only 1~1.5 seconds until the USB device entry comes back. So this takes less than 2 seconds in worst case.

hanyazou, 2016-01-24:
Or libfreenect2 records in some persistent storage that the device is in reboot sequence.

xlz, 2016-01-24:
I don't want to force artificial wait in libfreenect2. Maybe the user will stop using Kinect altogether. The user can wait externally if they want to restart the Kinect.

> sleep 1 # longer wait required

As is said, what's wrong with `sleep 3`?

The question is still here: How to not shut down the Kinect when the program exits? You proposed several ways to wait for the shut down, not what I'm looking for.

xlz, 2016-01-24:
I think the thing you are actually proposing is USB hotplug.

hanyazou, 2016-01-24:
> The user can wait externally if they want to restart the Kinect.

In this way, libfreenect2 library or manual should provide information how long the user must wait before enumeration and open. We can describe 'wait 3 seconds before reuse' in document. But shorter wait is better. To make the wait time shorter, we need more information from the device at runtime.

> The question is still here: How to not shut down the Kinect when the program exits? 

The stop() and close() kick shutdown sequence. There is no problem if the program exits suddenly with exit() or uncaught signal or exception.

This script runs without random errors. Because the device does not shutdown, Protonect shows the image window quickly.

> #!/bin/bash
test()
{
$* &
pid=$!
sleep 5
kill <b>-KILL</b> $pid || exit
}
test Protonect
test Protonect
test Protonect

xlz, 2016-01-27:
There may be an easy fix. Can you try commenting out this line in `src/libfreenect.cpp`?
```cpp
  command_tx_.execute(ShutdownCommand(nextCommandSeq()), result);
```

xlz, 2016-01-29:
I have good results with the above fix on Linux and Windows. How does the above fix work on Mac?

hanyazou, 2016-01-30:
@xlz I'm very sorry to find that your fix does not work on my Mac.

And I found that
> There is no problem if the program exits suddenly with exit() or uncaught signal or exception.
was not true.
If the Protonect call exit(1) in the sigint_hanlder() the device may reboot with probability 1/10 or so. I can't address the condition and timing in which the device reboot for now.

xlz, 2016-01-30:
ok, then what is the deterministically reproducible method to not shut down Kinect on Mac?

hanyazou, 2016-01-31:
I modified test script, added exit() in Protonect and run "./bin/Protonect cl -noviewer" 500 times to count how many times the Kinect 2 device will reboot when the protonect exit.

A. 69 / 500   # kill -KILL, protonect terminate immediately
B. 85 / 500   # kill -INT, exit() in sigint_handler()
C. 14 / 500   # kill -INT, exit() before stop()
D. 465 / 500   # kill -INT, exit() before close()
E. 472 / 500   # kill -INT, close() w/o shutdown command
F. 500 / 500   # kill -INT, close() with shutdown command

C. is the best timing to prevent shutdown the device. But C is not perfect. I thought D. was safe to exit program because the USB request submission is terminated and all pending transmission is canceled. But the result is opposite, the device 95% reboots when the protonect exit.

According to these result, I think there is no method to not shut down Kinect on MY Mac and MY Kinect.

>#!/bin/bash
reboot=0
test()
{
    $* > /dev/null 2>&1 &
    pid=$!
    wait=$((RANDOM % 3 + 3)).$((RANDOM % 100))
    echo sleep $wait
    sleep $wait
    #kill -KILL $pid
    kill -INT $pid
    sleep 1
    if system_profiler SPUSBDataType | grep Xbox; then
        :
    else
        echo "#### NO DEVICE"
        sleep 3 # wait for the device back
        reboot=`expr $reboot + 1`
    fi
}
i=0
while [ $i -lt 500 ]; do
    test ./bin/Protonect cl -noviewer
    i=`expr $i + 1`
done
echo $reboot / $i

xlz, 2016-01-31:
OK, so the results tell us some lines of code in `close()` would make the Kinect reboot.

There are not many lines of functioning code in `close()`. Can you try removing them one by one? They are `SetModeEnabledWith0x00640064Command`, `SetModeDisabledCommand`, `ShutdownCommand`, `releaseInterfaces`, `libusb_close`.

hanyazou, 2016-01-31:
> D. 465 / 500 # kill -INT, exit() before close()

This is almost equivalent to that I remove all lines in close(). If I remove all lines, it still reboot 90%.

> E. 472 / 500 # kill -INT, close() w/o shutdown command

This is exactly equivalent to that I remove ShurdownCommand in close(). Still reboot 95%.

> F. 500 / 500 # kill -INT, close() with shutdown command

I'm sure the ShutdownCommand will reboot the device 99.9%.

> Can you try removing them one by one? They are SetModeEnabledWith0x00640064Command, SetModeDisabledCommand, ShutdownCommand, releaseInterfaces, libusb_close.

Though these lines may affect the reboot probability, each one contributes only a few percent I think. Do you think that removing them one by one trial is needed?

xlz, 2016-01-31:
A few percent? It doesn't work like that. Whatever, I'm just saying you can identify which lines of code actually shuts down the Kinect.

xlz, 2016-01-31:
OK, actually according to this

> C. 14 / 500 # kill -INT, exit() before stop()

You should look for the problematic lines of code in `stop()`.

xlz, 2016-01-31:
Try to comment out various `usb_control_.xxx()` and `command_tx_.yyy()`, and see which are the offending ones.

hanyazou, 2016-01-31:
OK I will try. But I feel that timing of exit() affects the probability. I think LOG_INFO << "..." lines should be removed and I must try same binary many times. It seems that 100 times is not enough.

xlz, 2016-01-31:
I don't think calling exit() in a signal handler is well defined. Try adding exit() under certain line in stop().

hanyazou, 2016-02-14:
I identified which lines cause to shutdown the device in stop() and close().
If we comment out xxx_pool.cancel(), StopCommand() and ShutdownCommand() we can avoid shutdown and reboot. (but libusb_close() cause segmentation violation randomly.)

please refer attached log [test.sh-20160206-131909.log.txt](https://github.com/OpenKinect/libfreenect2/files/129619/test.sh-20160206-131909.log.txt)
It rebooted only 4/1000 times.

> Protonect
  while(!protonect_shutdown) {
    listener.waitForNewFrame(frames);
    registration->apply(rgb, depth, &undistorted, &registered);
  }
  // stop()
  rgb_transfer_pool_.disableSubmission();
  ir_transfer_pool_.disableSubmission();
  // rgb_transfer_pool_.cancel();
  // ir_transfer_pool_.cancel();
  usb_control_.setIrInterfaceState(UsbControl::Disabled);
  command_tx_.execute(SetModeEnabledWith0x00640064Command(nextCommandSeq()), result);
  command_tx_.execute(SetModeDisabledCommand(nextCommandSeq()), result);
  // command_tx_.execute(StopCommand(nextCommandSeq()), result);
  command_tx_.execute(SetStreamDisabledCommand(nextCommandSeq()), result);
  command_tx_.execute(SetModeEnabledCommand(nextCommandSeq()), result);
  command_tx_.execute(SetModeDisabledCommand(nextCommandSeq()), result);
  command_tx_.execute(SetModeEnabledCommand(nextCommandSeq()), result);
  command_tx_.execute(SetModeDisabledCommand(nextCommandSeq()), result);
  usb_control_.setVideoTransferFunctionState(UsbControl::Disabled);
  // close()
  command_tx_.execute(SetModeEnabledWith0x00640064Command(nextCommandSeq()), result);
  command_tx_.execute(SetModeDisabledCommand(nextCommandSeq()), result);
  // command_tx_.execute(ShutdownCommand(nextCommandSeq()), result);
  libusb_close(usb_device_handle_);
  exit()

If some of commented out lines executed, it will reboot with high probability.

And I found that it will shutdown and reboot if you insert short sleep before exit.

> Protonect
  while(!protonect_shutdown) {
    listener.waitForNewFrame(frames);
    registration->apply(rgb, depth, &undistorted, &registered);
  }
  // stop()
  libfreenect2::this_thread::sleep_for(libfreenect2::chrono::milliseconds(X));  // X = 1 ~ 500ms
  exit(1);

attached big log (22MB) tells that it will shutdown and reboot depending on how long it slept. And it looks cycling regularity, multiples of 33ms.
(Attached contains patches, scripts and more outputs.)

sleep duration (ms) : shutdown and reboot / number of trials
>81:    0 / 100  err=0    |                                                  |
82:    0 / 100  err=0    |                                                  |
83:    0 / 100  err=0    |                                                  |
84:    0 / 100  err=0    |                                                  |
85:    0 / 100  err=0    |                                                  |
86:    0 / 100  err=0    |                                                  |
87:    0 / 100  err=0    |                                                  |
88:    0 / 100  err=0    |                                                  |
89:    1 / 100  err=0    |                                                  |
90:    0 / 100  err=0    |                                                  |
91:    0 / 100  err=0    |                                                  |
92:    0 / 100  err=0    |                                                  |
93:    0 / 100  err=0    |                                                  |
94:    1 / 100  err=0    |                                                  |
95:    0 / 100  err=0    |                                                  |
96:    1 / 100  err=0    |                                                  |
97:    1 / 100  err=0    |                                                  |
98:    0 / 100  err=0    |                                                  |
99:    0 / 100  err=0    |                                                  |
100:    0 / 100  err=0    |                                                  |
[... 48 lines cut ...]
198:    0 / 100  err=0    |                                                  |
200:    0 / 100  err=0    |                                                  |
205:    3 / 100  err=0    |*                                                 |
210:    2 / 100  err=0    |*                                                 |
215:    0 / 100  err=0    |                                                  |
220:   13 / 100  err=0    |******                                            |
225:   22 / 100  err=55   |***********                                       |
230:    0 / 100  err=65   |                                                  |
235:    0 / 100  err=0    |                                                  |
240:    8 / 100  err=75   |****                                              |
245:    0 / 100  err=100  |                                                  |
250:    3 / 100  err=55   |*                                                 |
255:   15 / 100  err=0    |*******                                           |
260:    0 / 100  err=0    |                                                  |
265:    0 / 100  err=0    |                                                  |
270:    0 / 100  err=0    |                                                  |
275:    0 / 100  err=0    |                                                  |
280:    0 / 100  err=0    |                                                  |
285:    0 / 100  err=0    |                                                  |
290:    0 / 100  err=0    |                                                  |
295:    1 / 100  err=0    |                                                  |
300:    0 / 100  err=0    |                                                  |
305:    0 / 100  err=0    |                                                  |

[test.sh-20160211-102045.log.txt.zip](https://github.com/OpenKinect/libfreenect2/files/129620/test.sh-20160211-102045.log.txt.zip)

hanyazou, 2016-02-14:
According to these results, I think we can't avoid shutdown and reboot on exit. So I made a patch which tap the timing of issuing the shutdown command to wait the device back.

https://github.com/hanyazou/libfreenect2/commit/87e6ac98ef8d12926a9744958bab151df49ef55b

It's just a proof of concept and should be enhanced. But i'd like to hear from your comments.

xlz, 2016-02-14:
Thank you for the tremendous results. The results are too huge for me to parse so I have to trust your conclusion.

But I still don't understand why cancelling transfers would make it reboot?

> If some of commented out lines executed, it will reboot with high probability.

Do you mean: if **any one** of the commented lines are executed, the device will reboot with high probability?

Also I assume `// stop()` and `// close()` are not actually lines of code in your testing, just labeling the functions?

> libusb_close() cause segmentation violation randomly

Can you provide a stack trace? At least this segfault can be fixed.

> cycling regularity, multiples of 33ms

Periodic. And note that the range of the time delay causing reboots is ~12ms (Check the [isochronous transfer timing pattern](https://github.com/OpenKinect/libfreenect2/wiki/Linux-USB-Notes#design-of-alternative-transfer-pool), each packet takes ~12ms). So this means if the program exits during an active isochronous transfer, something will happen and cause the device to reboot.

Can you use Mac's USB Prober to provide some insight on what is happening at kernel level? 

> https://github.com/hanyazou/libfreenect2/commit/87e6ac98ef8d12926a9744958bab151df49ef55b

First, you can't change the API, which breaks the ABI. 

Removing `ShutdownCommand` altogether is OK, as I tested on other platforms.

Because this bug is specific to your platform and there is no way to identify the platforms/models of Mac that are affected, this cannot be enabled by default. Usually this kind of thing is done as a driver quirk. In this case, you can use e.g. `getenv("LIBFREENECT2_WAIT_FOR_REBOOT")` to conditionally execute the waiting code.

There needs to be some kind of waiting code indeed. And the detail of the waiting code can be improved indeed, but the detail is not important now.

hanyazou, 2016-02-15:
>Do you mean: if any one of the commented lines are executed, the device will reboot with high probability?

yes.

>Also I assume // stop() and // close() are not actually lines of code in your testing, just labeling the functions?

yes.

> > libusb_close() cause segmentation violation randomly

> Can you provide a stack trace? At least this segfault can be fixed.

I will investigate this separately and I will submit another issue.

> So this means if the program exits during an active isochronous transfer, something will happen and cause the device to reboot.

I've added some sleep before cancel(). I thought I can exit without reboot with this code bellow,

> while(!protonect_shutdown) {
listener.waitForNewFrame(frames);
registration->apply(rgb, depth, &undistorted, &registered);
}
rgb_transfer_pool_.disableSubmission();
ir_transfer_pool_.disableSubmission();
libfreenect2::this_thread::sleep_for(libfreenect2::chrono::milliseconds(X)); // X = 1 ~ 200ms
rgb_transfer_pool_.cancel();
ir_transfer_pool_.cancel();
exit(1);

But the result was opposite, it shutdown and reboot 100% on exit() regardless of sleep duration.

> Can you use Mac's USB Prober to provide some insight on what is happening at kernel level?

I've tried the USB Prober at first.  USB Logger returns an error bellow,

> USB Prober: [ERR] Could not IOServiceOpen on USB Controller client e00002c7

According to IOReturn.h, e00002c7 means 

> #define kIOReturnUnsupported     iokit_common_err(0x2c7) // unsupported function

So I could not use the tool.
(Yes. You are right. I need some tool like USB Prober.) 

I will spend more time on this problem next weekend.

hanyazou, 2016-02-15:
I'd prepared another pair of Mac and Kinect sensor and executed same test. It shows the same tendency.

1. iMac mid 2013 (nDIVIA GPU) + Kinect sensor (Xbox One day one version, 2013) + for Windows v3 adaptor (purchased in 2015)
2. MacBook Pro mid 2015 (AMD GPU) + Kinect for Windows v2 (purchased in 2015) 

Both of two mac were installed OS X 10.11. I newly installed the OS to MacBook Pro for this trial so that another application like VMware does not affect the test result.
These Macs have different CPU/GPU and manufacturing year. This means the problem is not a failure of my iMac nor my Kinect sensor.

xlz, 2016-02-19:
I have commented out `command_tx_.execute(ShutdownCommand(nextCommandSeq()), result);` in the latest master.

hanyazou, 2016-02-21:
OS X 10.11 (El Capitan) seems to have problem(s)
-------------------
I've clean installed 3 versions of Mac OS X and ran the same test script and Protonect which was build from same source code. Test target is modified Protonect derived from v0.1.1 not to issue the shutdown USB command.  The script and modified Protonect can be retrieved from my test branch, https://github.com/hanyazou/libfreenect2/tree/shutdown_test.

The results are here. OS X 10.11 (latest version) seems to have problem obviously. The previous version 10.10 also reproduced this shutdown issue once.

>OS X 10.9.5    **shutdown=0,** error=6 / trial=100  
https://github.com/hanyazou/libfreenect2/blob/e88a6e7b24f91780e78b7ba8708a13b20472d21a/test-logs/test.sh-20160220-171316.txt

>OS X 10.10.5  **shutdown=1,** error=7 / trial=100  
https://github.com/hanyazou/libfreenect2/blob/f20db16b9017c44f2e83e4f9b1b2d3a257839f9e/test-logs/test.sh-20160221-071707.txt

>OS X 10.11.3  **shutdown=90**, error=10 / trial=100   
https://github.com/hanyazou/libfreenect2/blob/23895a29c5917f329466223c06a25e3d901b2b40/test-logs/test.sh-20160220-194359.txt

Test procedure is very short after OS clean installation. 
(no additional apps, utils and extensions. no VMware nor Xcode.)

>$ history
    1  /usr/bin/ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install)"
    2  mkdir github
    3  cd github
    4  git clone https://github.com/hanyazou/libfreenect2.git libfreenect2-hanyazou
    6  brew install wget
    7  brew install git
    8  brew install cmake
    9  brew install pkg-config
   10  brew install libusb
   11  brew tap homebrew/versions
   12  brew install glfw3
   13  brew tap homebrew/science
   14  brew install jpeg-turbo
   15  cd libfreenect2-hanyazou/
   16  mkdir build
   17  cd build
   18  git checkout -b shutdown_test origin/shutdown_test
   19  cmake ..
   20  make
   21  sh ../test.sh

Shutdown issue can be reproduced on Mavericks
-------------------
Run this script bellow and hit ESC key as soon as possible after the Protonect image window pops up or hit ESC key repeatedly like 'F5 attack'. It display '#### NO DEVICE' and stop if the Kinect sensor goes shutdown and disappears. I can see the '#### NO DEVICE' output with probability 1/3 ~ 1/10 on OS X 10.9.

>#!/bin/bash
while date; do 
  echo `date "+%h %d %H:%M:%S.000"` "#### EXEC Protonect"
  # --debug_flags 253 is for exclude shutdown command submission in close().
  ./bin/Protonect --debug_flags 253 cl
  sleep 1
  if system_profiler SPUSBDataType | grep Xbox; then
    :
  else
    echo `date "+%h %d %H:%M:%S.000"` "#### NO DEVICE"
    exit 1
  fi
done

I've captured USBProber log while this script was running. (atatched)
(I could run USBProber logger function on OS X 10.9 only.)

[kinect-shutdown.log.txt](https://github.com/OpenKinect/libfreenect2/files/139400/kinect-shutdown.log.txt)

I guess the Kinect sensor might disappear around on Feb 20 16:52:36.849. But I can't find any notable error nor exception message.

Can you find some hint from this USBProber log?

I think this script can be executed on Linux if you replace the system_profiler command with lsusb and delete my debug_flags option. I will try when I have a chance.

xlz, 2016-02-21:
I don't understand your data. It says "shutdown=0" but later you said 10.9 does reboot the Kinect. Right now I interpret that as the "shutdown=0" test case did not show the error but 10.9 does have the error.

> might disappear around on Feb 20 16:52:36.849

Regardless of time, did it disappear or did it not? And did it come back?

I have not seen this issue on Linux. Okay, if you manage to install Linux on Mac it might be possible. That would quite some gymnastics though.

xlz, 2016-02-21:
RemoveDevice() does look suspicious. But its log doesn't look complete.

I think you need to use the usbtracer tool. A lot of logging points include information with `kTPHubPortRemoveDevice`.

xlz, 2016-02-21:
I can't find any entry points to RemoveDevice() if the log is complete. I used this source as reference https://opensource.apple.com/tarballs/IOUSBFamily/IOUSBFamily-560.4.2.tar.gz

hanyazou, 2016-02-21:
>I don't understand your data. It says "shutdown=0" but later you said 10.9 does reboot the Kinect. Right now I interpret that as the "shutdown=0" test case did not show the error but 10.9 does have the error.

Sorry. There was two test script.
The 1st  test script kill -INT $pid after 12~15 seconds launching Protonect. This script exec Protonect 100 times automatically. This script ran w/o shutdown on OS X 10.9.
You can see this script at https://github.com/hanyazou/libfreenect2/blob/shutdown_test/test.sh

2nd script is for manual testing.
https://github.com/hanyazou/libfreenect2/blob/shutdown_test/test-by-hand.sh
With 2nd script and hitting ESC key rapidly, I could reproduce the shutdown on OS X 10.9. 

>But its log doesn't look complete.

I will try the usbtracer.

hanyazou, 2016-02-21:
I could reproduce in simpler instruction.

>$ git clone https://github.com/OpenKinect/libfreenect2.git
$ cd libfreenect2
$ git branch
`* master`
$ git log | head
commit 9ef859612de01dbb0fe4a6bd2b3a8ed19671c00e
Author: Thiemo Wiedemeyer <wiedemeyer@cs.uni-bremen.de>
Date:   Fri Feb 19 13:37:54 2016 +0100
    opencl: Use a different profiling macro
    Enabling profiling in OpenCL effects the performance, so for
    profiling libfreenect2s processors, it should be disabled and only
    used when testing improvements of the OpenCL code itself.
$ git diff | cat
diff --git a/examples/Protonect.cpp b/examples/Protonect.cpp
[patch.txt](https://github.com/OpenKinect/libfreenect2/files/139452/patch.txt)
$ mkdir build
$ cd build
$ cmake ..
$ make
./bin/Protonect cl --frames 10

The Protonect shows the image window to display only 10 frames and exit then.
This reproduce the shutdown issue with possibility 1/2 ~ 1/10 on Max OS X 10.9.5.

I've executed this modified Protonect and usbtracer simultaneously.
>on terminal 1: ./usbtracer > ~/usbtracer.log 2>&1 &
on terminal 2: ./bin/Protonect cl --frames 10

attached usbtracer-shutdown.log.txt is the log of a case that the sensor device disappeared and be back after a few seconds. This log contains 'HubPort::RemoveDevice' lines.
[usbtracer-shutdown.log.txt](https://github.com/OpenKinect/libfreenect2/files/139456/usbtracer-shutdown.log.txt)

another log usbtracer-wo-shutdown.log.txt is a case that the sensor did not disappear. This log does not contain any 'remove'.
[usbtracer-wo-shutdown.log.txt](https://github.com/OpenKinect/libfreenect2/files/139458/usbtracer-wo-shutdown.log.txt)

I'm not sure I could capture the logs properly. There are too many lines I can't understand.

hanyazou, 2016-02-21:
I tried same Protonect patch on Linux (Jotson TK1). Now I'm sure that my Kinect sensor is not broken and the Linux does not have this shutdown issue. The modified Protonect could be re-execute very quickly for several dozen times.
[patch.txt](https://github.com/OpenKinect/libfreenect2/files/139452/patch.txt)

>for i in {1..50}; do
  echo ================ $i
  lsusb -v 2>/dev/null| | grep -i nui
  DISPLAY=:0 ./bin/Protonect gl --frames 1 || break
done

off-topic: TK1 kernel might have another instability problem.  Protonect stopped at $i=81 with segmentation fault and dmesg said,
>[ 1851.612674] tegra-xhci tegra-xhci: xHCI host not responding to stop endpoint command.
[ 1851.612893] tegra-xhci tegra-xhci: Assuming host is dying, halting host.
[ 1851.614590] tegra-xhci tegra-xhci: HC died; cleaning up
[ 1851.620989] tegra-xhci tegra-xhci: Timeout while waiting for configure endpoint command
[ 1851.621153] tegra-xhci tegra-xhci: Abort the command ring, but the xHCI is dead.

xlz, 2016-02-21:
> $i=81

What does this mean...

hanyazou, 2016-02-21:
@xlz when I tried "for i in {1..100}; do ... done", it stopped at i=81.

xlz, 2016-02-21:
Is there anything else in dmesg other what you posted? tegra-xhci is not super good indeed. It's a miracle it works at all.

hanyazou, 2016-02-21:
@xlz Sorry. The messages were lost already because I ran the script then. Dmesg only output these lines for now.

>[  130.504887] usb 2-1.1: reset SuperSpeed USB device number 3 using tegra-xhci
[  130.517048] usb 2-1.1: Parent hub missing LPM exit latency info.  Power management will be impacted.
[  130.582203] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 2 with no TDs queued?
[  130.727220] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 2 with no TDs queued?
[  130.734937] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 2 with no TDs queued?

xlz, 2016-02-21:
These are not errors. If you have anything before "xHCI host not responding to stop endpoint command", post it.

hanyazou, 2016-02-21:
@xlz I've tried rebooting and running my script several times. I found that my scripts could run i = 1 to 200~300 after reboot. Then, the kernel starts to kill the Protonect process by OOM killer. This means the kernel (may be the tegra-xhci) leaks some memory.  I couldn't reproduce 'tegra-xhci: xHCI host not responding to stop endpoint command.' But I can guess that the kernel and the driver couldn't use enough memory at that time. Attached dmesg log contains various memory warnings and errors.

[tegra-ubuntu-dmesg.txt](https://github.com/OpenKinect/libfreenect2/files/139506/tegra-ubuntu-dmesg.txt)

xlz, 2016-02-21:
I looked at the usbtracer logs. They are verbose but readable. Though I still couldn't find who called `RemoveDevice()`, which is really important.

Here is another way that looks promising. You can use DTrace for better diagnosis: http://lists.apple.com/archives/usb/2015/Jan/msg00004.html

I suppose you would do this
```
sudo dtrace -n '*:*:*HubPort:*:entry { stack(); }':
```
to produce a stack trace and see who calls `RemoveDevice()`.

Another thing is we need to beware of whether the device really disappears *because of* `RemoveDevice()` and eliminate the possibility of something else makes the device disappear.

xlz, 2016-02-22:
Any update on the dtrace method? It should be easy to do.

hanyazou, 2016-02-23:
It seems that `RemoveDevice` was called because USB controller detect that the sensor was removed.

>term-1$ ./bin/Protonect cl -frames 1

>term-2$ sudo dtrace -n '\*:\*:\*RemoveDevice\*:entry { stack(); ustack(); }'
dtrace: description '\*:\*:\*RemoveDevice\*:entry ' matched 7 probes
CPU     ID                    FUNCTION:NAME
  2 132694 _ZN15AppleUSBHubPort12RemoveDeviceEv:entry 
              AppleUSBHub`AppleUSBHubPort::DefaultConnectionChangeHandler(unsigned short, unsigned short)+0x54b
              AppleUSBHub`AppleUSBHubPort::PortStatusChangedHandler()+0x573
              AppleUSBHub`AppleUSBHubPort::PortStatusChangedHandlerEntry(OSObject*)+0x2b
              mach_kernel`0xffffff802e04a010+0x1fa
              mach_kernel`call_continuation+0x17
>
>
>
>^C

>term-2$ sudo dtrace -n '\*:\*:\*HubPort\*:entry { stack(); }'
dtrace: description '\*:\*:\*HubPort\*:entry ' matched 106 probes
CPU     ID                    FUNCTION:NAME
  6 142900 _ZN12AppleUSBEHCI20GetRootHubPortStatusEP14IOUSBHubStatust:entry 
              AppleUSBEHCI`AppleUSBEHCI::UIMRootHubStatusChange()+0x102
              IOUSBFamily`IOUSBControllerV3::CheckForRootHubChanges()+0x1e
              IOUSBFamily`IOUSBControllerV3::RootHubTimerFired(OSObject*, IOTimerEventSource*)+0x2ff
              mach_kernel`IOTimerEventSource::timeoutAndRelease(void*, void*)+0x10b
              mach_kernel`0xffffff802e04a010+0x1fa
              mach_kernel`call_continuation+0x17
>
>  0 142900 _ZN12AppleUSBEHCI20GetRootHubPortStatusEP14IOUSBHubStatust:entry 
              AppleUSBEHCI`AppleUSBEHCI::UIMRootHubStatusChange()+0x102
              IOUSBFamily`IOUSBControllerV3::CheckForRootHubChanges()+0x1e
              IOUSBFamily`IOUSBControllerV3::RootHubTimerFired(OSObject*, IOTimerEventSource*)+0x2ff
              mach_kernel`IOTimerEventSource::timeoutAndRelease(void*, void*)+0x10b
              mach_kernel`0xffffff802e04a010+0x1fa
              mach_kernel`call_continuation+0x17
... repeat ...

xlz, 2016-02-23:
`AppleUSBEHCI` is USB 2 controller. This part is irrelevant.

Is there anything from XHCI?

xlz, 2016-02-23:
It looks like a lot of USBLog didn't appear in the usb prober log.

xlz, 2016-02-23:
Let me trace the call chain
```
  2 132694 _ZN15AppleUSBHubPort12RemoveDeviceEv:entry 
              AppleUSBHub`AppleUSBHubPort::DefaultConnectionChangeHandler(unsigned short, unsigned short)+0x54b
              AppleUSBHub`AppleUSBHubPort::PortStatusChangedHandler()+0x573
              AppleUSBHub`AppleUSBHubPort::PortStatusChangedHandlerEntry(OSObject*)+0x2b
              mach_kernel`0xffffff802e04a010+0x1fa
              mach_kernel`call_continuation+0x17
_portStatusChangedHandlerThread
thread_call_enter(_portStatusChangedHandlerThread)
AppleUSBHubPort::StatusChanged
AppleUSBHub::ProcessStatusChanged
AppleUSBHub::ProcessStatusChangedEntry
...call_continuation
_workThread
thread_call_enter(_workThread)
AppleUSBHub::InterruptReadHandler
```
So basically the Kinect adaptor hub tells Apple Hub driver that this port is disconnected (kHubPortConnection == 0).

The question is what makes the port disconnect?

xlz, 2016-02-24:
I give up. I can't find anything meaningful before `AppleUSBHub::InterruptReadHandler`.

Here is one last proposal:
* Add a 10 second sleep after `disableSubmission()`
* And don't comment out the `StopCommand` (still comment out the `ShutdownCommand`)

Maybe something wrong happens if we call `libusb_transfer_cancel()` (in the middle of active transfers - as is also seen in other examples, exiting in the middle of active transfers makes it disconnect).

The alternative is to let active transfers naturally run out and do not submit any new transfers. This is what `disableSubmission()` does. Now the code waiting for all transfers to finish is in `.cancel()`, which we replace with a 10-second sleep.
```
Protonect
while(!protonect_shutdown) {
listener.waitForNewFrame(frames);
registration->apply(rgb, depth, &undistorted, &registered);
}
// stop()
rgb_transfer_pool_.disableSubmission();
ir_transfer_pool_.disableSubmission();
libfreenect2::this_thread::sleep_for(libfreenect2::chrono::milliseconds(10*1000)); 
// rgb_transfer_pool_.cancel();
// ir_transfer_pool_.cancel();
usb_control_.setIrInterfaceState(UsbControl::Disabled);
command_tx_.execute(SetModeEnabledWith0x00640064Command(nextCommandSeq()), result);
command_tx_.execute(SetModeDisabledCommand(nextCommandSeq()), result);
command_tx_.execute(StopCommand(nextCommandSeq()), result);
command_tx_.execute(SetStreamDisabledCommand(nextCommandSeq()), result);
command_tx_.execute(SetModeEnabledCommand(nextCommandSeq()), result);
command_tx_.execute(SetModeDisabledCommand(nextCommandSeq()), result);
command_tx_.execute(SetModeEnabledCommand(nextCommandSeq()), result);
command_tx_.execute(SetModeDisabledCommand(nextCommandSeq()), result);
usb_control_.setVideoTransferFunctionState(UsbControl::Disabled);
// close()
command_tx_.execute(SetModeEnabledWith0x00640064Command(nextCommandSeq()), result);
command_tx_.execute(SetModeDisabledCommand(nextCommandSeq()), result);
// command_tx_.execute(ShutdownCommand(nextCommandSeq()), result);
libusb_close(usb_device_handle_);
exit()
```

hanyazou, 2016-02-27:
>Here is one last proposal:
Add a 10 second sleep after disableSubmission()
And don't comment out the StopCommand (still comment out the ShutdownCommand)

Your proposal works on OS X 10.9 only. The Kinect always shutdown and reboot unexpectedly on exiting Protonect on OS X 10.11.

I could avoid this unexpected shutdown by issuing explicit shutdown command  like https://github.com/OpenKinect/libfreenect2/issues/539#issuecomment-183874645 (please refer `modification - 2` bellow).

But I found that the shutdown and reboot introduced another problem. Sometimes the Protonect stopped  around `status 0x090000: 1024` until I did un-plug and re-plug the device manually.  This is independent from that the shutdown is un-expected or explicit. And both OS version have the same problem. I tried libusb_reset_device() on the Xbox NUI sensor and on the Nui Sensor Adaptor both. But it didn't solve the problem. Re-execute the Protonect w/o un-plug and re-plug didn't solve the problem also.

Mac OS vers
--------------------
10.9 Maverick
>bash-3.2$ system_profiler SPSoftwareDataType
Software:
    System Software Overview:
      System Version: OS X 10.9.5 (13F34)
      Kernel Version: Darwin 13.4.0
      Boot Volume: OS X 10.9 Mavericks
      Boot Mode: Normal
      Computer Name: iMac Mavericks
      User Name: Zou Hanya (hanyazou)
      Secure Virtual Memory: Enabled
      Time since boot: 2 days 6:37

10.11 El Capitan
>bash-3.2$ system_profiler SPSoftwareDataType
Software:
    System Software Overview:
      System Version: OS X 10.11.3 (15D21)
      Kernel Version: Darwin 15.3.0
      Boot Volume: OS X 10.11 El Capitan
      Boot Mode: Normal
      Computer Name: Zou’s iMac
      User Name: Zou Hanya (hanyazou)
      Secure Virtual Memory: Enabled
      System Integrity Protection: Enabled
      Time since boot: 43 minutes

source code
--------------------
> bash-3.2$ git log -1 --oneline
664ff3c Protonect: Add '-frames' option

Modification - 1
>bash-3.2$ git diff
diff --git a/src/libfreenect2.cpp b/src/libfreenect2.cpp
index db8adf1..77102e9 100644
--- a/src/libfreenect2.cpp
+++ b/src/libfreenect2.cpp
@@ -782,15 +782,16 @@ bool Freenect2DeviceImpl::stop()
   {
     LOG_INFO << "canceling rgb transfers...";
     rgb_transfer_pool_.disableSubmission();
\-    rgb_transfer_pool_.cancel();
   }
   if (ir_transfer_pool_.enabled())
   {
     LOG_INFO << "canceling depth transfers...";
     ir_transfer_pool_.disableSubmission();
\-    ir_transfer_pool_.cancel();
   }
\+  libfreenect2::this_thread::sleep_for(libfreenect2::chrono::milliseconds(10*1000));

Modification - 2
>bash-3.2$ git diff
diff --git a/src/libfreenect2.cpp b/src/libfreenect2.cpp
index db8adf1..b72e7b7 100644
--- a/src/libfreenect2.cpp
+++ b/src/libfreenect2.cpp
@@ -832,7 +832,8 @@ bool Freenect2DeviceImpl::close()
   /\* This command actually reboots the device and makes it disappear for 3 seconds.
    \* Protonect can restart instantly without it.
    \*/
\-  //command_tx_.execute(ShutdownCommand(nextCommandSeq()), result);
\+  command_tx_.execute(ShutdownCommand(nextCommandSeq()), result);
\+  libfreenect2::this_thread::sleep_for(libfreenect2::chrono::milliseconds(5*1000));

test script
-------------
>#!/bin/bash
shutdown=0
error=0
shutdown_detected()
{
  echo "#### SHUTDOWN"
  shutdown=$(($shutdown + 1))
}
error_detected()
{
  echo "#### ERROR"
  error=$(($error + 1))
}
for i in {1..100}; do
  ./bin/Protonect cl -frames 1 || error_detected
  sleep 1
  if ioreg -p IOUSB | grep -i xbox; then
    :
  else
    shutdown_detected
    sleep 5
  fi
  echo "####  shutdown=$shutdown / $i  error=$error"
done

results
--------------
OX X 10.9 w/o MOD    shutdown=23 error=0
OS X 10.9 MOD-1       shutdown=0 error=0
OS X 10.9 MOD-2       shutdown=0 error=4
OS X 10.11 w/o MOD  shutdown=100 error=1
OS X 10.11 MOD-1     shutdown=100 error=9
OS X 10.11 MOD-2     shutdown=0 error=6

error
--------------
The error above was that Protonect was waiting for the changes of the status 0x090000 forever after shutdown and reboot the device.

>Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) Core(TM) i7-3770 CPU @ 3.40GHz^@ (CPU)[Intel^@]
[Info] [OpenCLDepthPacketProcessorImpl]   1: GeForce GTX 675MX^@ (GPU)[NVIDIA^@]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: GeForce GTX 675MX^@ (GPU)[NVIDIA^@]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @20:18 with serial 005697142447
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 1024

**// The status didn't change. I must unplug and plug the USB cable of Kinect.**

>[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_IO Input/Output Error
[Info] [Freenect2DeviceImpl] closing...
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_PIPE Pipe error
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_PIPE Pipe error
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_PIPE Pipe error
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed

xlz, 2016-02-27:
OK, Apple's USB is seriously messed up. Literally, let's have it as
```
#ifdef __APPLE__
  send shutdown command
  sleep 10 seconds
#endif
```
This is the price you have to pay for using Apple.

About being stuck in the status checking, I don't know if it's really essential to wait for it forever. Maybe you can set a limit of retries for that loop and see what happens.

commit 9c80ccd6 (2016-02-28): usb: Issue reboot command on Mac OS X

Without the ShutdownCommand, the Kinect still disappears randomly
on Mac OS X. Painstaking effort did not determine the cause.

So take the suboptimal way and shut it down explicitly.

Fixed #539.
commit bd08eb65 (2016-03-21): usb: Issue reboot command on Mac OS X

Without the ShutdownCommand, the Kinect still disappears randomly
on Mac OS X. Painstaking effort did not determine the cause.

So take the suboptimal way and shut it down explicitly.

Fixed #539.
closed 2016-02-28 by xlz