OpenKinect/libfreenect2 #541: Release 0.1.0 tarball checksum changed

de-vri-es, 2016-01-26:
The checksum for the 0.1.0 release tarball seems to have changed. Back in 2015-12-29 the sha256 checksum for `v0.1.0.tar.gz` was `ef4c41c16c2949c8e3513999fdc33160964d693554ed84c05843581c20a809d2`. Now it is `1617eff0bfd2f4a509617beca3a0b4b3a4a3b2f0f2e5f3ce6210ea3087d84d50`.

Not a big issue, but in #537 it was said that release tarballs do not change. It's probably good to be aware that, for some reason, it did change.

xlz, 2016-01-27:
0.1.0 was released on Jan 18, 2016. Where are you getting the tarballs?

de-vri-es, 2016-01-27:
Oops, no clue why I said 2015-12-29. I meant 2016-01-21.

floe, 2016-01-27:
Weird, I just downloaded both archives and I get

    5972e85e7ef625f5f1c91f49a712c2d03745488d5954006c36cf2d31dc37aeca  libfreenect2-0.1.0.tar.gz
    c9130f933583028141f1e8ec8edab85e8932c77d9ee93ba4026ceb82d2d105ae  libfreenect2-0.1.0.zip

None of which match your SHA256. Don't have a clue right now what's happening.

de-vri-es, 2016-01-27:
Aha, it seems I made a slight mistake in the URL. I forgot the trailing `.0` in `0.1.0`. Not sure how that works since there is no `0.1` release. And the checksum I got does not match any of the release tarballs.

So I wonder what github is doing here, but there is probably no issue with the correct URL.

closed 2016-01-27 by de-vri-es

floe, 2016-01-27:
I think since there's a v0.1 branch, Github will auto-generate a snapshot tarball based on that URL. Good to know :-)