OpenKinect/libfreenect2 #542: OpenNi2 driver - image convert to opencv

kruss84, 2016-01-27:
I'm trying to convert an image from openni::VideoFrameRef to cv::Mat.
When I use the examples given for color and depth, I just get gray images.
Resolution of the images is correct, only the data is bad.

http://stackoverflow.com/questions/15966722/capture-rgb-from-kinnect-with-openni-and-show-with-opencv
http://answers.opencv.org/question/33513/opencv-asus-xtion-sensor/

When I start the program SimpleRead of Openni2, the depth image looks very good.
The readout of the Kinect v2 is working perfectly.

I would appreciate any suggestions or ideas to solve this problem.

xlz, 2016-01-27:
@hanyazou

hanyazou, 2016-01-28:
@kruss84 I will investigate the driver and opencv sample code. But I don't have a lot of experience with opencv. Could you please provide your source code include main() so that I can compile it without adding lines.

closed 2016-02-29 by kruss84