OpenKinect/libfreenect2 #543: Jetson TK1 [RgbPacketStreamParser] skipping rgb packet!

vaghelan, 2016-01-29:
Hi,

Just downloaded latest sources from master.

I get error when run protonect for Kinect v2. Any ideas why ?

**[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!**

Version: 0.1.0
Environment variables: LOGFILE=<protonect.log>
Usage: bin/Protonect [gl | cl | cpu] [<device serial>] [-noviewer] [-help] [-version]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:7 with serial 198085435247
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9855
[Debug] [Freenect2DeviceImpl] status 0x090000: 9855
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 198085435247
device firmware: 4.0.3911.0
**[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!**
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [RgbPacketStreamParser] skipping rgb packet!

xlz, 2016-01-29:
This error happening only once at the start can be ignored

vaghelan, 2016-01-29:
But then I do not see anything in the window that opens and on the screen continuously skipping rgb packet! error is displayed....

vaghelan, 2016-01-29:
Protonect Windows is blank....

xlz, 2016-01-29:
have you maximized cpu frequency?

vaghelan, 2016-01-29:
Can you please tell me how to do that ?

xlz, 2016-01-29:
http://elinux.org/Jetson/Performance

closed 2016-02-14 by xlz

oneTop, 2017-09-01:
Hi,
    I have a question like yours. And, I have maximized cpu frequency, but the problem remains unresolved.
Can you give me some advice? Looking forward to your reply.