OpenKinect/libfreenect2 #545: Installation isn't working. Cmake is failing

mozilla2012, 2016-01-29:
Trying to install on 14.04 LTS, and I _think_ I do everything right when I get this output:

```
-- using tinythread as threading library
-- Could NOT find OpenCL (missing:  OpenCL_LIBRARY) (found version "1.2")
-- Linking with these libraries: 
 /usr/lib/x86_64-linux-gnu/libusb-1.0.so
 /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0
 pthread
 /usr/lib/x86_64-linux-gnu/libglfw.so
 /usr/lib/x86_64-linux-gnu/libGL.so
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- Configurating examples
-- Using in-tree freenect2 target
-- checking for module 'libopenni2'
--   package 'libopenni2' not found
-- Could NOT find OpenNI2 (missing:  OpenNI2_LIBRARY_DIR OpenNI2_INCLUDE_DIRS) 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/team23/libfreenect2/build
```

It appears I'm missing OpenNI2 and OpenCL. Easy fix, right?

It says in the instructions: `sudo apt-get install libopenni2-dev` to install OpenNI2 but it's giving me an error:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libopenni2-dev
```

Not sure why this is happening. This is a fresh install of 14.04, so it should be able to see this.
I look in the [package search](http://packages.ubuntu.com/search?keywords=libopenni2-dev&searchon=names&suite=wily&section=all) and the package is only listed for Wily (15.10).

How can I get it to install 14.04? The instructions are now out of date. Also, some clarification for how to install OpenCL would be great because I'm not sure about that either. But that's for another issue.

Thanks!

xlz, 2016-01-29:
The instruction says OpenNI2 and OpenCL are optional components (did you see this?). If you don't have them, nothing is wrong.

mozilla2012, 2016-01-29:
Ah, I see now it wasn't really failing... Gotcha. Thanks for updating the documentation!

closed 2016-01-29 by mozilla2012

purna15111, 2018-08-28:
You can install the openni2 just by following the steps in this link this has helped me to solve https://www.howtoinstall.co/en/ubuntu/xenial/libopenni2-dev