OpenKinect/libfreenect2 #546: Protonect: "bulk transfer failed: LIBUSB_ERROR_TIMEOUT Operation timed out"

mozilla2012, 2016-01-29:
Hello!
When I run `sudo ./bin/Protonect` I get the following:

```
Version: 0.1.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [gl | cl | cpu] [<device serial>] [-noviewer] [-help] [-version]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:7 with serial 004285250747
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_TIMEOUT Operation timed out
```

I tried to Google this, but the only other person with a similar issue was running this on a virtual machine, whereas I'm running it on native 14.04...

Probably useful information: 
lsusb:
```
Bus 004 Device 004: ID 413c:3012 Dell Computer Corp. Optical Wheel Mouse
Bus 004 Device 003: ID 413c:2105 Dell Computer Corp. Model L100 Keyboard
Bus 004 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 007: ID 045e:02c4 Microsoft Corp. 
Bus 002 Device 006: ID 045e:02d9 Microsoft Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 045e:02d9 Microsoft Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

lspci | grep USB
```
00:1a.0 USB controller: Intel Corporation C600/X79 series chipset USB2 Enhanced Host Controller #2 (rev 05)
00:1d.0 USB controller: Intel Corporation C600/X79 series chipset USB2 Enhanced Host Controller #1 (rev 05)
08:00.0 USB controller: NEC Corporation uPD720200 USB 3.0 Host Controller (rev 04)
```

uname -a
`Linux team23-lab 3.19.0-47-generic #53~14.04.1-Ubuntu SMP Mon Jan 18 16:09:14 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
`

This is running on a fresh install of 14.04, and installing libfreenect2 was the first thing I did.

Thanks for your help!

xlz, 2016-01-29:
dmesg?

mozilla2012, 2016-01-29:
Here's what comes up each time I run it:

```
[  139.720750] usb 2-1.1: reset SuperSpeed USB device number 3 using xhci_hcd
[  139.738351] xhci_hcd 0000:08:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff880438a35200
[  139.738358] xhci_hcd 0000:08:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff880438a35248
[  139.738362] xhci_hcd 0000:08:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff880438a352d8
[  139.738365] xhci_hcd 0000:08:00.0: xHCI xhci_drop_endpoint called with disabled ep ffff880438a35290
[  140.760629] Protonect[2516]: segfault at 7f2ecf005010 ip 00007f2ee2a1eea8 sp 00007ffc542ca630 error 4 in libfreenect2.so.0.1.0[7f2ee2a05000+3a000]
```

I can paste the whole thing if you like.

xlz, 2016-01-29:
Try this kernel http://packages.ubuntu.com/trusty/linux-image-4.2.0-25-generic

mozilla2012, 2016-01-29:
Will that work for 14.04? Sorry, I'm not super experienced at this.

xlz, 2016-01-29:
Package: linux-image-4.2.0-25-generic (4.2.0-25.30~14.04.1) [security]

xlz, 2016-01-29:
uPD720200 is one of the earliest known working USB controller. It really should work as-is with slightly older kernels.

Try LIBUSB_DEBUG=4 ./bin/Protonect and post the log.

xlz, 2016-01-29:
By the way, the instruction does not ask you to use sudo ./bin/Protonect. Do not ever do that again.

mozilla2012, 2016-01-30:
Well, I just tried to update my kernel:

`sudo apt-get install linux-image-4.2.0-25-generic`

And it killed my machine. 
[image]

Now I need to figure that out...

Also, I need to do sudo ./bin/Protonect because otherwise I get permission errors or something similar. (I'd tell you exactly, but my computer isn't booting right now)

xlz, 2016-01-30:
Well, just enter grub to switch back to old kernel. Nothing too hard.

https://github.com/OpenKinect/libfreenect2#protonect-complains-about-no-device-connected-or-failure-opening-device

> If that fixes things, place platform/linux/udev/90-kinect2.rules into /etc/udev/rules.d/ and re-plug the device.

mozilla2012, 2016-01-30:
Ah, I see........................ my bad.
Should that fix the initial "Bulk transfer failed" issue, do you think?

xlz, 2016-01-30:
No.

What you need to do:
1. Follow the instruction
2. Do not use sudo ever
3. Try LIBUSB_DEBUG=4 ./bin/Protonect and post the log.

mozilla2012, 2016-01-30:
Alright, followed the instructions...
Without sudo, I get the attached.
[output.txt](https://github.com/OpenKinect/libfreenect2/files/110588/output.txt)

xlz, 2016-01-30:
Can you also post `lsusb -t`?

mozilla2012, 2016-01-30:
```
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
        |__ Port 5: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 6: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/1p, 5000M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 3, If 1, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 3, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
        |__ Port 1: Dev 3, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/1p, 480M

```

xlz, 2016-01-30:
1. `sudo sh -c 'for i in /sys/bus/usb/devices/*/power/autosuspend; do echo -1 >$i; done'`
2. `grep . /sys/bus/usb/devices/*/power/autosuspend`
3. Try again?

mozilla2012, 2016-01-30:
Will try this next time I am in lab. Thanks.

xlz, 2016-01-30:
The best I can get out of this is that this is just a timeout. It should not time out, but it does.

If this above disabling power management does not work, try to really use 4.2 kernel. If 4.2 kernel doesn't work either, you can try to use Windows to update the firmware of the USB controller. If that doesn't work either, I don't know what more can be done.

mozilla2012, 2016-01-30:
That's super weird. Thanks.
It is an older computer, but new enough to have USB 3.0s.

ankitvora7, 2016-01-31:
Do you have an idea of what else could have gone wrong ? Or maybe some mistake during the installation ? Is there a way to debug ?

xlz, 2016-01-31:
A way to debug: https://www.kernel.org/doc/Documentation/usb/usbmon.txt

Post `1.mon.out` here.

ankitvora7, 2016-01-31:
This is the output of 'sudo cat /sys/kernel/debug/usb/devices':
[buses.txt](https://github.com/OpenKinect/libfreenect2/files/111365/buses.txt)

From above I chose the Bus2(since Kinect is connected to Bus2) and hence the 1.mon.out is the output of the following command
'sudo cat /sys/kernel/debug/usb/usbmon/2u > /tmp/1.mon.out'
[1.mon.out.txt](https://github.com/OpenKinect/libfreenect2/files/111370/1.mon.out.txt)

See if what I did was alright!

mozilla2012, 2016-02-01:
Re:
> 1. `sudo sh -c 'for i in /sys/bus/usb/devices/*/power/autosuspend; do echo -1 >$i; done'`
> 2. `grep . /sys/bus/usb/devices/*/power/autosuspend`
> 3. Try again?

Didn't work. Bummer. Booting into Windows to update the 3.0 driver. (Which hopefully will work - it wasn't recognizing my 3.0 flash drive.....)

mozilla2012, 2016-02-01:
Went into Windows, updated drivers...

Still getting:
```
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_TIMEOUT Operation timed out
Segmentation fault (core dumped)
```

I don't know what to do either. I'll see if there are 3.0 drivers for Ubuntu that need updating, but I doubt it...

brendandburns, 2016-02-01:
I got similar errors until I updated my kernel to 4.2.x

I also found that the kinect sometimes gets wedged and I have to power-cycle it.

xlz, 2016-02-02:
@ankitvora7 Sorry the plain text format is too hard for me to parse. Can you obtain a pcap capture with https://ask.wireshark.org/questions/2801/usbmon-captures?

@brendandburns Are you confirming 4.2.x fixes the bug for you?

brendandburns, 2016-02-02:
@xlz I don't know if what I saw was the same bug, but it had the same output.

and yes, switching to 4.2.x fixed it for me.

however, I am also attempting to use a Texas Instruments USB 3.0 controller (TUSB73x0), which has generally been super flaky with the Kinect.  I've order a Renesas based card that I'm planning on switching to.

--brendan

floe, 2016-02-02:
USB3 as a whole has been a constant PITA in this project. So far, general experience seems to indicate the least problems with Intel or NEC/Renesas controllers (see also https://github.com/OpenKinect/libfreenect2/wiki/Linux-USB-Notes).

xlz, 2016-02-28:
NOTOURBUG.

This bug was introduced by https://github.com/torvalds/linux/commit/e210c422b6fdd2dc123bedc588f399aefd8bf9de (upstream 4.3-rc7, backported to Ubuntu kernel 3.19.0-41.46).

The offending commit was reverted in https://github.com/torvalds/linux/commit/a6835090716a85f2297668ba593bd00e1051e662 (upstream 4.5-rc3, [Ubuntu tracking bug](https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1529077), fix backported to Ubuntu kernel 3.19.0-49.55 (the Ubuntu bug claims it was fixed in 3.19.0-47.53, which is not true)).

I actually wrote about the original offending commit https://github.com/OpenKinect/libfreenect2/wiki/Linux-USB-Notes#warn-event-trb-for-slot-x-ep-2-with-no-tds-queued. Its seems to be intended for fixing the useless warnings, but the fix was defeated by USB controllers not compliant with specs, such as, NEC Corporation uPD720200.

For Ubuntu users, simply avoid kernels >=3.19.0-41 and <=3.19.0-47. (I'm not sure about the 4.2 series, but the fix should be carried in its latest update anyway.)

closed 2016-02-28 by xlz