OpenKinect/libfreenect2 #547: bulk transfer failed: LIBUSB_ERROR_TIMEOUT Operation timed out

ankitvora7, 2016-01-31:
Hi,
I'm facing an issue when I'm trying to run kinect v2 on ubuntu 14.04. Here's the terminal activity when I run './bin/Protonect':

```
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [gl | cl | cpu] [<device serial>] [-noviewer]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 001806651047
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] ReadData0x14 response
[Debug] [Freenect2DeviceImpl] 92 bytes of raw data
0x0000:  00 00 15 00 00 00 00 00 01 00 00 00 43 c1 1f 41    2e2e2e2e2e2e2e2e2e2e2e2e432e2e41
0x0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e2e
0x0020:  0a 21 33 55 c2 00 17 20 00 08 00 00 10 00 00 00    2e2133552e2e2e202e2e2e2e2e2e2e2e
0x0030:  00 01 00 00 00 10 00 00 00 00 80 00 01 00 00 00    2e2e2e2e2e2e2e2e2e2e802e2e2e2e2e
0x0040:  31 33 00 00 00 05 00 0d 47 4d 46 39 33 31 2e 31    31332e2e2e2e2e2e474d463933312e31
0x0050:  58 00 00 00 00 00 00 00 07 00 00 00                582e2e2e2e2e2e2e2e2e2e2e

[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_TIMEOUT Operation timed out
Segmentation fault (core dumped)

```
Here's my output for 'dpkg -l libusb-1.0-*':

```
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
ii  libusb-1.0-0:a 2:1.0.20-1pp amd64        userspace USB programming library
ii  libusb-1.0-0-d 2:1.0.20-1pp amd64        userspace USB programming library
un  libusb-1.0-doc <none>       <none>       (no description available)
```
I've tried whatever stuff was available on this forum but nothing seemed to have worked.

xlz, 2016-01-31:
You are using a quite old version. Use the master or a released version.

If the problem persists, check #546 and report the difference in your setup.

ankitvora7, 2016-01-31:
I tried installing the latest version but no luck. I also tried the power management commands mentioned on your referred post but again that didn't help. Here are the details w.r.t to the setup in #546 :
When I run 'sudo ./bin/Protonect':

```
Version: 0.1.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [gl | cl | cpu] [<device serial>] [-noviewer] [-help] [-version]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 12 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 001806651047
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Error] [protocol::CommandTransaction] bulk transfer failed: LIBUSB_ERROR_TIMEOUT Operation timed out
```
lsusb give this:

```
Bus 004 Device 003: ID 05ac:12a8 Apple, Inc. 
Bus 004 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 003: ID 045e:02c4 Microsoft Corp. 
Bus 002 Device 002: ID 045e:02d9 Microsoft Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 045e:02d9 Microsoft Corp. 
Bus 001 Device 002: ID 0781:5571 SanDisk Corp. Cruzer Fit
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 003: ID 0408:2fb1 Quanta Computer, Inc. 
Bus 003 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
lspci | grep USB gives:

```
00:1a.0 USB controller: Intel Corporation 6 Series/C200 Series Chipset Family USB Enhanced Host Controller #2 (rev 05)
00:1d.0 USB controller: Intel Corporation 6 Series/C200 Series Chipset Family USB Enhanced Host Controller #1 (rev 05)
04:00.0 USB controller: NEC Corporation uPD720200 USB 3.0 Host Controller (rev 04)
```
This is the uname -a output:
`
Linux ankit-Dell-System-XPS-L502X 3.19.0-47-generic #53~14.04.1-Ubuntu SMP Mon Jan 18 16:09:14 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux`

Also, here the log file when I run LIBUSB_DEBUG=4 ./bin/Protonect
[log.txt](https://github.com/OpenKinect/libfreenect2/files/111324/log.txt)

xlz, 2016-01-31:
Exactly the same as #546. Close as a duplicate.

Also, don't ever use sudo to run Protonect again.

closed 2016-01-31 by xlz

Moheat, 2017-10-10:
How do we download the latest version of libfreenect2?

Moheat, 2017-10-10:
When I try running Protonect inside libfreenect2/build/bin/Protonect I get this error. I don't know how to fix this

Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./build/bin/Protonect [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 7 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:2 with serial 153981134047
[Info] [Freenect2Impl] found 1 devices
Segmentation fault (core dumped)