OpenKinect/libfreenect2 #550: Possible to use libfreenect2 alongside Kinect for Windows v2 SDK?

dbruning, 2016-02-01:
I have a project that needs to work with multiple Kinects, but I only need to "concentrate" (i.e. take the full 3D depth frame) from one Kinect at a time, while the others are basically being used as webcams.

Could I use the Kinect for Windows v2 SDK to communicate with one of the Kinects, while using libfreenect2 to just take the color frame off the others?

I doubt it's possible, because you need to switch between the two drivers at a Windows level, but I'm not very familiar with the libfreenext2 project so thought it worth asking.

Thanks,
Darren.

xlz, 2016-02-02:
It looks like the Microsoft SDK does not support multiple Kinects. Thus, I don't think it would try to capture multiple all the Kinects available, and you would be free to use the other Kinects with libfreenect2.

The method to switch drivers for Kinects is documented in README (Zadig, Device Manager). It is manual only, but it's doable.

dbruning, 2016-02-02:
Great, I'll give that a try when my second Kinect arrives.

Re the issue with ensuring the USB3 ports are on separate PCI buses - do you know whether it's possible to request just the color frame from the Kinect via libfreenect2, thereby reducing the load on the USB3 port? (Since I only need a webcam image from the second Kinect). Or does all the raw information from the camera always have to come in?

xlz, 2016-02-02:
I don't know. Nobody has thought of trying this yet. But I know what to look for.

dbruning, 2016-02-02:
OK, thanks. Hopefully I'll get time to try it out.

xlz, 2016-02-04:
Can you try PR #557? It provides a new API to disable depth.

I think it will save the bandwidth, but have not actually tested with multiple Kinects.

xlz, 2016-02-06:
The ability to disable rgb or depth stream is added.

closed 2016-02-06 by xlz

dbruning, 2016-02-06:
That's awesome - thanks. I have been diverted to another part of the project for now, but I look forward to trying this out when I get back to it!