OpenKinect/libfreenect2 #552: The same pointer (CommandTransaction::Result::data) may be deallocated more than once time

Harpseal, 2016-02-02:
The **data** pointer of the **CommandTransaction::Result** struct doesn't set to NULL after calling **delete[] data;** . 

[src/command_transaction.cpp line:65](https://github.com/OpenKinect/libfreenect2/blob/master/src/command_transaction.cpp#L65)
```
// 
void CommandTransaction::Result::deallocate()
{
  if (data != NULL)
  {
    delete[] data; //issue is here
  }
  length = 0;
  capacity = 0;
}
```
should be =>
```
void CommandTransaction::Result::deallocate()
{
  if (data != NULL)
  {
    delete[] data;
    data = NULL; 
  }
  length = 0;
  capacity = 0;
}
```

Take the following code for example, if the **CommandTransaction::send(command)** return **ResultCode::Error** in method **CommandTransaction::execute**, the **CommandTransaction::Result::deallocate()** will be call twice by Result::notSuccessfulThenDeallocate() and deconstructor. 
[src/command_transaction.cpp line:94](https://github.com/OpenKinect/libfreenect2/blob/master/src/command_transaction.cpp#L94)

```
void CommandTransaction::execute(const CommandBase& command, Result& result)
{
  result.allocate(command.maxResponseLength());

  // send command
  result.code = send(command);

  if(result.notSuccessfulThenDeallocate()) return;
....
```
This issue has be tested using Visual studio 2013 in Windows 10.
Thanks.

xlz, 2016-02-02:
Thanks for bringing this up. This is to be fixed in with other error reporting issues.

xlz, 2016-02-03:
I believe PR #556 solves this bug. If you are interested, take a look at it and leave some reviews.

closed 2016-02-04 by xlz