OpenKinect/libfreenect2 #553: RGB data format

whatisor, 2016-02-03:
Hello,
What is default format of RGB data which is input of the belows:
    int r = tjDecompress2(impl_->decompressor, packet.jpeg_buffer, packet.jpeg_buffer_length, impl_->frame->data, 1920, 1920 * tjPixelSize[TJPF_BGRX], 1080, TJPF_BGRX, 0);

As I understand it is jpeg format.
Why don't we get raw  format?

Thank you

xlz, 2016-02-03:
The format is TJPF_BGRX.

whatisor, 2016-02-03:
TJPF_BGRX is compressed data format?
Thank you

floe, 2016-02-03:
I'm pretty sure you can't get the raw Bayer data with the Windows SDK. The camera sensor directly streams JPEG data, otherwise, the bandwidth consumption on the USB would be far too high.

BGRX is simply 4 bytes per pixel: Blue, Green, Red, Empty (reserved for alpha).

whatisor, 2016-02-04:
Hi floe, I don't think so. Because I see SDK has these options:
https://msdn.microsoft.com/en-us/library/microsoft.kinect.kinect.colorimageformat.aspx
typedef enum _ColorImageFormat
{
    ColorImageFormat_None = 0,
    ColorImageFormat_Rgba = 1,
    ColorImageFormat_Yuv = 2,
    ColorImageFormat_Bgra = 3,
   ** ColorImageFormat_Bayer = 4,**
    ColorImageFormat_Yuy2 = 5
} ColorImageFormat, ColorImageFormat_None, ColorImageFormat_Rgba, ColorImageFormat_Yuv, ColorImageFormat_Bgra, ColorImageFormat_Bayer, ColorImageFormat_Yuy2;

floe, 2016-02-04:
OK, then the Microsoft SDK does conversion in software. Of course, you're free to do that with the BGRX data, too - but I can assure you that it is _not_ the raw RGB sensor data, there will always be a JPEG compression/decompression stage in between.

whatisor, 2016-02-04:
So, do you mean Kinect v2 cannot return raw data as Bayer format?
Kinect 1 can do this.
Maybe I need to check with Windows 8 ( I do not have this :D )
Have you ever checked this option,ColorImageFormat_Bayer = 4 on Window 8?
Thank you

floe, 2016-02-04:
> So, do you mean Kinect v2 cannot return raw data as Bayer format?
Yes.

> Kinect 1 can do this.
True, but the color resolution of Kinect v1 is just VGA (640x480). The Kinect v2 has FullHD (1920x1080). The bandwidth for the raw stream would be far too high, so it's always compressed.

> Have you ever checked this option,ColorImageFormat_Bayer = 4 on Window 8?
No, because I know that the Kinect v2 generates a JPEG stream in hardware.

closed 2016-02-04 by floe