OpenKinect/libfreenect2 #558: 0.2 tracking issue

xlz, 2016-02-05:
Major changes and remaining issues:

##### New features

- [x] OpenNI2 integration (Zou Hanya)
- [x] Depth only registration functions (Matthieu FT)
- [ ] Raw data dumping and replay interfaces/tools (Brendan Burns)
- [ ] Firmware upgrade tool
- [ ] VideoToolbox depth decoding
- [ ] VA-API depth decoding
- [ ] CUDA depth decoding
- [ ] Tegra JPEG decoding
- [ ] Gray output of JPEG decoders
- [ ] SWIG bindings
- [ ] Python binding (pyfreenect2)

##### Improvement and bug fixes

- [x] CPU depth processor cleanup (Albert Hofkamp)
- [ ] Ability to disable RGB or depth stream - WIP
- [ ] Improve USB error checking - WIP
- [ ] Documentation for Jetson TK1

Probably not possible for 0.2:

- [ ] Zerocopy libusb transfer
- [ ] Zerocopy between packet processors and frame listeners

xlz, 2016-02-29:
@floe People are having weird issues with beignet 1.0.3 https://github.com/code-iai/iai_kinect2/issues/236, where Protonect works fine but beignet produces some mysterious errors within iai_kinect2.

Is it possible to provide beignet 1.1.1 in your ppa? 

For the record, I built a ported beignet 1.1.1 to Ubuntu 14.04 with the beignet sources (1.1.1) from Ubuntu 16.04, changed `debian/control` to remove build dependencies on `clang, llvm`, installed `clang-3.6 libclang-3.6-dev`, and `apt-src build beignet-dev` built some beignet 1.1.1 debs.

floe, 2016-03-05:
Sorry for the silence, been sick for a while. I'll see what I can put together, doesn't sound like a big problem for the PPA.

floe, 2016-03-10:
I've created a new PPA: https://launchpad.net/~floe/+archive/ubuntu/beignet
Beignet 1.1.1 is available for trusty:amd64, doesn't seem to build for i386...?

floe, 2016-03-10:
Update: took some fiddling, now also available for i386.

floe, 2016-03-11:
I guess it's not worth to support other non-LTS releases via the PPAs, as 16.04 will be out soon anyway?

sunnysunnyran, 2016-03-22:
somebody who know how to matting from multi kinect?thankyou very much

floe, 2016-03-22:
@sunnysunnyran please open a new issue with a specific question and don't just dump your comments into arbitrary threads.

xlz, 2016-03-25:
rc2

xlz, 2016-04-13:
For anyone interested, what is left is just updating the docs and testing. The code is pretty much finished. 

Nvidia's kayccc is following up with the problem on Jetson TX1, but I don't think he will have an answer any time soon.

I'm busy these two weeks. Hopefully I can make a release by the end of this month.

xlz, 2016-04-28:
I published 0.2 release.

closed 2016-04-28 by xlz

floe, 2016-04-29:
Great work! I'll check with Zenodo if anything needs to be done regarding the DOI.