OpenKinect/libfreenect2 #559: New to Kinect V2, need help

sankethrit2013, 2016-02-05:
Regarding the following piece of code in the "protonect.cpp" example:

/// [main]
/**
 * Main application entry point.
 *
 * Accepted argumemnts:
 * - cpu Perform depth processing with the CPU.
 * - gl  Perform depth processing with OpenGL.
 * - cl  Perform depth processing with OpenCL.
 * - <number> Serial number of the device to open.
 * - -noviewer Disable viewer window.
 */

I understood that i have to give these arguments in the code somewhere, I just couldn't figure out where. There is no variable which can be edited to include these string values.

It would be great if someone could help me with this. As I think this is what is causing nothing to show up when I run my code. The code runs, but nothing shows up, just a cmd console as below:

[image]

xlz, 2016-02-05:
Your libusb is not correctly built with the libusbK backend. If you don't know how you built it wrong, you can simply download the prebuilt binary https://github.com/OpenKinect/libfreenect2/releases/tag/v0.1.0

sankethrit2013, 2016-02-10:
[image]

Same issue, did not solve with the prebuilt libraries. This time around, it does not even run. It just closes.

xlz, 2016-02-10:
Did you use Zadig?

sankethrit2013, 2016-02-10:
Yes. I have the driver installed properly. It shows 'Xbox NUI Sensor
(Composite Parent).

I am assuming it is grabbing the RGB data, but nothing is being displayed.

​
Same as how it worked when I built it.

*Sanketh S. Moudgalya*

*PhD Student*

*Chester F. Carlson Center for Imaging Science*

*Graduate Research and Teaching Assistant*

*Image, Video and Computer Vision Laboratory*

*Department of Electrical and Microelectronic EngineeringRochester
Institute of Technology*

xlz, 2016-02-10:
I'm unable to reproduce your issue. I just downloaded the binary, and it ran correctly.

xlz, 2016-02-15:
I reproduced this issue by building libusb with `install_libusb_vs2015.cmd`. The original winiso branch is broken by upstream merge conflicts and it didn't get built in.

Now I have tested and there is no longer this error.

closed 2016-02-15 by xlz