OpenKinect/libfreenect2 #561: openni2 errors

lakehanne, 2016-02-06:
I got the following error during a build on ubuntu 12.04. Openni2 is installed btw.

```bash
/home/local/ANT/ogunmolu/Documents/libfreenect2/src/openni2/VideoStream.cpp: In member function 'bool Freenect2Driver::VideoStream::buildFrame(libfreenect2::Frame*)':
/home/local/ANT/ogunmolu/Documents/libfreenect2/src/openni2/VideoStream.cpp:113:36: error: 'getServices' was not declared in this scope
/home/local/ANT/ogunmolu/Documents/libfreenect2/src/openni2/VideoStream.cpp:140:25: error: no matching function for call to 'Freenect2Driver::VideoStream::raiseNewFrame(OniFrame*&)'
/home/local/ANT/ogunmolu/Documents/libfreenect2/src/openni2/VideoStream.cpp:140:25: note: candidate is:
/usr/include/openni2/Driver/OniDriverAPI.h:67:7: note: void oni::driver::StreamBase::raiseNewFrame(OniDriverFrame*)
/usr/include/openni2/Driver/OniDriverAPI.h:67:7: note:   no known conversion for argument 1 from 'OniFrame*' to 'OniDriverFrame*'
make[2]: *** [CMakeFiles/freenect2-openni2.dir/src/openni2/VideoStream.cpp.o] Error 1
make[1]: *** [CMakeFiles/freenect2-openni2.dir/all] Error 2
make: *** [all] Error 2
```

xlz, 2016-02-06:
You are using an old version of openni2. What is the version?

lakehanne, 2016-02-07:
I installed  the one from the Ubuntu repositories. Something of the sort: libopenni2-dev.

I am on Ubuntu precise  btw

xlz, 2016-02-07:
libopenni2-dev is not available in the Ubuntu precise repo http://packages.ubuntu.com/search?keywords=libopenni2-dev

lakehanne, 2016-02-07:
Interesting. My company uses a custom build of Ubuntu precise which I believe I must have used. Maybe I should try compiling the libopenni2-dev  from source? Do you have a good reference for this?

xlz, 2016-02-07:
Use the binary here http://structure.io/openni?

closed 2016-02-26 by xlz