OpenKinect/libfreenect2 #564: Segmentation Fault

pedropgusmao, 2016-02-09:
Hi,

I am trying to run install libfreenect2 and run the Protonect executable. However I keep getting Segmentation Fault. Both infrared and xbox logo light up and don turn off after I ry to run the example. During the first try a black window appears for half a second then it closes  with the segmentation fault message.

Here are both outputs of both calling Protonect and dmesg...

Thanks for your help.

```
Version: 0.1.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [gl | cl | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 14 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:8 with serial 038870245147
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 038870245147
device firmware: 4.0.3911.0
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
Segmentation fault (core dumped)
```

```
[332468.676039] usb 1-4: new high-speed USB device number 18 using xhci_hcd
[332468.989284] usb 2-4: new SuperSpeed USB device number 7 using xhci_hcd
[332469.173894] usb 1-4: New USB device found, idVendor=045e, idProduct=02d9
[332469.173898] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[332469.173900] usb 1-4: Product: NuiSensor Adaptor      
[332469.173902] usb 1-4: Manufacturer: Microsoft Corporation  
[332469.174503] hub 1-4:1.0: USB hub found
[332469.174751] hub 1-4:1.0: 1 port detected
[332471.967960] usb 2-4: New USB device found, idVendor=045e, idProduct=02d9
[332471.967963] usb 2-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[332471.967965] usb 2-4: Product: NuiSensor Adaptor      
[332471.967967] usb 2-4: Manufacturer: Microsoft Corporation  
[332471.970057] hub 2-4:1.0: USB hub found
[332471.970207] hub 2-4:1.0: 1 port detected
[332472.436132] usb 2-4.1: new SuperSpeed USB device number 8 using xhci_hcd
[332472.452757] usb 2-4.1: New USB device found, idVendor=045e, idProduct=02c4
[332472.452763] usb 2-4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[332472.452766] usb 2-4.1: Product: Xbox NUI Sensor
[332472.452768] usb 2-4.1: Manufacturer: Microsoft
[332472.452771] usb 2-4.1: SerialNumber: 038870245147
[332494.368248] usb 2-4.1: reset SuperSpeed USB device number 8 using xhci_hcd
[332494.384421] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88085654cc00
[332494.384425] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88085654cc48
[332494.384426] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88085654ccd8
[332494.384427] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88085654cc90
[332495.387063] Protonect[26892]: segfault at 0 ip 00007f4c85583527 sp 00007f4c75808e90 error 4 in libfreenect2.so.0.1.0[7f4c85574000+3b000] ```

floe, 2016-02-09:
Try all 3 depth packet processors by running `Protonect gl`, `Protonect cl` and `Protonect cpu`. Also, run Protonect with gdb and no other parameters and see if you can produce a backtrace for the segfault.

pedropgusmao, 2016-02-09:
Wow, that was quick!
Here are my outputs

output of ```Protonect gl```:

``` 
pedro@skynet:~/repositories/libfreenect2/build/bin$ ./Protonect gl
Version: 0.1.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [gl | cl | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 14 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:13 with serial 038870245147
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Info] [Freenect2DeviceImpl] started
device serial: 038870245147
device firmware: 4.0.3911.0
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
Segmentation fault (core dumped)
```
################################################

Output of ```Protonect cl``` :
```
pedro@skynet:~/repositories/libfreenect2/build/bin$ ./Protonect cl
Version: 0.1.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [gl | cl | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
To pause and unpause: pkill -USR1 Protonect
OpenCL pipeline is not supported!
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 14 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:14 with serial 038870245147
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 1024
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9759
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Info] [Freenect2DeviceImpl] started
device serial: 038870245147
device firmware: 4.0.3911.0
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[... 85 lines cut ...]
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Info] [Freenect2DeviceImpl] started
device serial: 038870245147
device firmware: 4.0.3911.0
[Debug] [DepthPacketStreamParser] not all subsequences received 512
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe7ed5700 (LWP 27912)]
0x00007ffff7ba7527 in execute (this=0x91caf0) at /home/pedro/repositories/libfreenect2/include/internal/libfreenect2/async_packet_processor.h:121
121	        processor_->process(current_packet_);
(gdb) 
```
Here the Viewer window opens but then hangs with no image displayed.

Again thanks for your help.

floe, 2016-02-09:
OK, that's not really supposed to happen :-) Please type `bt` (for backtrace) at the gdb prompt and post the result.

floe, 2016-02-09:
BTW, are you running libusb-1.0.20?

pedropgusmao, 2016-02-09:
Regarding libusb, I belive so, I get the following from ```dpkg``` : 
``` 
pedro@skynet:~/repositories/libfreenect2$ dpkg -s libusb-1.0-0-dev 
Package: libusb-1.0-0-dev
Status: install ok installed
Priority: optional
Section: libdevel
Installed-Size: 292
Maintainer: Aurelien Jarno <aurel32@debian.org>
Architecture: amd64
Multi-Arch: same
Source: libusb-1.0
Version: 2:1.0.20-1ppa1~trusty1
Depends: libusb-1.0-0 (= 2:1.0.20-1ppa1~trusty1)
Recommends: libusb-1.0-doc
Description: userspace USB programming library development files
 Library for programming USB applications without the knowledge
 of Linux kernel internals.
 .
 This package contains what you need for compiling sources that
 use this library in your own code.
Homepage: http://www.linux-usb.org/
``` 

And the output from gdb is: 

```
pedro@skynet:~/repositories/libfreenect2/build$ gdb ./bin/Protonect 
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./bin/Protonect...done.
(gdb) run
Starting program: /home/pedro/repositories/libfreenect2/build/bin/Protonect 
Traceback (most recent call last):
  File "/usr/share/gdb/auto-load/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.19-gdb.py", line 63, in <module>
    from libstdcxx.v6.printers import register_libstdcxx_printers
ImportError: No module named 'libstdcxx'
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Version: 0.1.0
Environment variables: LOGFILE=<protonect.log>
Usage: /home/pedro/repositories/libfreenect2/build/bin/Protonect [gl | cl | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
To pause and unpause: pkill -USR1 Protonect
[New Thread 0x7ffff1769700 (LWP 28336)]
[New Thread 0x7ffff0f68700 (LWP 28337)]
[New Thread 0x7fffe7ed5700 (LWP 28338)]
[New Thread 0x7fffe76d4700 (LWP 28340)]
device serial: 038870245147
device firmware: 4.0.3911.0

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe7ed5700 (LWP 28338)]
0x00007ffff7ba7527 in execute (this=0x8d9ad0) at /home/pedro/repositories/libfreenect2/include/internal/libfreenect2/async_packet_processor.h:121
121	        processor_->process(current_packet_);
(gdb) bt
#0  0x00007ffff7ba7527 in execute (this=0x8d9ad0) at /home/pedro/repositories/libfreenect2/include/internal/libfreenect2/async_packet_processor.h:121
#1  libfreenect2::AsyncPacketProcessor<libfreenect2::RgbPacket>::static_execute (data=0x8d9ad0) at /home/pedro/repositories/libfreenect2/include/internal/libfreenect2/async_packet_processor.h:106
#2  0x00007ffff7bb705f in tthread::thread::wrapper_function (aArg=0x7ad3f0) at /home/pedro/repositories/libfreenect2/src/tinythread/tinythread.cpp:167
#3  0x00007ffff6948182 in start_thread (arg=0x7fffe7ed5700) at pthread_create.c:312
#4  0x00007ffff6e7047d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
(gdb) 
```

xlz, 2016-02-09:
`bt full`? What is the address of `processor_` or `current_packet_`?

pedropgusmao, 2016-02-09:
There you have it:
```
pedro@skynet:~/repositories/libfreenect2/build$ gdb ./bin/Protonect 
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./bin/Protonect...done.
(gdb) run
Starting program: /home/pedro/repositories/libfreenect2/build/bin/Protonect 
Traceback (most recent call last):
  File "/usr/share/gdb/auto-load/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.19-gdb.py", line 63, in <module>
    from libstdcxx.v6.printers import register_libstdcxx_printers
ImportError: No module named 'libstdcxx'
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Version: 0.1.0
Environment variables: LOGFILE=<protonect.log>
Usage: /home/pedro/repositories/libfreenect2/build/bin/Protonect [gl | cl | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
To pause and unpause: pkill -USR1 Protonect
[New Thread 0x7ffff1769700 (LWP 29280)]
[New Thread 0x7ffff0f68700 (LWP 29281)]
[New Thread 0x7fffe7ed5700 (LWP 29282)]
[New Thread 0x7fffe76d4700 (LWP 29283)]

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe7ed5700 (LWP 29282)]
0x00007ffff7ba7527 in execute (this=0x91e700) at /home/pedro/repositories/libfreenect2/include/internal/libfreenect2/async_packet_processor.h:121
121	        processor_->process(current_packet_);
(gdb) bt full
#0  0x00007ffff7ba7527 in execute (this=0x91e700) at /home/pedro/repositories/libfreenect2/include/internal/libfreenect2/async_packet_processor.h:121
No locals.
#1  libfreenect2::AsyncPacketProcessor<libfreenect2::RgbPacket>::static_execute (data=0x91e700) at /home/pedro/repositories/libfreenect2/include/internal/libfreenect2/async_packet_processor.h:106
No locals.
#2  0x00007ffff7bb705f in tthread::thread::wrapper_function (aArg=0x7ad3f0) at /home/pedro/repositories/libfreenect2/src/tinythread/tinythread.cpp:167
        ti = 0x7ad3f0
        guard = <optimized out>
#3  0x00007ffff6948182 in start_thread (arg=0x7fffe7ed5700) at pthread_create.c:312
        __res = <optimized out>
        pd = 0x7fffe7ed5700
        now = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140737084479232, 1112596798112836517, 1, 0, 140737084479936, 140737084479232, -1112543871805379675, -1112580639286449243}, mask_was_saved = 0}}, 
          priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
        pagesize_m1 = <optimized out>
        sp = <optimized out>
        freesize = <optimized out>
        __PRETTY_FUNCTION__ = "start_thread"
#4  0x00007ffff6e7047d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
No locals.
(gdb) p &processor_ 
$1 = (libfreenect2::AsyncPacketProcessor<libfreenect2::RgbPacket>::PacketProcessorPtr *) 0x91e708
(gdb) p &current_packet_
$2 = (libfreenect2::RgbPacket *) 0x91e718
(gdb) 
```

xlz, 2016-02-09:
`p processor_` and `p current_packet_`. They are both pointers. I just need their values (not the addresses of themselves.)

pedropgusmao, 2016-02-09:
Sorry for that, here is the corrected log.  I guess there is something wrong with ```processor_ ``` ...

```
pedro@skynet:~/repositories/libfreenect2/build$ gdb ./bin/Protonect 
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./bin/Protonect...done.
(gdb) run
Starting program: /home/pedro/repositories/libfreenect2/build/bin/Protonect 
Traceback (most recent call last):
  File "/usr/share/gdb/auto-load/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.19-gdb.py", line 63, in <module>
    from libstdcxx.v6.printers import register_libstdcxx_printers
ImportError: No module named 'libstdcxx'
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Version: 0.1.0
Environment variables: LOGFILE=<protonect.log>
Usage: /home/pedro/repositories/libfreenect2/build/bin/Protonect [gl | cl | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
To pause and unpause: pkill -USR1 Protonect
[New Thread 0x7ffff1769700 (LWP 29848)]
[New Thread 0x7ffff0f68700 (LWP 29849)]
[New Thread 0x7fffe7ed5700 (LWP 29851)]
[New Thread 0x7fffe76d4700 (LWP 29852)]
device serial: 038870245147
device firmware: 4.0.3911.0

Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe7ed5700 (LWP 29851)]
0x00007ffff7ba7527 in execute (this=0x91e700) at /home/pedro/repositories/libfreenect2/include/internal/libfreenect2/async_packet_processor.h:121
121	        processor_->process(current_packet_);
(gdb) bt full
#0  0x00007ffff7ba7527 in execute (this=0x91e700) at /home/pedro/repositories/libfreenect2/include/internal/libfreenect2/async_packet_processor.h:121
No locals.
#1  libfreenect2::AsyncPacketProcessor<libfreenect2::RgbPacket>::static_execute (data=0x91e700) at /home/pedro/repositories/libfreenect2/include/internal/libfreenect2/async_packet_processor.h:106
No locals.
#2  0x00007ffff7bb705f in tthread::thread::wrapper_function (aArg=0x7ad3f0) at /home/pedro/repositories/libfreenect2/src/tinythread/tinythread.cpp:167
        ti = 0x7ad3f0
        guard = <optimized out>
#3  0x00007ffff6948182 in start_thread (arg=0x7fffe7ed5700) at pthread_create.c:312
        __res = <optimized out>
        pd = 0x7fffe7ed5700
        now = <optimized out>
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {140737084479232, -2912808726539726445, 1, 0, 140737084479936, 140737084479232, 2912791293840134547, 2912829281000952211}, mask_was_saved = 0}}, priv = {
            pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
        pagesize_m1 = <optimized out>
        sp = <optimized out>
        freesize = <optimized out>
        __PRETTY_FUNCTION__ = "start_thread"
#4  0x00007ffff6e7047d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
No locals.
(gdb) p processor_
$1 = (libfreenect2::AsyncPacketProcessor<libfreenect2::RgbPacket>::PacketProcessorPtr) 0x0
(gdb) p current_packet_ 
$2 = {sequence = 10, timestamp = 20195, jpeg_buffer = 0x7fffe8488018 "\377\330\377", <incomplete sequence \342>, jpeg_buffer_length = 622767, exposure = 16,6820621, gain = 1,00072396, gamma = 1}
(gdb) 
```

mcgi5sr2, 2016-02-09:
Hi, 
I have come across what I believe to be the exact same issue using micro pcs with only an on board intel graphics card. We found a solution which involved rolling back this repo to 
9f39b744a8fe89d4d86e38bb2c494303ab41ae86

xlz, 2016-02-09:
Can you show the content of `build/libfreenect2/config.h`?

Can you try to `rm -r *` under the build directory, and rebuild?

pedropgusmao, 2016-02-09:
I have rebuilt as suggested and I still get the same error. 
Also, I don't know if it changes anything, but I have not install OpenNI2 since it was optional...
 My config.h is: 
```
/*
* This file is part of the OpenKinect Project. http://www.openkinect.org
*
* Copyright (c) 2014 individual OpenKinect contributors. See the CONTRIB file
* for details.
*
* This code is licensed to you under the terms of the Apache License, version
* 2.0, or, at your option, the terms of the GNU General Public License,
* version 2.0. See the APACHE20 and GPL2 files for the text of the licenses,
* or the following URLs:
* http://www.apache.org/licenses/LICENSE-2.0
* http://www.gnu.org/licenses/gpl-2.0.txt
*
* If you redistribute this file in source form, modified or unmodified, you
* may:
*   1) Leave this header intact and distribute it under the same terms,
*      accompanying it with the APACHE20 and GPL20 files, or
*   2) Delete the Apache 2.0 clause and accompany it with the GPL2 file, or
*   3) Delete the GPL v2 clause and accompany it with the APACHE20 file
* In all cases you must keep the copyright notice intact and include a copy
* of the CONTRIB file.
*
* Binary distributions must follow the binary distribution requirements of
* either License.
*/

#ifndef LIBFREENECT2_CONFIG_H
#define LIBFREENECT2_CONFIG_H

#define LIBFREENECT2_VERSION "0.1.0"
#define LIBFREENECT2_API_VERSION ((0 << 16) | 1)

#ifdef _MSC_VER
#define LIBFREENECT2_PACK( __Declaration__ ) __pragma( pack(push, 1) ) __Declaration__ __pragma( pack(pop) )
#else
#define LIBFREENECT2_PACK( __Declaration__ ) __Declaration__ __attribute__((__packed__))
#endif

#include <libfreenect2/export.h>
#define LIBFREENECT2_API LIBFREENECT2_EXPORT

#define LIBFREENECT2_WITH_OPENGL_SUPPORT

/* #undef LIBFREENECT2_WITH_OPENCL_SUPPORT */
/* #undef LIBFREENECT2_OPENCL_ICD_LOADER_IS_OLD */

/* #undef LIBFREENECT2_WITH_VT_SUPPORT */

/* #undef LIBFREENECT2_WITH_TURBOJPEG_SUPPORT */

/* #undef LIBFREENECT2_THREADING_STDLIB */

#define LIBFREENECT2_THREADING_TINYTHREAD

/* #undef LIBFREENECT2_WITH_CXX11_SUPPORT */

#endif // LIBFREENECT2_CONFIG_H
```

haxan7, 2016-02-09:
I too am facing this exact issue on my [HP 15-j137tx](http://support.hp.com/us-en/product/HP-ENVY-15-Notebook-PC-series/5401191/model/6754027/document/c04074949/) running Ubuntu 14.04 kernel 3.13, with all the dependencies installed with versions as suggested by the README.md.

mcgi5sr2, 2016-02-09:
@haxan7 and @pedropgusmao have you tried revert back to the commit I suggested?

haxan7, 2016-02-09:
@mcgi5sr2 I don't have access to a kinect at the moment. I will try it tomorrow.

matlabbe, 2016-02-09:
I got the seg fault when creating the Registration object:
```cpp
//default registration params
libfreenect2::Freenect2Device::IrCameraParams depthParams = dev_->getIrCameraParams();
libfreenect2::Freenect2Device::ColorCameraParams colorParams = dev_->getColorCameraParams();
reg_ = new libfreenect2::Registration(depthParams, colorParams); //<-- seg fault here
```
I think this may be related as the commit suggested by @mcgi5sr2 is referring to changes in Registration class.

cheers

xlz, 2016-02-09:
Somehow turbojpeg is not detected/ enabled in your build...

xlz, 2016-02-09:
@fran6co what was your test results wrt making turbojpeg optional on Mac?

matlabbe, 2016-02-09:
My system for info: Ubuntu 14.04
I tried with version 9f39b74 and it worked. I switched back to latest code revision (4d03b71) just to check if turbojpeg is detected and it is working now! Well, I didn't clear the CMake cache when I've updated the first time. Here is the cmake output (turbojpeg is detected):
```
~/workspace/libfreenect2/build$ cmake ..
-- using tinythread as threading library
-- Could NOT find OpenCL (missing:  OpenCL_LIBRARY OpenCL_INCLUDE_DIR) 
-- RPATH set to 
-- Linking with these libraries: 
 /home/mathieu/workspace/libfreenect2/depends/libusb/lib/libusb-1.0.so
 pthread
 /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0
 /home/mathieu/workspace/libfreenect2/depends/glfw/lib/libglfw.so
 /usr/lib/x86_64-linux-gnu/libGL.so
-- Configurating examples
-- Using in-tree freenect2 target
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mathieu/workspace/libfreenect2/build
```

xlz, 2016-02-09:
The root cause: our CMakeLists.txt uses `TurboJPEG_FOUND` to detect presence of TurboJPEG. CMake 3.2 and earlier only provide `TURBOJPEG_FOUND`. CMake 3.3+ provide both `TurboJPEG_FOUND` and `TURBOJPEG_FOUND`.

Without enabling TurboJPEG by default, the default rgb processor is NULL, which is an ill designed runtime check.

xlz, 2016-02-09:
Proposed fixes in #566.

xlz, 2016-02-09:
To improve test coverage of future similar issues, CMake related changes must be tested using different versions of CMake.

xlz, 2016-02-10:
Any objections? I will merge #566 if no other opinions.

pedropgusmao, 2016-02-10:
@xlz and  @mcgi5sr2 , I rolled back to https://github.com/OpenKinect/libfreenect2/commit/9f39b744a8fe89d4d86e38bb2c494303ab41ae86 and everything is working now. 
Thanks.

pedropgusmao, 2016-02-10:
Also installed everything from the beginning (deleted previous library, cloned repository) and applied changes from https://github.com/OpenKinect/libfreenect2/issues/566 . It also worked flawlessly.

haxan7, 2016-02-10:
Both  #566 and 9f39b744a8fe89d4d86e38bb2c494303ab41ae86 working for me too. I also had to update my kernel from 3.13.0-24 to 3.13.0-77.

xlz, 2016-02-10:
Rolling back is avoiding the problem, not fixing the problem.

closed 2016-02-10 by xlz

mcgi5sr2, 2016-02-10:
@xlz totally agree, just thought mentioning last working version we had would help in the bug squash

PR #566 (2016-02-10): Fix CMake's UPPERCASE_FOUND package detection and resulting segfaults
Fix #564.

@fran6co