OpenKinect/libfreenect2 #565: Protonect error show failed to claim interface with IrInterfaceId

chenchengshao, 2016-02-09:
Ubuntu Indigo system, trying to install libfreenect2, after installing following the instruction, can't open the kinect 2, attach the error msg and dmesg message for reference,  need help to see where is the problem?

chencheng@OpNeAR3:~/catkin_ws/src/libfreenect2/build/bin$ export LIBUSB_DEBUG=3
chencheng@OpNeAR3:~/catkin_ws/src/libfreenect2/build/bin$ sudo ./Protonect
modprobe: FATAL: Module nvidia not found.
Version: 0.1.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [gl | cl | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:58 with serial 501185143042
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Error] [protocol::UsbControl] failed to claim interface with IrInterfaceId(=1)! LIBUSB_ERROR_BUSY Resource busy. Try debugging with environment variable: export LIBUSB_DEBUG=3 .
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Error] [Freenect2Impl] failed to open Kinect v2: @2:58
failure opening device!
chencheng@OpNeAR3:~/catkin_ws/src/libfreenect2/build/bin$ 

Error on dmesg

[1527170.552786] usb 2-1.3.1: unable to get BOS descriptor
[1527171.066551] retire_capture_urb: 33 callbacks suppressed
[1527171.246576] usb 2-1.3.1: USB disconnect, device number 57
[1527172.851056] usb 2-1.3.1: new high-speed USB device number 58 using ehci-pci
[1527172.956188] usb 2-1.3.1: unable to get BOS descriptor
[1527172.957188] usb 2-1.3.1: New USB device found, idVendor=045e, idProduct=02d8
[1527172.957192] usb 2-1.3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[1527172.957195] usb 2-1.3.1: Product: Xbox NUI Sensor
[1527172.957198] usb 2-1.3.1: Manufacturer: Microsoft
[1527172.957201] usb 2-1.3.1: SerialNumber: 501185143042
[1527184.571799] usb 2-1.3.1: reset high-speed USB device number 58 using ehci-pci
[1527184.676965] usb 2-1.3.1: unable to get BOS descriptor
[1527185.190981] retire_capture_urb: 170 callbacks suppressed
[1527185.246782] usb 2-1.3.1: USB disconnect, device number 58
[1527186.975263] usb 2-1.3.1: new full-speed USB device number 59 using ehci-pci
[1527187.059392] usb 2-1.3.1: device descriptor read/64, error -32
[1527187.247658] usb 2-1.3.1: device descriptor read/64, error -32
[1527187.435921] usb 2-1.3.1: new full-speed USB device number 60 using ehci-pci
[1527187.520013] usb 2-1.3.1: device descriptor read/64, error -32
[1527187.708318] usb 2-1.3.1: device descriptor read/64, error -32
[1527187.896588] usb 2-1.3.1: new full-speed USB device number 61 using ehci-pci
[1527188.305171] usb 2-1.3.1: device not accepting address 61, error -32
[1527188.389290] usb 2-1.3.1: new full-speed USB device number 62 using ehci-pci
[1527188.797859] usb 2-1.3.1: device not accepting address 62, error -32
[1527188.798071] hub 2-1.3:1.0: unable to enumerate USB device on port 1
[1527190.356106] usb 2-1.3.1: new full-speed USB device number 63 using ehci-pci
[1527190.440226] usb 2-1.3.1: device descriptor read/64, error -32
[1527190.628460] usb 2-1.3.1: device descriptor read/64, error -32
[1527190.816762] usb 2-1.3.1: new full-speed USB device number 64 using ehci-pci
[1527190.900888] usb 2-1.3.1: device descriptor read/64, error -32
[1527191.089156] usb 2-1.3.1: device descriptor read/64, error -32
[1527191.277426] usb 2-1.3.1: new full-speed USB device number 65 using ehci-pci
[1527191.686013] usb 2-1.3.1: device not accepting address 65, error -32
[1527191.770096] usb 2-1.3.1: new full-speed USB device number 66 using ehci-pci
[1527192.178713] usb 2-1.3.1: device not accepting address 66, error -32
[1527192.178997] hub 2-1.3:1.0: unable to enumerate USB device on port 1
[1527193.208189] usb 2-1.3.1: new high-speed USB device number 67 using ehci-pci
[1527193.313322] usb 2-1.3.1: unable to get BOS descriptor
[1527193.314321] usb 2-1.3.1: New USB device found, idVendor=045e, idProduct=02d8
[1527193.314325] usb 2-1.3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[1527193.314329] usb 2-1.3.1: Product: Xbox NUI Sensor
[1527193.314331] usb 2-1.3.1: Manufacturer: Microsoft
[1527193.314347] usb 2-1.3.1: SerialNumber: 501185143042

xlz, 2016-02-09:
Looks like the Kinect is connected to a Usb 2.0 port?

chenchengshao, 2016-02-09:
Thank you, you are right about it, after I changed Usb 3.0, have other problem.

Starting program: /home/chencheng/catkin_ws/src/libfreenect2/build/bin/Protonect 
Traceback (most recent call last):
  File "/usr/share/gdb/auto-load/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.19-gdb.py", line 63, in <module>
    from libstdcxx.v6.printers import register_libstdcxx_printers
ImportError: No module named 'libstdcxx'
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
modprobe: ERROR: could not insert 'fglrx': No such device
Error: Fail to load fglrx kernel module!
Error! Fail to load fglrx kernel module! Maybe you can switch to root user to load kernel module directly
modprobe: ERROR: could not insert 'fglrx': No such device
Error: Fail to load fglrx kernel module!
Error! Fail to load fglrx kernel module! Maybe you can switch to root user to load kernel module directly
modprobe: ERROR: could not insert 'fglrx': No such device
Error: Fail to load fglrx kernel module!
Error! Fail to load fglrx kernel module! Maybe you can switch to root user to load kernel module directly
Version: 0.1.0
Environment variables: LOGFILE=<protonect.log>
Usage: /home/chencheng/catkin_ws/src/libfreenect2/build/bin/Protonect [gl | cl | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
To pause and unpause: pkill -USR1 Protonect
[New Thread 0x7fffec39d700 (LWP 4997)]
[New Thread 0x7fffebb9c700 (LWP 4998)]
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 12 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:3 with serial 501185143042
[Info] [Freenect2Impl] found 1 devices
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65544 GLX: Failed to create context: BadAlloc (insufficient resources for operation)
[Error] [OpenGLDepthPacketProcessor] Failed to create opengl window.
[Thread 0x7fffebb9c700 (LWP 4998) exited]
[Thread 0x7fffec39d700 (LWP 4997) exited]
[Inferior 1 (process 4986) exited with code 0377]
**
# I reinstall the driver , and OpenGL information is :

**
chencheng@opnear1:~$ glxinfo | grep OpenGL
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: GeForce GT 430/PCIe/SSE2
OpenGL version string: 2.1.2 NVIDIA 304.131
OpenGL shading language version string: 1.20 NVIDIA via Cg compiler
OpenGL extensions:

xlz, 2016-02-09:
The OpenGL version is surely too low. See the README:

>The following minimum requirements must be met in order to enable the optional features:
> OpenGL depth processing: OpenGL 3.1

chenchengshao, 2016-02-10:
Thank you, but after re-install the NVIDIA driver and upgrade the kernel from 3.13 to 3.19.

The result is still the same, 

> modprobe: ERROR: could not insert 'fglrx': No such device
> Error: Fail to load fglrx kernel module!

and after reporting 

> status 0x090000:9729, status 0x090000:9731,

 the system crashed, I have to manually reboot the system.

Below is the other system information.

> 
> $uname -r   ->  3.19.0-49-generic
> 
> chencheng@opnear1:~$ glxinfo | grep OpenGL
> OpenGL vendor string: NVIDIA Corporation
> OpenGL renderer string: GeForce GT 430/PCIe/SSE2
> OpenGL core profile version string: 4.3.0 NVIDIA 352.63
> OpenGL core profile shading language version string: 4.30 NVIDIA via Cg compiler
> OpenGL core profile context flags: (none)
> OpenGL core profile profile mask: core profile
> OpenGL core profile extensions:
> OpenGL version string: 4.5.0 NVIDIA 352.63
> OpenGL shading language version string: 4.50 NVIDIA
> OpenGL context flags: (none)
> OpenGL profile mask: (none)
> OpenGL extensions

floe, 2016-02-10:
1. Don't run Protonect with sudo. Use platform/linux/udev/90-kinect2.rules, if necessary.
2. Does `Protonect cpu` work?

xlz, 2016-02-10:
> the system crashed

How did it crash? Need more information.

chenchengshao, 2016-02-10:
Protonect cpu also no work, it also crashed.

I tried to run in the command line gdb Protonect, here is the output:
> 
> Starting program: /home/chencheng/catkin_ws/src/libfreenect2/build/bin/Protonect 
> Traceback (most recent call last):
>   File "/usr/share/gdb/auto-load/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.19-gdb.py", line 63, in <module>
>     from libstdcxx.v6.printers import register_libstdcxx_printers
> ImportError: No module named 'libstdcxx'
> [Thread debugging using libthread_db enabled]
> Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
> [New Thread 0x7fffec39d700 (LWP 3124)]
> [New Thread 0x7fffebb9c700 (LWP 3125)]
> [Thread 0x7fffebb9c700 (LWP 3125) exited]
> [Thread 0x7ffff7fbc7c0 (LWP 3113) exited]
> [Inferior 1 (process 3113) exited with code 0377]
> ~                                                    

more information:
> chencheng@opnear1:~/catkin_ws/src/libfreenect2/build/bin$ ldd ./Protonect | grep libusb
> 	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f38123e3000)
> 
>
dmesg error:

> 
> [    3.919640] NVRM: Your system is not currently configured to drive a VGA console
> [    3.919642] NVRM: on the primary VGA device. The NVIDIA Linux graphics driver
> [    3.919643] NVRM: requires the use of a text-mode VGA console. Use of other console
> [    3.919644] NVRM: drivers including, but not limited to, vesafb, may result in
> [    3.919644] NVRM: corruption and stability problems, and is not supported.
> [    4.167893] init: plymouth-upstart-bridge main process ended, respawning
> [    4.170361] init: plymouth-upstart-bridge main process (1296) terminated with status 1
> [    4.170368] init: plymouth-upstart-bridge main process ended, respawning
> [    6.070104] e1000e: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
> [    6.070137] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
> [   32.949224] audit_printk_skb: 171 callbacks suppressed
> [   32.949226] audit: type=1400 audit(1455121835.843:68): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/lib/cups/backend/cups-pdf" pid=2675 comm="apparmor_parser"
> [   32.949231] audit: type=1400 audit(1455121835.843:69): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/sbin/cupsd" pid=2675 comm="apparmor_parser"
> [   32.949459] audit: type=1400 audit(1455121835.843:70): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/sbin/cupsd" pid=2675 comm="apparmor_parser"
> [  390.073531] AMD IOMMUv2 driver by Joerg Roedel <joerg.roedel@amd.com>
> [  390.073537] AMD IOMMUv2 functionality not available on this system
> [  390.140206] <6>[fglrx] Maximum main memory to use for locked dma buffers: 31461 MBytes.
> [  390.141791] <3>[fglrx:firegl_init_device_list] *ERROR* No supported display adapters were found
> [  390.141794] <3>[fglrx:firegl_init_module] *ERROR* firegl_init_devices failed
> [  390.224279] <6>[fglrx] Maximum main memory to use for locked dma buffers: 31461 MBytes.
> [  390.225842] <3>[fglrx:firegl_init_device_list] *ERROR* No supported display adapters were found
> [  390.225845] <3>[fglrx:firegl_init_module] *ERROR* firegl_init_devices failed
> [  390.280232] <6>[fglrx] Maximum main memory to use for locked dma buffers: 31461 MBytes.
> [  390.281493] <3>[fglrx:firegl_init_device_list] *ERROR* No supported display adapters were found
> [  390.281496] <3>[fglrx:firegl_init_module] *ERROR* firegl_init_devices failed

xlz, 2016-02-10:
What crashed? How did it crash? Please describe it precisely. Did your computer explode? Did you operating system panic?

Your gdb output has nothing of relevance. Provide a stack trace at least.

Your dmesg does not contain any messages that happened during Protonect.

chenchengshao, 2016-02-10:
After running ulimit -c unlimint,  ./Protonect, I didn't see any core file in the folder.
The screen will freeze and the system is stuck,then I have to manually reboot the system, below is the link for the screen shot.
https://goo.gl/photos/Syxzvsi5tRfXhXRA8.
And I also tried to run gdb Protonect, hit run in the gdb command line, it's the same result.
I'm new to this one, if I miss any steps?

xlz, 2016-02-10:
I think it's a problem in your graphics driver. Nvidia graphics driver is a giant mess. I can't help you.

xlz, 2016-02-10:
I happen to get hold of a dual Intel/Nvidia graphics laptop. My suggestion: disable or blacklist nouveau, use Intel graphics for OpenGL/OpenCL.

chenchengshao, 2016-02-11:
Thank you for helping, I'm going to try another server with AMD graphic.

closed 2016-02-11 by chenchengshao