OpenKinect/libfreenect2 #567: I run Protonect and output ends with device firmware: xxxxx, occasionally window flashes but closes

hyderjake, 2016-02-09:
Usage: ./bin/Protonect [gl | cl | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 12 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 143374634247
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9791
[Debug] [Freenect2DeviceImpl] status 0x090000: 9791
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 143374634247
device firmware: 4.0.3912.0
using: optirun ./bin/Protonect gl and cl with same error

hyderjake, 2016-02-09:
Also outputs not all subsequences received 0, but not every time

hyderjake, 2016-02-09:
This has worked before but I needed to have a clean install of ubuntu and now it does not run

hyderjake, 2016-02-09:
Kubuntu 14.04 w/ Nvidia 970m driver 355, running through bumblebee

floe, 2016-02-10:
Kernel version?

hyderjake, 2016-02-10:
Linux Ghost 3.19.0-49-generic #55~14.04.1-Ubuntu SMP Fri Jan 22 11:24:31 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

xlz, 2016-02-10:
> with same error

I don't see any error in your report.

hyderjake, 2016-02-10:
It is runtime I believe, the program ends unexpectedly, no window the rgb or ir appears, but if it does appear it closes immediately. the gdb displayed a segmentation fault, but without gdb, I occasionally get errors with the rgb parser package or a seg fault, they aren't consistent

sammell, 2016-02-10:
l have the same error :
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [gl | cl | cpu] [<device serial>]
[-noviewer] [-norgb | -nodepth] [-help] [-version]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 12 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 017503551547
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9757
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 017503551547
device firmware: 4.0.3911.0
[Debug] [DepthPacketStreamParser] not all subsequences received 0
Segmentation fault (core dumped)

xlz, 2016-02-10:
For segfault, it may be duplicate to #564.

Update to latest master and retry.

sammell, 2016-02-10:
Thank you but l can't find solution in #564 can you please explain for me 
and for the second error   
[Debug] [DepthPacketStreamParser] not all subsequences received 0
or [Debug] [DepthPacketStreamParser] not all subsequences received 1020 
the value change every time i execue ./Protonect

xlz, 2016-02-10:
These are not error. You see "[Debug]"? That's an indicator for debug messages.

sammell, 2016-02-10:
then what that's mean :(not all subsequences received 0)

xlz, 2016-02-10:
Some temporary minor packet loss

sammell, 2016-02-10:
ok that's good 
i have ubuntu 14.04 and kinect V2 and i used libfreenect2 (OpenKinect/libfreenect2) but i found problems with openni2 so i installed from structure.io/openi : OpenNi2.2.0.33 Beta (x86)  and when i run NiViewer it gives me openDevice failed:
DeviceOpen using default: no devices found
  but  when i run Protonect it give me 
Environment variables: LOGFILE=<protonect.log>
Usage: ./Protonect [gl | cl | cpu] [<device serial>]
[-noviewer] [-norgb | -nodepth] [-help] [-version]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 12 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:3 with serial 017503551547
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9757
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 017503551547
device firmware: 4.0.3911.0
[Debug] [DepthPacketStreamParser] not all subsequences received 0
Segmentation fault (core dumped)
i can't found a solution can you give me some solution please

xlz, 2016-02-10:
> Update to latest master and retry.

Have you done it

sammell, 2016-02-10:
No , l did not update to latest master . because i don't found install.sh in libfreenect2-master

xlz, 2016-02-10:
That is the solution. And I have given you a solution. That's all.

hyderjake, 2016-02-10:
Thank You, I have it working now

closed 2016-02-10 by xlz