OpenKinect/libfreenect2 #568: [ubuntu 14.04]Run Protonect, after  around 30s , it return LIBUSB_ERROR_NO_DEVICE and sytem stuck.

chenchengshao, 2016-02-11:
After run Protonect for about 30s, return fault LIBUSB_ERROR_NO_DEVICE,and system stuck, need to manually reboot it.
Below is the log file for .Protonect.
[log.txt](https://github.com/OpenKinect/libfreenect2/files/125849/log.txt)

**During its running, keep monitor dmesg, below is the log.**
[dmesg.txt](https://github.com/OpenKinect/libfreenect2/files/125850/dmesg.txt)

[  180.812253] xhci_hcd 0000:07:00.0: WARN Cannot submit Set TR Deq Ptr
[  180.812261] xhci_hcd 0000:07:00.0: A Set TR Deq Ptr command is pending.
[  180.876120] xhci_hcd 0000:07:00.0: WARN Cannot submit Set TR Deq Ptr
[  180.876127] xhci_hcd 0000:07:00.0: A Set TR Deq Ptr command is pending.
[  189.991734] xhci_hcd 0000:07:00.0: xHCI host not responding to stop endpoint command.
[  189.991742] xhci_hcd 0000:07:00.0: Assuming host is dying, halting host.

**Is this problem related to USB controller: ASMedia or something else?**

**system information:**

> chencheng@OpNeAR4:~/catkin_ws/src/libfreenect2/build/bin$ uname -r
> 3.19.0-49-generic

> chencheng@OpNeAR4:~/workspace$ lspci |grep USB 
> 00:1a.0 USB controller: Intel Corporation C600/X79 series chipset USB2 Enhanced Host Controller #2 (rev 06) 
> 00:1d.0 USB controller: Intel Corporation C600/X79 series chipset USB2 Enhanced Host Controller #1 (rev 06) 
_> 06:00.0 USB controller: ASMedia Technology Inc. Device 1142 
> 07:00.0 USB controller: ASMedia Technology Inc. Device 1142 _
> 
> 
> chencheng@OpNeAR4:~/workspace$ lsusb 
> Bus 006 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub 
> Bus 006 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub 
> Bus 004 Device 003: ID 045e:02d8 Microsoft Corp. 
> Bus 004 Device 002: ID 045e:02d9 Microsoft Corp. 
> Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub 
> Bus 003 Device 003: ID 0d3d:0040 Tangtop Technology Co., Ltd 
> Bus 003 Device 002: ID 045e:02d9 Microsoft Corp. 
> Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub 
> Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub 
> Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub 
> Bus 005 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub 
> Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub 
> chencheng@OpNeAR4:~/workspace$ 
> 
> 
> chencheng@OpNeAR4:~/catkin_ws/src/libfreenect2/build/bin$ ldd Protonect |grep usb 
> 	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f35e6100000) 
> chencheng@OpNeAR4:~/catkin_ws/src/libfreenect2/build/bin$ 
> 
> 
> chencheng@OpNeAR4:~/catkin_ws/src/libfreenect2/build/bin$ glxinfo | grep OpenGL
> OpenGL vendor string: Advanced Micro Devices, Inc.
> OpenGL renderer string: AMD Radeon HD 6450
> OpenGL core profile version string: 4.3.13374 Core Profile Context 15.20.1013
> OpenGL core profile shading language version string: 4.40
> OpenGL core profile context flags: (none)
> OpenGL core profile profile mask: core profile
> OpenGL core profile extensions:
> OpenGL version string: 4.4.13374 Compatibility Profile Context 15.20.1013
> OpenGL shading language version string: 4.40
> OpenGL context flags: (none)
> OpenGL profile mask: compatibility profile
> OpenGL extensions:
> chencheng@OpNeAR4:~/catkin_ws/src/libfreenect2/build/b

xlz, 2016-02-11:
Asmedia 1142 is known to not work. https://github.com/OpenKinect/libfreenect2#im-getting-lots-of-usb-transfer-errors-andor-only-blank-windows

chenchengshao, 2016-02-11:
Thank you for confirmation.

closed 2016-02-11 by chenchengshao