OpenKinect/libfreenect2 #569: Depth Registration Question

soulslicer, 2016-02-11:
When using your API to get the Depth image. Is there any upscaling or preprocessing done on the image internally?

xlz, 2016-02-11:
There is no upscaling by default. You can get upsamping one if you want. There is some median filtering and edge removal filtering.

soulslicer, 2016-02-11:
Can I know the API to get this? Also, can I know which part of the code
does these steps?

xlz, 2016-02-11:
Get what?

soulslicer, 2016-02-11:
Essentially I'm trying to rewrite a point cloud generating software using OpenCV but on a Windows machine. I've taken the raw point cloud, and created a registered point cloud using projection matrices etc.

The Kinect sensor Windows API gives a 512x424 pixel image. However, the point cloud I generate is nowhere near is flat/dense as the one that is generated by the IAI_Kinect driver code which uses libfreenect2. Their code looks very basic, and I'm trying to find out why the points generated from the depth image gotten from libfreenect2 seem superior.

It is clear that libfreenect2 is doing some form of upsampling. I'd like to replicate this on my Windows end. I've attempted to look through the code and have had trouble finding the section that does this. I would like to know exactly what upsampling techniques are used, whether OpenCV libraries were used to do these etc.?

xlz, 2016-02-11:
iai_kinect2 does upsampling. Upsampling does not introduce more information.

Use bigdepth in https://openkinect.github.io/libfreenect2/classlibfreenect2_1_1Registration.html#a8be3a548ff440e30f5e89a4b053e04fa to get similar upsampled results.

soulslicer, 2016-02-11:
Okay..but what algorithms are used in bigdepth step?

xlz, 2016-02-11:
A 5x3 dilation filter.

soulslicer, 2016-02-11:
Okay thanks. So essentially that just copies the points over a neighbouring area. Is there any median/mean filtering done?

soulslicer, 2016-02-11:
Lastly..what is your observation of depth maps generated from your system vs that of the Windows API?

xlz, 2016-02-11:
Bilateral filtering is applied in depth processing https://openkinect.github.io/libfreenect2/structlibfreenect2_1_1Freenect2Device_1_1Config.html

I don't have any observation of the Microsoft SDK.

soulslicer, 2016-02-11:
I'm familiar with Bilateral filtering but not have Edge Aware filtering. Can I know what implementation of this is used?

soulslicer, 2016-02-11:
Thanks for answering my questions

closed 2016-02-11 by soulslicer