OpenKinect/libfreenect2 #570: iai_kinect2 does not work with latest commit of libfreenect2.

haxan7, 2016-02-11:
`Protonect` works with `gl cl and cpu` but `roslaunch kinect2_bridge kinect2_bridge.launch` fails to publish anything.

Everything works as expected when libfreenect2 is rolled back to 9f39b744a8fe89d4d86e38bb2c494303ab41ae86.

I have also opened this issue on [iai_kinect2](https://github.com/code-iai/iai_kinect2/issues/223).

xlz, 2016-02-11:
What is the first commit that doesn't work?

wiedemeyer, 2016-02-11:
Good question.
Commit 9f19d7a8c8d65f40fe489ebd083c83511f837535 seems to work, while the next commit 834ad740a0039dd9ddc027f08bf1575c4b5c0d44 segfaults. The first which just hangs and does not segfault is f7a9aa1d336bd52a0fb84c8a3124f442239239e6.

xlz, 2016-02-11:
Case in point, if you just want something that works, you can just use the 0.1 series release version instead of a commit version.

xlz, 2016-02-11:
There are two typos in f7a9aa1 `usb: Add more error checking`, one is:

```diff
-  usb_control_.setIrInterfaceState(UsbControl::Disabled);
+  if (!usb_control_.setIrInterfaceState(UsbControl::Disabled) != UsbControl::Success) return false;
```

I didn't spot this and tests didn't show it.

The other thing is, now the semantics of `Freenect2Device::{start,stop}()` has been improved and their return values are now meaningful: if they return false, the usb device has errors and is in an undefined state and you must close it. iai_kinect2 should use the new error checking if it can (@wiedemeyer). In this case iai_kinect2 hangs because it calls `stop()` then `stop()` (which fails, though unjustifiably), and then the second `start()` fails to actually start , and iai_kinect2 then waits for the unstarted data stream forever.

closed 2016-02-11 by xlz

wiedemeyer, 2016-02-12:
Thanks for the quick fix. I updated kinect2_bridge to react to the return values.

haxan7, 2016-02-12:
Thanks @wiedemeyer  and @xlz for prompt fixes.

commit 6dcc5e9a (2016-02-11): usb: Fix typos in error reporting

The typos made iai_kinect2 hang.

Fixes #570