OpenKinect/libfreenect2 #571: how can i get data from Kinect v2 to PCL?

kerolos, 2016-02-11:
hello,
the libfreenect2 is working very fine in my PC, but i wanna use this data from Kinect v2 into PCL. i followed this link "http://robotica.unileon.es/mediawiki/index.php/PCL/OpenNI_tutorial_1:_Installing_and_testing" but i think , it is only for kinect v1 by calling this function "Grabber* openniGrabber;  " .
thanks in advance

xlz, 2016-02-11:
There is OpenNI2 interface you can probably use. PCL seems to have several OpenNI2 examples.

moonsh, 2016-02-13:
How about following this ?

"https://github.com/giacomodabisias/libfreenect2pclgrabber"

He combined Libfreenect2 and PCL

ahundt, 2016-02-13:
I also have a modified version of the above at https://github.com/ahundt/libfreenect2pclgrabber

kerolos, 2016-02-13:
thanks a lot Moonsh and Ahundt for your replay. i really appreciate that.

closed 2016-02-14 by xlz