OpenKinect/libfreenect2 #572: Jetson TK1 R21.4   LIBUSB_ERROR_IO Input/Output Error

bingyo, 2016-02-13:
I try to install in Jetson TK1.
But I do not see anything in the Protonect window that opens and on the screen continuously skipping rgb packet! error is displayed.

The setup and result are as follows:

```
$ uname -a
Linux tegra-ubuntu 3.10.40-gdacac96 #1 SMP PREEMPT Thu Jun 25 15:25:11 PDT 2015 armv7l armv7l armv7l GNU/Linux
```

I use [postFlash](https://github.com/jetsonhacks/postFlash) (Finished usb_port_owner_info=2, autosuspend=-1 .etc)

```
$ LIBUSB_DEBUG=3 ./Protonect 
```
[Protonect.txt](https://github.com/OpenKinect/libfreenect2/files/129008/Protonect.txt)

```
$ dmesg
```
[dmesg.txt](https://github.com/OpenKinect/libfreenect2/files/129009/dmesg.txt)

Please help me.

bingyo, 2016-02-13:
I used libfreenect2 v0.1.1

xlz, 2016-02-15:
I am unable to reproduce.

Kernel cmdline:
```
console=ttyS0,115200n8 console=tty1 no_console_suspend=1 lp0_vec=2064@0xf46ff000 mem=2015M@2048M memtype=255 ddr_die=2048M@2048M section=256M pmuboard=0x0177:0x0000:0x02:0x43:0x00 tsec=32M@3913M otf_key=c75e5bb91eb3bd947560357b64422f85 usbcore.old_scheme_first=1 core_edp_mv=1150 core_edp_ma=4000 tegraid=40.1.1.0.0 debug_uartport=lsport,3 power_supply=Adapter audio_codec=rt5640 modem_id=0 android.kerneltype=normal fbcon=map:1 commchip_id=0 usb_port_owner_info=2 lane_owner_info=6 emc_max_dvfs=0 touch_id=0@0 board_info=0x0177:0x0000:0x02:0x43:0x00 root=/dev/mmcblk0p1 rw rootwait tegraboot=sdmmc gpt usbcore.autosuspend=-1
```

Kernel version
```
Linux version 3.10.40-gdacac96 (buildbrain@mobile-u64-286) (gcc version 4.5.3 (crosstool-NG hg_unknown@20110628.165246) ) #1 SMP PREEMPT Thu Jun 25 15:25:11 PDT 2015
```

Protonect output
```
$ DISPLAY=:0.0 ./bin/Protonect 
Version: 0.1.1
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [gl | cl | cpu] [<device serial>] [-noviewer] [-help] [-version]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 6 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:4 with serial 002676751047
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] enabling usb transfer submission...
[Info] [Freenect2DeviceImpl] submitting usb transfers...
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Info] [Freenect2DeviceImpl] started
device serial: 002676751047
device firmware: 4.0.3911.0
[Debug] [DepthPacketStreamParser] not all subsequences received 992
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
...
[Debug] [DepthPacketStreamParser] skipping depth packet
[Info] [OpenGLDepthPacketProcessor] avg. time: 26.4314ms -> ~37.8338Hz
```

bingyo, 2016-02-16:
I have the same Kernel cmdline, Kernel version as you.
I do not know...
Do you use v0.1.1(libfreenect2)?

lsusb -t
Maybe same you.
```
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=tegra-xhci/2p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=tegra-xhci/6p, 480M
    |__ Port 3: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 4: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 4: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 4: Dev 3, If 2, Class=Human Interface Device, Driver=usbhid, 12M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/1p, 5000M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 3, If 1, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 3, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
        |__ Port 1: Dev 3, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/1p, 480M
```

xlz, 2016-02-16:
> Do you use v0.1.1(libfreenect2)?

```
$ DISPLAY=:0.0 ./bin/Protonect 
Version: 0.1.1
```

It looks like your USB 3 hardware configuration is different from mine. I use the onboard  blue USB 3.0 port only without any additional hub or extension card. If you did the same, you shouldn't be able to have additional USB 2 devices as shown in your lsusb -t.

bingyo, 2016-02-17:
>$ DISPLAY=:0.0 ./bin/Protonect 
>Version: 0.1.1

I understand same Version.

>If you did the same, you shouldn't be able to have additional USB 2 devices as shown in your lsusb -t.

I used the onboard blue USB 3.0 port.
When It's work!
I can see RGB, IR, Depth image.
Thank you very much!
Please close issues.

consult...
lsusb -t
```
ubuntu@tegra-ubuntu:~/rtabmap/build$ lsusb -t   
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=tegra-ehci/1p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=tegra-xhci/2p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/1p, 5000M
        |__ Port 1: Dev 7, If 0, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 7, If 1, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 7, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
        |__ Port 1: Dev 7, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=tegra-xhci/6p, 480M
    |__ Port 3: Dev 2, If 0, Class=Hub, Driver=hub/1p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 4: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 4: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 4: Dev 3, If 2, Class=Human Interface Device, Driver=usbhid, 12M
```

closed 2016-02-17 by floe