OpenKinect/libfreenect2 #578: listener.waitForNewFrame(frames); Access violation

idencl, 2016-02-18:
I'm having this run-time problem, namely when I compile Protonect example it runs fine until this particular line of code is executed:

```
listener.waitForNewFrame(frames);
```
it gives access violation error:
First-chance exception at 0x00007fff30f389ef (freenect2.dll) in xyz.exe:
0xC0000005: Access violation reading location 0x0000000000000008.

Any ideas what could be the problem?
Much appreciated!

xlz, 2016-02-18:
It seems you are using Visual Studio. Can you provide some information on which line of code generates the segfault?

idencl, 2016-02-18:
Exactly, I am compiling with Visual Studio 2010<tel:2010>. I am AFK at the moment, but the error is on line ~1477<tel:1477> in xtree, function Clear, first line. I think it goes: "_Clear(_Root());"

I'm compiling with Release config, and I'm using "Release x64" compiled libfreenect2 too.

Thanks!

xlz, 2016-02-18:
Can you pinpoint the line of code that is within libfreenect2? I don't recognize where the line 1477 comes from.

idencl, 2016-02-18:
Sorry, I was referring to xtree in my last comment, not libfreenect2. This is the pinpointed line that is causing segfault actually.

And the code that I am compiling is Protonect.cpp without Viewer, line 322:  "listener.waitForNewFrame(frames);"

xlz, 2016-02-18:
The variables, `listener_` and `frames`. Do they look OK under the debugger?

idencl, 2016-02-19:
Yup...
```listener```is of type **SyncMultiFrameListener** - and works fine for instance for following command: ```listener.hasNewFrame();```

```frames``` is an empty **FrameMap**

So to pinpoint when it breaks: frames take me to definition of FrameMap which is of type std::map, which inherits from class _Tree (this is in the file *xtree*), function ```void clear()``` on line 1471.

What I noticed is that here the first address is 32bit, and second one is 64bit, could I be doing something wrong with libraries?
> First-chance exception at 0x00007fff30f389ef (freenect2.dll) in xyz.exe:
> 0xC0000005: Access violation reading location 0x0000000000000008

idencl, 2016-02-19:
Apparently it was a library mismatch. I recompiled libfreenect2 for VS14,  added new path to the linker and compiled Protonect example with VS14, and now it works.

Anyone experiencing strange segfaults - this might be the issue...

closed 2016-02-19 by idencl