OpenKinect/libfreenect2 #580: Protonect on windows: failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED

mmeliabit, 2016-02-19:
Hello,

I was trying to run Protonect on our Windows 8.1 and I can't get to make it run. I don't know what I am doing wrong. I followed the instructions and the only error I got was on the .cmd execution to build libusb, and I finished the compilation with VS using the libusb_2013.sln generated by the command.

I associated the libusbK 3.0.7.0 driver with Xbox NUI Sensor (Composite Parent). However, if I run again Zadig I do not see this device anymore (only Xbox NUI Sensor (without "Composite Parent"), and two NUI Sensor Adaptors.

The first error I get in the logs when I run Protonect is: [Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform

The rest of the log:
[ 0.000000] [0000141c] libusb: debug [libusb_init] created default context
[ 0.000000] [0000141c] libusb: debug [libusb_init] libusb v1.0.20.11049
[ 0.000000] [0000141c] libusb: debug [windows_init] Windows 8.1 64-bit
[ 0.000000] [0000141c] libusb: debug [setup_cancel_io] Will use CancelIoEx for I/O cancellation
[ 0.000000] [0000141c] libusb: debug [winusbx_init] using libusbK DLL for universal access
[ 0.000000] [0000141c] libusb: debug [winusbx_init] libusbK version: 3.0.7.0
[ 0.000000] [0000141c] libusb: debug [winusbx_init] initalized sub API libusbK
[ 0.000000] [0000141c] libusb: debug [winusbx_init] initalized sub API libusb0
[ 0.000000] [0000141c] libusb: debug [winusbx_init] initalized sub API WinUSB
[ 0.000000] [0000141c] libusb: debug [windows_init_clock] hires timer available (Frequency: 14318180 Hz)
[ 0.000000] [0000141c] libusb: debug [windows_init_clock] timer thread will run on core #0
[ 0.000000] [0000141c] libusb: debug [htab_create] using 1021 entries hash table
[ 0.000000] [0000141c] libusb: debug [usbi_add_pollfd] add fd 0 events 1
[ 0.000000] [000015f4] libusb: debug [libusb_get_next_timeout] no URBs, no timeout!
[ 0.000000] [000015f4] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.000000] [000015f4] libusb: debug [handle_events] poll fds modified, reallocating
[ 0.000000] [000015f4] libusb: debug [handle_events] poll() 1 fds with timeout in 100ms
[ 0.015625] [0000141c] libusb: debug [libusb_get_device_list] 
[ 0.015625] [0000141c] libusb: debug [windows_get_device_list] allocating new device for session [16B]
[ 0.015625] [0000141c] libusb: debug [windows_get_device_list] allocating new device for session [215]
[ 0.015625] [0000141c] libusb: debug [windows_get_device_list] allocating new device for session [356]
[ 0.015625] [0000141c] libusb: debug [windows_get_device_list] allocating new device for session [24D]
[ 0.015625] [0000141c] libusb: debug [windows_get_device_list] allocating new device for session [255]
[ 0.015625] [0000141c] libusb: debug [windows_get_device_list] allocating new device for session [214]
[ 0.015625] [0000141c] libusb: debug [windows_get_device_list] allocating new device for session [365]
[ 0.015625] [0000141c] libusb: debug [get_api_type] driver(s): usbhub
[ 0.015625] [0000141c] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.015625] [0000141c] libusb: debug [windows_get_device_list] allocating new device for session [6B]
[ 0.015625] [0000141c] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.015625] [0000141c] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.015625] [0000141c] libusb: debug [windows_get_device_list] allocating new device for session [13A]
[ 0.015625] [0000141c] libusb: debug [get_api_type] driver(s): usbhub
[ 0.015625] [0000141c] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.015625] [0000141c] libusb: debug [windows_get_device_list] allocating new device for session [5E]
[ 0.015625] [0000141c] libusb: debug [get_api_type] driver(s): usbhub
[ 0.015625] [0000141c] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.015625] [0000141c] libusb: debug [windows_get_device_list] allocating new device for session [19E]
[ 0.015625] [0000141c] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.015625] [0000141c] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.015625] [0000141c] libusb: debug [windows_get_device_list] allocating new device for session [D1]
[ 0.015625] [0000141c] libusb: debug [get_api_type] driver(s): usbhub
[ 0.015625] [0000141c] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.015625] [0000141c] libusb: debug [windows_get_device_list] allocating new device for session [336]
[ 0.015625] [0000141c] libusb: debug [get_api_type] driver(s): usbhub
[ 0.015625] [0000141c] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.015625] [0000141c] libusb: debug [windows_get_device_list] allocating new device for session [2FF]
[ 0.015625] [0000141c] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.015625] [0000141c] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.015625] [0000141c] libusb: debug [windows_get_device_list] allocating new device for session [D3]
[ 0.015625] [0000141c] libusb: debug [get_api_type] driver(s): USBHUB3
[ 0.015625] [0000141c] libusb: debug [get_api_type] matched driver name against HUB API
[ 0.015625] [0000141c] libusb: debug [windows_get_device_list] allocating new device for session [3E7]
[ 0.015625] [0000141c] libusb: debug [windows_get_device_list] found existing device for session [5E] (0.0)
[ 0.015625] [0000141c] libusb: debug [init_device] (bus: 1, addr: 1, depth: 0, port: 0): '\\.\USB#ROOT_HUB#4&3241D132&0'
[ 0.015625] [0000141c] libusb: debug [windows_get_device_list] found existing device for session [3E7] (0.0)
[ 0.015625] [0000141c] libusb: debug [init_device] (bus: 6, addr: 1, depth: 0, port: 0): '\\.\USB#ROOT_HUB30#4&2D1E3512&0&0'
[ 0.015625] [0000141c] libusb: debug [windows_get_device_list] found existing device for session [336] (0.0)
[ 0.015625] [0000141c] libusb: debug [init_device] (bus: 3, addr: 1, depth: 0, port: 0): '\\.\USB#ROOT_HUB20#4&D40F25F&0'
[ 0.015625] [0000141c] libusb: debug [windows_get_device_list] found existing device for session [19E] (0.0)
[ 0.015625] [0000141c] libusb: debug [init_device] (bus: 4, addr: 1, depth: 0, port: 0): '\\.\USB#ROOT_HUB#4&310F6F76&0'
[ 0.015625] [0000141c] libusb: debug [windows_get_device_list] allocating new device for session [1A1]
[... 1938 lines cut ...]
[ 1.969293] [0000141c] libusb: debug [libusb_free_transfer] transfer 0000004F76306440
[ 1.969293] [0000141c] libusb: debug [libusb_free_transfer] transfer 0000004F76304A60
[ 1.969293] [0000141c] libusb: debug [libusb_free_transfer] transfer 0000004F76306320
[ 1.969293] [0000141c] libusb: debug [libusb_free_transfer] transfer 0000004F76305120
[ 1.969293] [0000141c] libusb: debug [libusb_free_transfer] transfer 0000004F76306560
[ 1.969293] [0000141c] libusb: debug [libusb_free_transfer] transfer 0000004F79DD5B70
[ 1.969293] [0000141c] libusb: debug [libusb_free_transfer] transfer 0000004F79DC9460
[ 1.969293] [0000141c] libusb: debug [libusb_free_transfer] transfer 0000004F79D43F10
[ 1.969293] [0000141c] libusb: debug [libusb_free_transfer] transfer 0000004F79DC8420
[ 1.969293] [0000141c] libusb: debug [libusb_free_transfer] transfer 0000004F755DFAD0
[ 1.969293] [0000141c] libusb: debug [libusb_free_transfer] transfer 0000004F763121D0
[ 1.969293] [0000141c] libusb: debug [libusb_free_transfer] transfer 0000004F7584D8B0
[ 1.969293] [0000141c] libusb: debug [libusb_free_transfer] transfer 0000004F756C23E0
[ 1.969293] [0000141c] libusb: debug [libusb_free_transfer] transfer 0000004F79DAEDF0
[ 1.969293] [0000141c] libusb: debug [libusb_free_transfer] transfer 0000004F79DB9B40
[ 1.969293] [0000141c] libusb: debug [libusb_free_transfer] transfer 0000004F79DB99C0
[ 1.969293] [0000141c] libusb: debug [libusb_free_transfer] transfer 0000004F79DB93C0
[ 1.969293] [0000141c] libusb: debug [libusb_free_transfer] transfer 0000004F79DB9CC0
[ 1.969293] [0000141c] libusb: debug [libusb_free_transfer] transfer 0000004F79DB9540
[ 1.969293] [0000141c] libusb: debug [libusb_free_transfer] transfer 0000004F79DB96C0
[ 1.969293] [0000141c] libusb: debug [libusb_free_transfer] transfer 0000004F79DB9E40
[ 1.969293] [0000141c] libusb: debug [libusb_free_transfer] transfer 0000004F79DB90C0
[ 1.969293] [0000141c] libusb: debug [libusb_free_transfer] transfer 0000004F79DB9840
[ 1.969293] [0000141c] libusb: debug [libusb_free_transfer] transfer 0000004F79DB9240
[ 1.969293] [0000141c] libusb: debug [libusb_free_transfer] transfer 0000004F7A5134D0

mmeliabit, 2016-02-19:
I replaced my freenect and libusb DLLs with prebuild ones download from here: https://github.com/OpenKinect/libfreenect2/releases/tag/v0.1.0

Now it is working. I don't know what I did wrong.

xlz, 2016-02-19:
https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting#libusb_error_notsupported

> Also check the build log of install_libusb_vs201?.cmd to look for any errors that disable the libusbK backend.

?

xlz, 2016-02-19:
> the only error I got was on the .cmd execution to build libusb

What error? Can you post the log?

langmingzhou, 2016-02-22:
Hello,
  I had meet the same error of @mmeliabit. I debug into Protonect, and it was stop in the line "return -1":
  if (enable_rgb && enable_depth)
  {
    if (!dev->start())
      return -1; //step out there
  }
and the information output in the screen is: [Error][usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_NOT_SUPPORTED Operation not supported or unimplemented on this platform.

Also, I set enable_rgb=true and enable_depth = false. Then run Protonect again, then I got those error:
  while(!protonect_shutdown)
  {
    listener.waitForNewFrame(frames); // step out there
    libfreenect2::Frame *rgb = frames[libfreenect2::Frame::Color];
    libfreenect2::Frame *ir = frames[libfreenect2::Frame::Ir];
    libfreenect2::Frame *depth = frames[libfreenect2::Frame::Depth];
and the information output in the screen is: [Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!

My OS is window 10. And the old version:
https://github.com/OpenKinect/libfreenect2/releases/tag/v0.1.0
can work successfully in my computer. The same confusion: I don't know what I did wrong.

xlz, 2016-02-22:
@langmingzhou 

> Also check the build log of install_libusb_vs201?.cmd to look for any errors that disable the libusbK backend.

Do you see any errors?

langmingzhou, 2016-02-22:
I run install_libusb_vs2013.cmd in git shell. And there are some errors, I record them in a screenshot. 
[image]

langmingzhou, 2016-02-22:
But I can finished the compilation with VS using the libusb_2013.sln generated by the command "install_libusb.sh". The previous debug work of protonect is based on such libusb-1.0 dll and libusb-1.0 lib. And the version of those dll and lib are win32.

xlz, 2016-02-22:
OK, I found out I forgot to update the script for VS2013. This is the merge conflict that makes your libusb not work.

Try the latest master now.

langmingzhou, 2016-02-22:
Ok, it is working. Thanks a lot for your work. And another question: can I change the platform to win32 from original x64 cause my main framework is based on win32?

xlz, 2016-02-22:
I don't know. It is not tested under win32 so there is no guarantee it will work.

Btw, building with win32 results in worse performance from a test long time ago.

closed 2016-02-22 by xlz