OpenKinect/libfreenect2 #581: Libfreenect2 not working on Jetson TX1

menlocaleb, 2016-02-19:
I'm trying to run libfreenect2 on an Nvidia Jetson TX1 Developer Kit and can't seem to get it working. Here's the output I'm getting:
`$ LIBUSB_DEBUG=3 ./bin/Protonect 
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 18 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:6 with serial 068176734547
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9855
[Debug] [Freenect2DeviceImpl] status 0x090000: 9855
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] subpacket too large
[Info] [Freenect2DeviceImpl] started
device serial: 068176734547
device firmware: 4.0.3911.0
[Debug] [DepthPacketStreamParser] image data too short!
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] image data too short!
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] image data too short!
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] image data too short!
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] image data too short!
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] image data too short!
[Debug] [DepthPacketStreamParser] not all subsequences received 0
`

I've ensured that USB 3 is enabled in /boot/extlinux/extlinux.conf and turned off usb autosuspend using `for i in /sys/bus/usb/devices/*/power/autosuspend; do echo -1 >$i; done`

I've checked dmesg for all of the issues mentioned in the Troubleshooting wiki and couldn't find anything.

Output from lspci:
`00:01.0 PCI bridge: NVIDIA Corporation Device 0fae (rev a1)
01:00.0 PCI bridge: Pericom Semiconductor Device 2608
02:01.0 PCI bridge: Pericom Semiconductor Device 2608
02:02.0 PCI bridge: Pericom Semiconductor Device 2608
02:03.0 PCI bridge: Pericom Semiconductor Device 2608
02:04.0 PCI bridge: Pericom Semiconductor Device 2608
03:00.0 USB controller: Renesas Technology Corp. uPD720202 USB 3.0 Host Controller (rev 02)
04:00.0 USB controller: Renesas Technology Corp. uPD720202 USB 3.0 Host Controller (rev 02)
05:00.0 USB controller: Renesas Technology Corp. uPD720202 USB 3.0 Host Controller (rev 02)
06:00.0 USB controller: Renesas Technology Corp. uPD720202 USB 3.0 Host Controller (rev 02)
`

Output from lsusb -t:
`/:  Bus 10.Port 1: Dev 1, Class=root_hub, Driver=tegra-xhci/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Vendor Specific Class, Driver=r8152, 5000M
    |__ Port 2: Dev 3, If 0, Class=Hub, Driver=hub/4p, 5000M
/:  Bus 09.Port 1: Dev 1, Class=root_hub, Driver=tegra-xhci/5p, 480M
    |__ Port 3: Dev 3, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 1: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 2: Dev 6, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
/:  Bus 08.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 5000M
/:  Bus 07.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 5000M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 5000M
    |__ Port 1: Dev 5, If 0, Class=Hub, Driver=hub/1p, 5000M
        |__ Port 1: Dev 6, If 0, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 6, If 1, Class=Vendor Specific Class, Driver=, 5000M
        |__ Port 1: Dev 6, If 2, Class=Audio, Driver=snd-usb-audio, 5000M
        |__ Port 1: Dev 6, If 3, Class=Audio, Driver=snd-usb-audio, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 4, If 0, Class=Hub, Driver=hub/1p, 480M`

I did succeed in getting the [jetsonhacks version of libfreenect2](http://jetsonhacks.com/2015/02/26/install-kinect-v2-using-libfreenect2-on-nvidia-jetson-tk1/) to work after a few small tweaks so I know it's possible to run on the TX1, I just can't figure out how to get the main version of libfreenect2 to work. I need to do this because I want to use the iai_kinect2 code to integrate with ROS, and that code needs the main version of libfreenect2. Any help would be greatly appreciated!

xlz, 2016-02-19:
> I did succeed in getting the jetsonhacks version of libfreenect2 to work after a few small tweaks

How? Anything more specific? jetsonhacks was mostly using one of my branch which has been incorporated into the upstream libfreenect2 so there shouldn't be any functional difference. It's the a few small tweaks that made the difference.

Frankly, several people have come to ask about problems with TX1. As I don't have a TX1, some of the investigation can only be done on the user side. And they did not follow up so the problems are never solved.

menlocaleb, 2016-02-20:
I changed the wget line in jetsonhacks' install script to a new version of gstjpeg: http://developer.download.nvidia.com/embedded/L4T/r23_Release_v1.0/source/gstjpeg_src.tbz2

I also changed line 113 of CMakeLists.txt from "jpeg" to "nvjpeg" to fix a build error.

Those are the only changes I made. However, I also ran the jetsonhacks version without CUDA or the Tegra acceleration in order to try to get it to work with the iai_kinect2 code and it still worked (the altered iai_kinect2 code crashes though). So I think something other than the optimizations (CUDA and Tegra) that you added are making it work.

Are there any other commands I can run to help you help me debug? Or any issues you think it might be?

xlz, 2016-02-20:
> You may need to maximize CPU and GPU clock to maintain basic Kinect functionalities, see http://elinux.org/Jetson/Performance for instructions.

I suppose you have done this? Also checked other issues here https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting#jetson-tk1-issues ?

We definitely have one working version and one non-working version to compare with. Several things to look for

* Can you download http://ports.ubuntu.com/ubuntu-ports/pool/main/libu/libusb-1.0/libusb-1.0-0_1.0.20-1_armhf.deb http://ports.ubuntu.com/ubuntu-ports/pool/main/libu/libusb-1.0/libusb-1.0-0-dev_1.0.20-1_armhf.deb, `dpkg -i` them, and built the jetsonhacks version with this 1.0.20 binary version instead of building libusb from source? (Remove depends/libusb* first).
* Are there any libusb errors from `LIBUSB_DEBUG=3 ./bin/Protonect`?
* Are there anything relevant in `dmesg`?

menlocaleb, 2016-02-20:
I maximized all CPU's and set the GPU clock speed to 10x what it was (76800 kHz -> 768000 kHz). This didn't fix anything.

I have the correct udev rules in place and I've enabled usb3 with usb_port_owner_info=2 and disabled autosuspend with usbcore.autosuspend=-1.

When I run the Kinect through a USB3 hub rather than a PCI Express USB 3.0 card I only get the following error: `[Debug] [RgbPacketStreamParser] skipping rgb packet!` I don't get any depth errors. However when I run the jetson version, the opposite is true: it works when run through the PCI board, but fails with the following error when run through the USB 3 hub: `[DepthPacketStreamParser::handleNewData] subpacket incomplete (needed 298496 received 28160)`. It is worth noting that the jetson version when it works also outputs a number of `[Debug] [RgbPacketStreamParser] skipping rgb packet!` errors, but it still outputs at a decent framerate.

Using the libusb debs you linked to didn't fix anything either. However, I didn't have a depends/libusb* folder or file, I only have the install_libusb.sh script and the two deb files in the debs folder.

There weren't any libusb errors or messages when run with `LIBUSB_DEBUG=3`.

I get the following errors in dmesg:
```usb 4-1.1: reset SuperSpeed USB device number 7 using xhci_hcd```
```usb 4-1.1: Parent hub missing LPM exit latency info.  Power management will be impacted```
```xhci_hcd 0000:04:00.0: WARN Successful completion on short TX: needs XHCI_TRUST_TX_LENGTH quirk?```

Does that tell you anything?

xlz, 2016-02-20:
Everyone thinks `[Debug] [RgbPacketStreamParser] skipping rgb packet!` is an error. It's not, sigh. It means the JPEG decoder is too slow and some usb data is received and then discarded.

> [Debug] [RgbPacketStreamParser] skipping rgb packet!
> [DepthPacketStreamParser::handleNewData] subpacket incomplete (needed 298496 received 28160)

These are not errors. They just show inefficiency. They are irrelevant in determining if it works or not.

> usb 4-1.1: reset SuperSpeed USB device number 7 using xhci_hcd
> usb 4-1.1: Parent hub missing LPM exit latency info. Power management will be impacted

Not errors.

> xhci_hcd 0000:04:00.0: WARN Successful completion on short TX: needs XHCI_TRUST_TX_LENGTH quirk?

Not an error, but could indicate something indirect about your USB controller.

It's not clear to me from your report which is working and which is not. Also, you really need to clarify your hardware configuration. Something unusual is going on with your `lspci`, four USB controllers really?

* With the upstream libfreenect2 (Btw I can tell you are not using the latest version because added command line options is not in your Protonect output)
    - "When I run the Kinect through a USB3 hub", 
        1. Is there framerate stats printed?
        2. Does the viewer show images?
    - "When I run the Kinect through a PCI Express USB 3.0 card", 
        1. Is there framerate stats printed?
        2. Does the viewer show images?
    - When you connect the Kinect directly to the onboard USB 3 port, instead of a hub or an USB extension card?
        1. Is there framerate stats printed?
        2. Does the viewer show images?
* With jetsonhacks version
    - "When I run the Kinect through a USB3 hub", 
        1. Is there framerate stats printed?
        2. Does the viewer show images?
    - "When I run the Kinect through a PCI Express USB 3.0 card", 
        1. Is there framerate stats printed?
        2. Does the viewer show images?
    - When you connect the Kinect directly to the onboard USB 3 port, instead of a hub or an USB extension card?
        1. Is there framerate stats printed?
        2. Does the viewer show images?

menlocaleb, 2016-02-20:
- Upstream libfreenect2 (now up to date)
  - USB 3 Hub
    - TurboJpegRgbPacketProcesser ~23.78Hz and no depth frame rate
    - The viewer is just a black screen
  - PCI Express USB 3.0 Card
    - TurboJpegRgbPacketProcesser ~23.69Hz and no depth frame rate
    - The viewer is just a black screen
 - jestonhacks version
  - USB 3 Hub
    - no frame rate output
    - viewers don't work
  - PCI Express USB 3.0 Card
    - TurboJpeg ~22.81 Hz and OpenGL ~61.96 Hz
    - Viewers all work

I had trouble testing the onboard USB 3 port because I wasn't able to run the program over ssh because of a GLFW error. Could it be because of a version mismatch between glfw on my workstation vs the TX1?

I'm running the TX1 development board with a 4 port PCI Express USB 3.0 Card with 4 dedicated 5Gbps Channels as well as a USB 3.0 4 port hub plugged into the built-in TX1 usb port.

xlz, 2016-02-20:
You can run it over ssh using `DISPLAY=:0.0 ./bin/Protonect`.

xlz, 2016-02-21:
Previously there are several cases when people use external USB extension cards or USB hubs on Jetson and it didn't work. Once they use the built-in port it worked. So definitely check the built-in port.

This hints at mostly USB problems. Yes, the upstream libfreenect2 should work even with the USB extension card. I don't recall any dramatic changes to the USB transfer code in libfreenect2 since the jetsonhacks version.

Besides the test with the built-in port, can you do the test using the upstream version and the extension card port, and provide:
* Log of `LIBUSB_DEBUG=4 ./bin/Protonect` (this will be lengthy)
* Full `dmesg`

menlocaleb, 2016-02-22:
Running with built-in usb port (run over ssh -X with DISPLAY=:0.0)
- upstream
  - Turbo jpeg ~24 Hz, not depth hz output
  - No viewer at all
- jetsonhacks
  - No frame rate info or viewer at all

xlz, 2016-02-22:
I really need log of `LIBUSB_DEBUG=4 ./bin/Protonect` and the full `dmesg`. If you can, run it with `./bin/Protonect cuda`.

Even with Jetson TK1, TegraJPEG and CUDA processors provide 85Hz and 94Hz.

Also, it seems your build didn't enable TegraJPEG support. A user https://github.com/OpenKinect/libfreenect2/issues/480#issuecomment-185791572 had the same issue but hasn't followed up. The problem is there should be `gstjpeg_src/nv_headers` extracted under `build` directory, but it didn't. Maybe you can find out why.

menlocaleb, 2016-02-22:
Running upstream libfreenect with PCI express card (not CUDA or Tegra accelerated:
```
acme@tegra-ubuntu:~/dev/libfreenect2/build$ LIBUSB_DEBUG=4 ./bin/Protonect 
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
To pause and unpause: pkill -USR1 Protonect
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000005] [00003191] libusb: debug [libusb_init] created default context
[ 0.000099] [00003191] libusb: debug [libusb_init] libusb v1.0.20.11004
[ 0.000187] [00003191] libusb: debug [find_usbfs_path] found usbfs at /dev/bus/usb
[ 0.000237] [00003191] libusb: debug [op_init] bulk continuation flag supported
[ 0.000267] [00003191] libusb: debug [op_init] zero length packet flag supported
[ 0.000317] [00003191] libusb: debug [op_init] sysfs can relate devices
[ 0.000343] [00003191] libusb: debug [op_init] sysfs has complete descriptors
[ 0.000653] [00003192] libusb: debug [linux_udev_event_thread_main] udev event thread entering.
[ 0.002135] [00003191] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.002189] [00003191] libusb: debug [linux_get_device_address] scan usb1
[ 0.002377] [00003191] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.002412] [00003191] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.002438] [00003191] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.002825] [00003191] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.002868] [00003191] libusb: debug [linux_get_device_address] scan usb2
[ 0.003079] [00003191] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.003111] [00003191] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.003137] [00003191] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.003424] [00003191] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
[ 0.003457] [00003191] libusb: debug [linux_get_device_address] scan usb3
[ 0.003587] [00003191] libusb: debug [linux_get_device_address] bus=3 dev=1
[ 0.003615] [00003191] libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769
[ 0.003641] [00003191] libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)
[ 0.003928] [00003191] libusb: debug [linux_get_device_address] getting address for device: 3-1 detached: 0
[ 0.003961] [00003191] libusb: debug [linux_get_device_address] scan 3-1
[ 0.004092] [00003191] libusb: debug [linux_get_device_address] bus=3 dev=7
[ 0.004123] [00003191] libusb: debug [linux_enumerate_device] busnum 3 devaddr 7 session_id 775
[ 0.004150] [00003191] libusb: debug [linux_enumerate_device] allocating new device for 3/7 (session 775)
[ 0.004283] [00003191] libusb: debug [linux_get_parent_info] Dev 0x1c6e8 (3-1) has parent 0x1c318 (usb3) port 1
[ 0.004492] [00003191] libusb: debug [linux_get_device_address] getting address for device: usb4 detached: 0
[ 0.004525] [00003191] libusb: debug [linux_get_device_address] scan usb4
[ 0.004654] [00003191] libusb: debug [linux_get_device_address] bus=4 dev=1
[ 0.004687] [00003191] libusb: debug [linux_enumerate_device] busnum 4 devaddr 1 session_id 1025
[ 0.004715] [00003191] libusb: debug [linux_enumerate_device] allocating new device for 4/1 (session 1025)
[ 0.004995] [00003191] libusb: debug [linux_get_device_address] getting address for device: 4-1 detached: 0
[ 0.005028] [00003191] libusb: debug [linux_get_device_address] scan 4-1
[ 0.005160] [00003191] libusb: debug [linux_get_device_address] bus=4 dev=12
[ 0.005192] [00003191] libusb: debug [linux_enumerate_device] busnum 4 devaddr 12 session_id 1036
[ 0.005219] [00003191] libusb: debug [linux_enumerate_device] allocating new device for 4/12 (session 1036)
[ 0.005338] [00003191] libusb: debug [linux_get_parent_info] Dev 0x1df08 (4-1) has parent 0x1c408 (usb4) port 1
[ 0.005466] [00003191] libusb: debug [linux_get_device_address] getting address for device: 4-1.1 detached: 0
[ 0.005498] [00003191] libusb: debug [linux_get_device_address] scan 4-1.1
[ 0.005630] [00003191] libusb: debug [linux_get_device_address] bus=4 dev=13
[ 0.005654] [00003191] libusb: debug [linux_enumerate_device] busnum 4 devaddr 13 session_id 1037
[ 0.005675] [00003191] libusb: debug [linux_enumerate_device] allocating new device for 4/13 (session 1037)
[ 0.005789] [00003191] libusb: debug [linux_get_parent_info] Dev 0x1e388 (4-1.1) has parent 0x1df08 (4-1) port 1
[ 0.006335] [00003191] libusb: debug [linux_get_device_address] getting address for device: usb5 detached: 0
[ 0.006363] [00003191] libusb: debug [linux_get_device_address] scan usb5
[ 0.006488] [00003191] libusb: debug [linux_get_device_address] bus=5 dev=1
[ 0.006514] [00003191] libusb: debug [linux_enumerate_device] busnum 5 devaddr 1 session_id 1281
[ 0.006536] [00003191] libusb: debug [linux_enumerate_device] allocating new device for 5/1 (session 1281)
[ 0.006985] [00003191] libusb: debug [linux_get_device_address] getting address for device: usb6 detached: 0
[ 0.007015] [00003191] libusb: debug [linux_get_device_address] scan usb6
[ 0.007142] [00003191] libusb: debug [linux_get_device_address] bus=6 dev=1
[... cut ...]
[ 0.942383] [00003193] libusb: debug [reap_for_handle] urb type=0 status=0 transferred=135168
[ 0.942396] [00003193] libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 1/1
[ 0.942405] [00003193] libusb: debug [handle_iso_completion] last URB in transfer --> complete!
[ 0.942415] [00003193] libusb: debug [usbi_handle_transfer_completion] transfer 0x1e5738 has callback 0xf73fb335
```

I've attached my dmesg.
[dmesg.txt](https://github.com/OpenKinect/libfreenect2/files/141480/dmesg.txt)

xlz, 2016-02-22:
Does the output of `LIBUSB_DEBUG=4 ./bin/Protonect` stop after
```
[ 0.942415] [00003193] libusb: debug [usbi_handle_transfer_completion] transfer 0x1e5738 has callback 0xf73fb335
```
?

menlocaleb, 2016-02-22:
No but I wasn't sure how much more to do, by that point it was seeming like it was repeating.

By the time I stopped it (after running for 5 - 10 seconds) this is what was approximately repeating:
```
[Debug] [DepthPacketStreamParser] subpacket too large
[ 5.068774] [00003193] libusb: debug [libusb_submit_transfer] transfer 0x1e5c38
[ 5.069475] [00003193] libusb: debug [submit_iso_transfer] need 1 6144k URBs for transfer
[ 5.070595] [00003193] libusb: debug [reap_for_handle] urb type=0 status=0 transferred=264856
[ 5.070713] [00003193] libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 1/1
[ 5.071177] [00003193] libusb: debug [handle_iso_completion] last URB in transfer --> complete!
[ 5.071201] [00003193] libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 1000ms
[ 5.071225] [00003193] libusb: debug [usbi_handle_transfer_completion] transfer 0x1e5d38 has callback 0xf73fb335
[Debug] [DepthPacketStreamParser] subpacket too large
[ 5.071361] [00003193] libusb: debug [libusb_submit_transfer] transfer 0x1e5d38
[ 5.071392] [00003193] libusb: debug [submit_iso_transfer] need 1 6144k URBs for transfer
[ 5.071567] [00003193] libusb: debug [reap_for_handle] urb type=0 status=0 transferred=270336
[ 5.071596] [00003193] libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 1/1
[ 5.071617] [00003193] libusb: debug [handle_iso_completion] last URB in transfer --> complete!
[ 5.071637] [00003193] libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 1000ms
[ 5.071660] [00003193] libusb: debug [usbi_handle_transfer_completion] transfer 0x1e5e38 has callback
```

xlz, 2016-02-22:
So I suppose you are still getting a stream of 
```
[Debug] [DepthPacketStreamParser] image data too short!
[Debug] [DepthPacketStreamParser] subpacket too large
```
? This points to data loss at kernel driver level. It's probably some changes in libusb behaviors that expose bugs in the kernel driver for uPD720202.

This can be verified by printing packet lengths here https://github.com/jetsonhacks/libfreenect2/blob/jetsontk1/examples/protonect/src/depth_packet_stream_parser.cpp#L65

Add
```cpp
std::cout << "pkt len " << length << std::endl;
```
And record it.

I still want to verify which libusb this test is run with. Can you provide the output of `ldd ./bin/Protonect | grep libusb`?

xlz, 2016-02-22:
The output of `ldd ./bin/Protonect | grep libusb` for the upstream version and jetsonhacks version.

I really think there is some difference in their libusb.

menlocaleb, 2016-02-22:
In regards to the tegra headers, in the jetsonhacks version I changed the following line of the build file from 
```
 FIND_LIBRARY(TEGRA_JPEG_LIBRARY jpeg PATHS /usr/lib/arm-linux-gnueabihf/tegra NO_DEFAULT_PATH)
```
to
```
 FIND_LIBRARY(TEGRA_JPEG_LIBRARY nvjpeg PATHS /usr/lib/arm-linux-gnueabihf/tegra NO_DEFAULT_PATH)
```
(changed jpeg to nvjpeg).

For packet lengths in the jetsonhacks version (where all viewer windows work) there seem to be two variations:
```
pkt len 28312
pkt len 33792
```

Libusb versions:
- Jetsonhacks (some of path altered to hide real directory names)
```
$ ldd ./bin/Protonect | grep libusb
	libusb-1.0.so.0 => /home/user_name/dir1/root_of_jetsonhacks_repo/examples/protonect/../../depends/libusb/lib/libusb-1.0.so.0 (0xf696f000)
```
- upstream
```
ldd ./bin/Protonect | grep libusb
	libusb-1.0.so.0 => /lib/arm-linux-gnueabihf/libusb-1.0.so.0 (0xf71a8000)
```

You were right, there is a difference: the jetsonhacks version uses one from it's depends directory.

xlz, 2016-02-22:
What happens if
```
LD_PRELOAD=jetsonhacks/libusb.so.0 upstream/ build/bin/Protonect
```
?

xlz, 2016-02-22:
Also, what happens if jetsonhacks version uses the system libusb? (Remove the libusb in depends directory)

menlocaleb, 2016-02-22:
```
$ LD_PRELOAD=/home/user_name/dir1/jetsonhacks_root_dir/depends/libusb/lib/libusb-1.0.so.0 ./bin/Protonect 
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
To pause and unpause: pkill -USR1 Protonect
[Error] [Freenect2Impl] Your libusb does not support large iso buffer!
no device connected!
$ lsusb
Bus 009 Device 030: ID 046d:c077 Logitech, Inc. 
Bus 009 Device 029: ID 046d:c31c Logitech, Inc. Keyboard K120 for Business
Bus 009 Device 028: ID 2109:3431  
Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 010 Device 033: ID 2109:0811  
Bus 010 Device 002: ID 0955:09ff NVidia Corp. 
Bus 010 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 015: ID 045e:02c4 Microsoft Corp. 
Bus 004 Device 014: ID 045e:02d9 Microsoft Corp. 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 008: ID 045e:02d9 Microsoft Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
Am I doing something wrong? My Kinect is connected as my lsusb shows.

menlocaleb, 2016-02-22:
If I move the libusb directory out of the depends directory and try to build the jetsonhacks version I get a build error
```
$ make
[ 45%] Built target glfw
[ 47%] Built target generate_resources_tool
Scanning dependencies of target freenect2
[ 50%] Building CXX object CMakeFiles/freenect2.dir/src/transfer_pool.cpp.o
In file included from /home/user_name/dev/jetson_libfreenect2/examples/protonect/src/transfer_pool.cpp:27:0:
/home/user_name/dev/jetson_libfreenect2/examples/protonect/include/libfreenect2/usb/transfer_pool.h:31:20: fatal error: libusb.h: No such file or directory
 #include <libusb.h>
                    ^
compilation terminated.
make[2]: *** [CMakeFiles/freenect2.dir/src/transfer_pool.cpp.o] Error 1
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
make: *** [all] Error 2
```

Here are the offending lines of the cmake I assume:
```
INCLUDE_DIRECTORIES("${MY_DIR}/../../depends/libusb/include/libusb-1.0/")
LINK_DIRECTORIES("${MY_DIR}/../../depends/libusb/lib/")
```

xlz, 2016-02-22:
You didn't do anything wrong. Maybe it's the code structure which won't allow my proposed test.

I suspect the difference in libusb is this: https://github.com/jetsonhacks/libfreenect2/commit/e818be215f8e863dc6b5c0bea424d85d626dc7ec

Can you `git revert e818be215f8e863dc6b5c0bea424d85d626dc7ec` in the jetsonhacks directory, rebuild, and see if it gets the same error as upstream libfreenect2?

xlz, 2016-02-23:
If this commit indeed makes the difference, you can try to build libusb from source by patching it to change `MAX_ISO_BUFFER_LENGTH` from `49152 * 128` to `49152`.

menlocaleb, 2016-02-23:
You're right! When I reverted that commit the jetsonhacks version also broke. So in the upstream version I ran install_libusb.sh in the depends directory, then edited `MAX_ISO_BUFFER_LENGTH` to be `49152`. To get it to build I added the following lines to my CMakeLists.txt, but there might be a better way.
```
INCLUDE_DIRECTORIES("${MY_DIR}/depends/libusb/include/libusb-1.0/")
LINK_DIRECTORIES("${MY_DIR}/depends/libusb/lib/")
```

Now the upstream libfreenect2 works on the TX1!

However, now when I run the kinect2_bridge it outputs nanms/nanHz for the depth processing. This is better than before when it wasn't working at all, but I'd rather it worked all the way!

xlz, 2016-02-23:
I don't think you are supposed to need `INCLUDE_DIRECTORIES("${MY_DIR}/depends/libusb/include/libusb-1.0/")`?

After you built libusb in `depends` directory, and `rm -rf *` under `build` directory, what does `cmake ..` outputs?

xlz, 2016-02-23:
Now there are two remaining issues:

* Does the upstream with the patched libusb work with the built-in USB port?
* Are you able to enable the TegraJPEG decoder?

menlocaleb, 2016-02-23:
You're right, I didn't need those two CMake additions. With the original CMakeLists.txt file here is my output:
```
$ cmake ..
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- using tinythread as threading library
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- checking for module 'libusb-1.0>=1.0.20'
--   found libusb-1.0, version 1.0.20
-- checking for modules 'libva;libva-drm'
--   package 'libva' not found
--   package 'libva-drm' not found
-- Found JPEG: /usr/lib/arm-linux-gnueabihf/libjpeg.so  
-- Found Linux4Tegra 23.1.1
-- Could NOT find TegraJPEG (missing:  TegraJPEG_INCLUDE_DIRS TegraJPEG_WORKS) 
-- Performing Test TURBOJPEG_WORKS
-- Performing Test TURBOJPEG_WORKS - Success
-- Found TurboJPEG: /usr/lib/arm-linux-gnueabihf/libturbojpeg.so.0  
-- checking for module 'glfw3'
--   found glfw3, version 3.0.4
-- Looking for XOpenDisplay in /usr/lib/arm-linux-gnueabihf/libX11.so;/usr/lib/arm-linux-gnueabihf/libXext.so
-- Looking for XOpenDisplay in /usr/lib/arm-linux-gnueabihf/libX11.so;/usr/lib/arm-linux-gnueabihf/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/arm-linux-gnueabihf/libX11.so
-- Found OpenGL: /usr/lib/arm-linux-gnueabihf/libGL.so  
-- Looking for CL_VERSION_2_0
-- Looking for CL_VERSION_2_0 - not found
-- Looking for CL_VERSION_1_2
-- Looking for CL_VERSION_1_2 - not found
-- Looking for CL_VERSION_1_1
-- Looking for CL_VERSION_1_1 - not found
-- Looking for CL_VERSION_1_0
-- Looking for CL_VERSION_1_0 - not found
-- Could NOT find OpenCL (missing:  OpenCL_INCLUDE_DIR) 
CMake Warning (dev) in CMakeLists.txt:
  A logical block opening on the line

    /home/user_name/dev/libfreenect2/CMakeLists.txt:333 (IF)

  closes on the line

    /home/user_name/dev/libfreenect2/CMakeLists.txt:371 (ENDIF)

  with mis-matching arguments.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- RPATH set to /home/user_name/dev/libfreenect2/depends/libusb/lib
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Linking with these libraries: 
 /home/user_name/dev/libfreenect2/depends/libusb/lib/libusb-1.0.so
 pthread
 /usr/lib/arm-linux-gnueabihf/libturbojpeg.so.0
 /usr/local/lib/libglfw.so
 /usr/lib/arm-linux-gnueabihf/libGL.so
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.6") 
-- Configurating examples
-- Using in-tree freenect2 target
-- checking for module 'libopenni2'
--   package 'libopenni2' not found
-- Could NOT find OpenNI2 (missing:  OpenNI2_LIBRARY_DIR OpenNI2_INCLUDE_DIRS) 
-- Feature list:
--   CUDA    disablsed
--   CXX11    disabled
--   Examples    yes
--   OpenCL    no
--   OpenGL    yes
--   OpenNI2    no
--   TegraJPEG    no
--   Threading    tinythread
--   TurboJPEG    yes
--   VAAPI    no
--   VideoToolbox    no (Apple only)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user_name/dev/libfreenect2/build
```

When I run the upstream libfreenect2 with libusb patched I don't get any errors, but no windows show up on my machine. I'm running `ssh -X` then `DISPLAY=:0.0 ./bin/Protonect`

I'll have to look into Tegra more tomorrow, I haven't disabled it, but CMake can't find the right headers.

xlz, 2016-02-23:
A few things that don't look right to me:

* `A logical block opening on the line...    /home/user_name/dev/libfreenect2/CMakeLists.txt:333 (IF)`: what happened to `ENABLE_CUDA`?
* Is there something that made you unable to enable CUDA?
* `/usr/local/lib/libglfw.so` (you built libglfw3 and make install'd it)? The upstream libfreenect2 asks to use `download_debs_trusty.sh` to download glfw debs to install.

I don't know your setup. The viewer is supposed to show up on the monitor that is connected to the Jetson. If no monitor is connected to it, framerate information can also prove it works. I don't know if you still get nanHz or 0Hz, but `cmake -DENABLE_CXX11=ON` may solve it.

There are some new commits these two days, remember to update.

menlocaleb, 2016-02-25:
Sorry I misunderstood - the viewer does launch on the monitor connected to the TX1 when I run it over ssh -X with `DISPLAY=:0.0 ./bin/Protonect`. When the Kinect is using the PCI Express USB slot it works, but I get a lot of "skipping rgb packet!" and "[DepthPacketStreamParser] 3 packets were lost" or "skipping depth packet". I get frame rates for both color and depth though (18 and 60 Hz respectively).

When running the Kinect through the built-in USB port the viewer appears but is only a black screen. I get lots of "skipping rgb packet!" messages and a color frame rate, but nothing for depth. For both of these tests I built using `cmake -DENABLE_CXX11=ON` using the most up to date code from the repo.

I was able to get the TegraJPEG decoder to build, but I had to replace the the current section in CMakeLists.txt relating to TegraJPEG with (adapted from the jetsonhacks CMakeLists.txt):
```
SET(HAVE_TegraJPEG disabled)
IF(ENABLE_TEGRAJPEG AND EXISTS "/etc/nv_tegra_release")
  FIND_LIBRARY(TEGRA_JPEG_LIBRARY nvjpeg PATHS /usr/lib/arm-linux-gnueabihf/tegra NO_DEFAULT_PATH)
  IF(NOT TEGRA_JPEG_LIBRARY)
    SET(HAVE_TegraJPEG no)
    MESSAGE(FATAL_ERROR "Did not find Tegra JPEG library")
  ENDIF(NOT TEGRA_JPEG_LIBRARY)

  INCLUDE_DIRECTORIES("${MY_DIR}/depends")
  CHECK_SYMBOL_EXISTS(TEGRA_ACCELERATE "${MY_DIR}/depends/nv_headers/jconfig.h" TEGRA_JPEG_HEADERS)
  IF(NOT TEGRA_JPEG_HEADERS)
    MESSAGE(FATAL_ERROR "Did not find headers for Tegra JPEG library in ${MY_DIR}/depends/nv_headers")
  ENDIF(NOT TEGRA_JPEG_HEADERS)

  MESSAGE(STATUS "Found Tegra hardware accelerated JPEG support")
  SET(LIBFREENECT2_WITH_TEGRAJPEG_SUPPORT 1)
  SET(HAVE_TegraJPEG yes)

  LIST(APPEND SOURCES
    src/tegra_jpeg_rgb_packet_processor.cpp
  )
  LIST(APPEND LIBRARIES
    ${TEGRA_JPEG_LIBRARY}
  )
ENDIF()
```

I also had to download the tegra headers (adapted from [jetsonhacks install script](https://gist.github.com/jetsonhacks/1bd2830cc1a5790a6ac2#file-installlibfreenect2-sh)):
```
wget http://developer.download.nvidia.com/embedded/L4T/r23_Release_v1.0/source/gstjpeg_src.tbz2
tar -xvf gstjpeg_src.tbz2 gstjpeg_src/nv_headers
mv gstjpeg_src/nv_headers/ libfreenect2/depends/
# Cleanup the empty folder
rmdir gstjpeg_src/
```

I also disabled VA-API support to ensure the Tegra version would be run. However, after all this, when I run `./bin/Protonect' I get the following error:
```
[Error] [libjpeg_handle()] dlsym: ./bin/Protonect: undefined symbol: jpeg_std_error
```
This error happens in src/packet_pipeline.cpp when the tegra processor is tested. The `tegra->good()` check results in the else running, deleting the tegra processor and defaulting to the turbo processor.
```
RgbPacketProcessor *tegra = new TegraJpegRgbPacketProcessor();
if (tegra->good())
    return tegra;
else {
    delete tegra;
}
return new TurboJpegRgbPacketProcessor();
```

I'm going to continue to debug the Tegra and CUDA optimizations tomorrow. Do you notice anything wrong that you know how to fix?

xlz, 2016-02-25:
> I had to replace the the current section in CMakeLists.txt relating to TegraJPEG

This will not work. And I'm unable to support the old CMake code. You can't simply mix the old code with new code and expect it to work.

If you take a look, https://github.com/OpenKinect/libfreenect2/blob/master/cmake_modules/FindTegraJPEG.cmake#L38, CMake supposedly downloads the headers to `depends/gstjpeg/gstjpeg_src.tbz2` and extracts this tarball into `build/gstjpeg_src/nv_headers`.

Now the problem is somehow it fails to extract the headers. Some very trivial error happens here. It just needs someone with TX1 to take a look.

xlz, 2016-02-25:
OK, I know the problem now.

xlz, 2016-02-25:
Try this tree https://github.com/xlz/libfreenect2/tree/tegra-url

marketto89, 2016-02-25:
@xlz, sorry for not replying to #480 I have been busy.

I just tried your tree tegra-url and I confirm that now cmake downloads and extract the tarball.

Here is the final output:

```
ubuntu@tegra-ubuntu:~/workspace/libfreenect2/build$ cmake .. -DCMAKE_INSTALL_PREFIX=$HOME/freenect2
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- using tinythread as threading library
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- checking for module 'libusb-1.0>=1.0.20'
--   found libusb-1.0, version 1.0.20
-- checking for modules 'libva;libva-drm'
--   package 'libva' not found
--   package 'libva-drm' not found
-- Found JPEG: /usr/lib/arm-linux-gnueabihf/libjpeg.so  
-- Found Linux4Tegra 23.1.1
-- Downloading gstjpeg_src.tbz2 to /home/ubuntu/workspace/libfreenect2/depends...
-- Performing Test TegraJPEG_WORKS
-- Performing Test TegraJPEG_WORKS - Success
-- Found TegraJPEG: /usr/lib/arm-linux-gnueabihf/tegra/libnvjpeg.so  
-- Performing Test TURBOJPEG_WORKS
-- Performing Test TURBOJPEG_WORKS - Success
-- Found TurboJPEG: /usr/lib/arm-linux-gnueabihf/libturbojpeg.so.0  
-- checking for module 'glfw3'
--   found glfw3, version 3.1.2
-- Looking for XOpenDisplay in /usr/lib/arm-linux-gnueabihf/libX11.so;/usr/lib/arm-linux-gnueabihf/libXext.so
-- Looking for XOpenDisplay in /usr/lib/arm-linux-gnueabihf/libX11.so;/usr/lib/arm-linux-gnueabihf/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/arm-linux-gnueabihf/libX11.so
-- Found OpenGL: /usr/lib/arm-linux-gnueabihf/libGL.so  
-- Looking for CL_VERSION_2_0
-- Looking for CL_VERSION_2_0 - not found
-- Looking for CL_VERSION_1_2
-- Looking for CL_VERSION_1_2 - found
-- Found OpenCL: /usr/lib/arm-linux-gnueabihf/libOpenCL.so (found version "1.2") 
-- Performing Test OpenCL_C_WORKS
-- Performing Test OpenCL_C_WORKS - Success
-- Performing Test OpenCL_CXX_WORKS
-- Performing Test OpenCL_CXX_WORKS - Success
-- Found CUDA: /usr/local/cuda (found version "6.5") 
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Linking with these libraries: 
 /usr/lib/arm-linux-gnueabihf/libusb-1.0.so
 pthread
 dl
 /usr/lib/arm-linux-gnueabihf/libturbojpeg.so.0
 /usr/lib/arm-linux-gnueabihf/libglfw.so
 /usr/lib/arm-linux-gnueabihf/libGL.so
 /usr/lib/arm-linux-gnueabihf/libOpenCL.so
 /usr/local/cuda/lib/libcudart.so
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.6") 
-- Configurating examples
-- Using in-tree freenect2 target
-- checking for module 'libopenni2'
--   package 'libopenni2' not found
-- Could NOT find OpenNI2 (missing:  OpenNI2_LIBRARY_DIR OpenNI2_INCLUDE_DIRS) 
-- Feature list:
--   CUDA    yes
--   CXX11    disabled
--   Examples    yes
--   OpenCL    yes
--   OpenGL    yes
--   OpenNI2    no
--   TegraJPEG    yes
--   Threading    tinythread
--   TurboJPEG    yes
--   VAAPI    no
--   VideoToolbox    no (Apple only)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/workspace/libfreenect2/build
ubuntu@tegra-ubuntu:~/workspace/libfreenect2/build$
```

marketto89, 2016-02-25:
After some reboots it seems that TegraJpeg started to work, but I still can't see any depth information from the output.

```
ubuntu@tegra-ubuntu:~/workspace/libfreenect2_xlz/build$ ./bin/Protonect cuda -noviewer
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [CudaDepthPacketProcessorImpl] device 0: GM20B @ 72MHz Memory 3853MB
[Info] [CudaDepthPacketProcessorImpl] selected device 0
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 6 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:10 with serial 010354144947
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9855
[Debug] [Freenect2DeviceImpl] status 0x090000: 9855
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 010354144947
device firmware: 2.3.3913.0
[Info] [TegraJpegRgbPacketProcessor] avg. time: 5.3481ms -> ~186.982Hz
[Info] [TegraJpegRgbPacketProcessor] avg. time: 5.29576ms -> ~188.83Hz
[Info] [TegraJpegRgbPacketProcessor] avg. time: 5.19811ms -> ~192.378Hz
[Info] [TegraJpegRgbPacketProcessor] avg. time: 5.21103ms -> ~191.9Hz
[Info] [TegraJpegRgbPacketProcessor] avg. time: 4.97272ms -> ~201.097Hz
[Info] [TegraJpegRgbPacketProcessor] avg. time: 4.98701ms -> ~200.521Hz
[Info] [TegraJpegRgbPacketProcessor] avg. time: 4.91539ms -> ~203.443Hz
[Info] [TegraJpegRgbPacketProcessor] avg. time: 4.93047ms -> ~202.82Hz
[Info] [TegraJpegRgbPacketProcessor] avg. time: 4.93109ms -> ~202.795Hz
[Info] [TegraJpegRgbPacketProcessor] avg. time: 5.03911ms -> ~198.448Hz
Killed
```

@menlocaleb, how did you set your TX1 for max performance? My CPU speed is 1.9 GHz, the GPU clock speed is 998.4 MHz and the GPU memory clock is 1.6 GHz. Do these numbers make sense to you?

xlz, 2016-02-25:
@marketto89 Your setup is different from @menlocaleb

@menlocaleb has been using a uPD720202 extension card with *patched libusb*.

Right now both of you could not get libfreenect2 to work with the built-in USB 3 port.

Regarding the built-in USB 3 port, there are several things to verify:

(See the logs here https://github.com/OpenKinect/libfreenect2/issues/480#issuecomment-160745936)
* Does `LIBUSB_DEBUG=4 ./bin/Protonect` show `[handle_iso_completion] low-level USB error -18`?
* What does `dmesg` show?
* Does patching and building libusb 1.0.20 (instructions: https://github.com/OpenKinect/libfreenect2/issues/581#issuecomment-187454035) from source make it work?

marketto89, 2016-02-25:
> [handle_iso_completion] low-level USB error -18

Yes I see many of these errors!
> What does dmesg show?

Here is the tail of my dmesg:
```
[17471.437320] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.471820] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.477565] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.501194] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.535944] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.539829] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.566815] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.569314] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.605003] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.605137] smmu_dump_pagetable(): fault_address=0x0000000080200000 pa=0x0000000000000000 bytes=1000 #pte=492 in L2
[17471.605178] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.605181] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.605422] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.605425] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.605665] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.605668] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.605881] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.605926] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.605929] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.617827] mc-err: [mcerr] (20) csw_xusb_hostw: EMEM decode error on PDE or PTE entry
[17471.627947] mc-err: [mcerr]   status = 0x6001004b; addr = 0x80200000
[17471.635432] mc-err: [mcerr]   secure: no, access-type: write, SMMU fault: nr-nw-s
[17471.636089] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.649069] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.700755] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.700894] smmu_dump_pagetable(): fault_address=0x0000000080280000 pa=0x0000000000000000 bytes=1000 #pte=412 in L2
[17471.700926] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.700929] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.701175] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.701178] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.701428] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.701431] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.701678] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.701681] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.701684] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.703533] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.703674] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.703678] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.703924] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.703927] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.704175] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.704179] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.704426] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.704429] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.704432] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.706209] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.710710] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.713596] mc-err: [mcerr] (20) csw_xusb_hostw: EMEM decode error on PDE or PTE entry
[17471.723732] mc-err: [mcerr]   status = 0x6001004b; addr = 0x80280000
[17471.731221] mc-err: [mcerr]   secure: no, access-type: write, SMMU fault: nr-nw-s
[17471.732863] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.737584] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.866825] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.871447] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.877564] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.905003] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.905145] smmu_dump_pagetable(): fault_address=0x0000000080180000 pa=0x0000000000000000 bytes=1000 #pte=413 in L2
[17471.905180] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[17471.905184] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 8 with no TDs queued?
[... cut ...]
[17610.582148] usb 2-2.1: Parent hub missing LPM exit latency info.  Power management will be impacted.
[17610.605448] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 2 with no TDs queued?
[17610.701931] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 2 with no TDs queued?
[17610.710400] tegra-xhci tegra-xhci: WARN Event TRB for slot 3 ep 2 with no TDs queued?
```
> Does patching and building libusb 1.0.20 (instructions: #581 (comment)) from source make it work?

I will inform you if this fix works!

marketto89, 2016-02-25:
> Does patching and building libusb 1.0.20 (instructions: #581 (comment)) from source make it work?

I followed the instructions and verified that Protonect were linked to the patched libusb but it is still not working:

```
ubuntu@tegra-ubuntu:~/workspace/libfreenect2_xlz/build$ DISPLAY=:0.0 ./bin/Protonect cuda  
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [CudaDepthPacketProcessorImpl] device 0: GM20B @ 72MHz Memory 3853MB
[Info] [CudaDepthPacketProcessorImpl] selected device 0
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 6 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:4 with serial 010354144947
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9739
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 010354144947
device firmware: 2.3.3913.0
[Info] [TegraJpegRgbPacketProcessor] avg. time: 5.56633ms -> ~179.651Hz
[Info] [TegraJpegRgbPacketProcessor] avg. time: 5.32445ms -> ~187.813Hz
[Info] [TegraJpegRgbPacketProcessor] avg. time: 5.17759ms -> ~193.14Hz
[Info] [TegraJpegRgbPacketProcessor] avg. time: 5.12778ms -> ~195.016Hz
[Info] [TegraJpegRgbPacketProcessor] avg. time: 5.15028ms -> ~194.164Hz
Killed
```

xlz, 2016-02-25:
OK this confirms it's the same USB kernel driver issue as last time. There is no solution to that yet. I will try to contact Nvidia developers about this bug.

In the meantime, I think the best workaround is to acquire a uPD720202 extension card as menlocaleb did.

xlz, 2016-02-25:
The thread is here https://devtalk.nvidia.com/default/topic/919354/jetson-tx1/usb-3-transfer-failures/

marketto89, 2016-02-29:
Given the answers, is there anything we can do to test the built-in USB 3.0 port?

xlz, 2016-03-02:
@marketto89 I think the easiest thing to do right now is get a uPD720202 extension card.

xlz, 2016-03-11:
I have obtained a TX1 and I have reproduced all the errors you've had.

I'm working on it, but this one is really hard.

menlocaleb, 2016-03-16:
@xlz sorry for not responding on this thread recently. I was just looking into this problem again and went to try your new branch  https://github.com/xlz/libfreenect2/tree/tegra-url but it looks like it's deleted. Was the problem just changing a url for the jpeg headers to be compatible with the X1?

Is there anything I should try to help, or should I wait for you to fix on your X1?

xlz, 2016-03-16:
@menlocaleb I thought your had it working https://github.com/OpenKinect/libfreenect2/issues/581#issuecomment-187462906, what is the new problem?

tegra-url is merged into upstream so it's deleted.

I have put the recommendation of "uPD720202 extension card with libusb 49152" into docs. I can't fix this issue because it's a problem with Nvidia's tegra kernel driver.

menlocaleb, 2016-03-16:
Sorry for not being specific: I can run the upstream libfreenect, I was working on fixing the tegra acceleration, which it sounds like should work now that you've checked in the fix.

Last time I tried the iai_kinect2 code was crashing, but maybe after the tegra acceleration fix and rebuilding it will work. If it doesn't I'll let you know.

RedAlice64, 2016-03-26:
Hi,
I am following the instructions and I am still getting streams of image data too short and subpacket too large.

Here's the message from 
`LIBUSB_DEBUG=4 ./bin/Protonect cuda | grep libusb`

`[10.821083] [00004f0a] libusb: debug [submit_iso_transfer] need 1 6144k URBs for transfer
[10.821272] [00004f0a] libusb: debug [reap_for_handle] urb type=0 status=0 transferred=135168
[10.821299] [00004f0a] libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 1/1
[10.821321] [00004f0a] libusb: debug [handle_iso_completion] last URB in transfer --> complete!
[10.821343] [00004f0a] libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 1000ms
[10.821368] [00004f0a] libusb: debug [usbi_handle_transfer_completion] transfer 0x3fcb48 has callback 0xf7717dd9
[10.821432] [00004f0a] libusb: debug [libusb_submit_transfer] transfer 0x3fcb48
[10.821462] [00004f0a] libusb: debug [submit_iso_transfer] need 1 6144k URBs for transfer
[10.821648] [00004f0a] libusb: debug [reap_for_handle] urb type=0 status=0 transferred=135168
[10.821674] [00004f0a] libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 1/1
[10.821696] [00004f0a] libusb: debug [handle_iso_completion] last URB in transfer --> complete!
[10.821718] [00004f0a] libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 1000ms
[10.821743] [00004f0a] libusb: debug [usbi_handle_transfer_completion] transfer 0x3fb948 has callback 0xf7717dd9
[10.821811] [00004f0a] libusb: debug [libusb_submit_transfer] transfer 0x3fb948
[10.821842] [00004f0a] libusb: debug [submit_iso_transfer] need 1 6144k URBs for transfer
[10.822034] [00004f0a] libusb: debug [reap_for_handle] urb type=0 status=0 transferred=135168
[10.822064] [00004f0a] libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 1/1
[10.822085] [00004f0a] libusb: debug [handle_iso_completion] last URB in transfer --> complete!
[10.822106] [00004f0a] libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 1000ms
[10.822132] [00004f0a] libusb: debug [usbi_handle_transfer_completion] transfer 0x3fcc48 has callback 0xf7717dd9
[10.822201] [00004f0a] libusb: debug [libusb_submit_transfer] transfer 0x3fcc48
[10.822228] [00004f0a] libusb: debug [submit_iso_transfer] need 1 6144k URBs for transfer
[10.822421] [00004f0a] libusb: debug [reap_for_handle] urb type=0 status=0 transferred=135168
[10.822448] [00004f0a] libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 1/1
[10.822469] [00004f0a] libusb: debug [handle_iso_completion] last URB in transfer --> complete!
[10.822491] [00004f0a] libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 1000ms
[10.822516] [00004f0a] libusb: debug [usbi_handle_transfer_completion] transfer 0x3fcd48 has callback 0xf7717dd9
[10.822584] [00004f0a] libusb: debug [libusb_submit_transfer] transfer 0x3fcd48
[10.822610] [00004f0a] libusb: debug [submit_iso_transfer] need 1 6144k URBs for transfer
[10.822806] [00004f0a] libusb: debug [reap_for_handle] urb type=0 status=0 transferred=135168
[10.822833] [00004f0a] libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 1/1
[10.822856] [00004f0a] libusb: debug [handle_iso_completion] last URB in transfer --> complete!
[10.822877] [00004f0a] libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 1000ms
[10.822903] [00004f0a] libusb: debug [usbi_handle_transfer_completion] transfer 0x3fce48 has callback 0xf7717dd9
[10.822969] [00004f0a] libusb: debug [libusb_submit_transfer] transfer 0x3fce48
[10.822997] [00004f0a] libusb: debug [submit_iso_transfer] need 1 6144k URBs for transfer
[10.823196] [00004f0a] libusb: debug [reap_for_handle] urb type=0 status=0 transferred=135168
[10.823224] [00004f0a] libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 1/1
[10.823235] [00004f0a] libusb: debug [handle_iso_completion] last URB in transfer --> complete!
[10.823246] [00004f0a] libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 1000ms
[10.823259] [00004f0a] libusb: debug [usbi_handle_transfer_completion] transfer 0x3fba48 has callback 0xf7717dd9
[10.823335] [00004f0a] libusb: debug [libusb_submit_transfer] transfer 0x3fba48
[10.823369] [00004f0a] libusb: debug [submit_iso_transfer] need 1 6144k URBs for transfer
[10.823566] [00004f0a] libusb: debug [reap_for_handle] urb type=0 status=0 transferred=135168
[10.823597] [00004f0a] libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 1/1
[10.823622] [00004f0a] libusb: debug [handle_iso_completion] last URB in transfer --> complete!
[10.823646] [00004f0a] libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 1000ms
[10.823672] [00004f0a] libusb: debug [usbi_handle_transfer_completion] transfer 0x3fcf48 has callback 0xf7717dd9
[10.823743] [00004f0a] libusb: debug [libusb_submit_transfer] transfer 0x3fcf48
[10.823776] [00004f0a] libusb: debug [submit_iso_transfer] need 1 6144k URBs for transfer
[10.823972] [00004f0a] libusb: debug [reap_for_handle] urb type=0 status=0 transferred=202752
[10.824004] [00004f0a] libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 1/1
[10.824030] [00004f0a] libusb: debug [handle_iso_completion] last URB in transfer --> complete!
[10.824054] [00004f0a] libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 1000ms
[... cut ...]
[10.900691] [00004f0a] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 16384
[10.900840] [00004f0a] libusb: debug [reap_for_handle] urb type=0 status=0 transferred=264856
[10.900861] [00004f0a] libusb: debug [handle_iso_completion] handling completion status 0 of iso urb 1/1
[10.900876] [00004f0a] libusb: debug [handle_iso_completion] last URB in transfer --> complete!
[10.900891] [00004f0a] libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 1000ms
`

nznobody, 2016-05-05:
Hello, I have read through a majority of this and other treads related to the Tegra TX1 and the Kinect v2 on the native built in USB3.0 port. has there been any progress?

I have seen the progress at [https://devtalk.nvidia.com/default/topic/919354/jetson-tx1/usb-3-transfer-failures/post/4836312/#4836312](url)

I can reproduce all the known problems resulting in no viewer output. Attached are reproduced logs for consistencies sake.

[other_logs.txt](https://github.com/OpenKinect/libfreenect2/files/249918/other_logs.txt)
[Protonect_Log_DEBUG4.txt](https://github.com/OpenKinect/libfreenect2/files/249919/Protonect_Log_DEBUG4.txt)
[dmesg.txt](https://github.com/OpenKinect/libfreenect2/files/249917/dmesg.txt)

Any progress apart from buying the suggested PCIe card?

xlz, 2016-05-05:
Nvidia's kay was looking at the issue but there is no update. So, no progress.

jetsonhacks, 2016-06-11:
NVIDIA has posted a USB firmware patch that appears to resolve the issue:

[https://devtalk.nvidia.com/default/topic/919354/jetson-tx1/usb-3-transfer-failures/post/4899105/#4899105]

The data appears to be transferred correctly, though the RGB color stream appears to have tint issues, suggesting a possible color decoding problem.

xlz, 2016-06-11:
Cool news.

@Jetsonhacks how does the tint look? Can you post a screen shot?

jetsonhacks, 2016-06-11:
I'm a monster, but one of a different color. The shirt is black as reference, painting is brown/gold/orange.

[image]

It's been several months, but I seem to recall that the Jetson TK1 has the same issue with coloring.

xlz, 2016-06-12:
Oh, this is because Jetson hardware JPEG decoder outputs image in RGBA format but the viewer is expecting BGRA. You can check this by checking frame->format.

jetsonhacks, 2016-06-13:
Yes, converting to BGRA solves the issue. Thank you very much for your help.

Abbyls, 2016-08-26:
1

Abbyls, 2016-08-26:
@menlocaleb @xlz 
Hi, I used the kinect2 with TX1 and I met the same issue mentioned above! 
As:
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Info] [OpenGLDepthPacketProcessor] avg. time: 21.9789ms -> ~45.4983Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 41.8277ms -> ~23.9076Hz
[Info] [DepthPacketStreamParser] 5 packets were lost

Abbyls, 2016-08-26:
@xlz @menlocaleb 
Hi, I used the kinect2 with TX1 and I met the same issue mentioned above! 
As:
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Info] [OpenGLDepthPacketProcessor] avg. time: 21.9789ms -> ~45.4983Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 41.8277ms -> ~23.9076Hz
[Info] [DepthPacketStreamParser] 5 packets were lost
but i didn't find the solutions.
Have you solved it ? Can you tell me how?

xlz, 2016-12-05:
I believe this has been resolved by newer TX1 firmware.

closed 2016-12-05 by xlz