OpenKinect/libfreenect2 #582: can't compile on old Mac OS X (10.9, VideoToolbox)

hanyazou, 2016-02-19:
On clean installed Mac OS X 10.9, I can't compile libfreenect2 because of an error about VideoToolbox.
(v0.1.1 was compiled successfully)  

>/Users/hanyazou/github/libfreenect2/src/vt_rgb_packet_processor.cpp:150:5: error: use of undeclared identifier
      'CMSampleBufferCreateReady'
    CMSampleBufferCreateReady(
    ^
1 error generated.

What version of OS X should be supported by the libfreenect2?
Shall I add 'IF (DARWIN_VERSION GREATER ... ' in CMakeList.txt? 

detail log
------
bash-3.2$ uname -a
Darwin iMac.local 13.4.0 Darwin Kernel Version 13.4.0: Sun Aug 17 19:50:11 PDT 2014; root:xnu-2422.115.4~1/RELEASE_X86_64 x86_64
bash-3.2$ system_profiler SPSoftwareDataType
Software:

    System Software Overview:

      System Version: OS X 10.9.5 (13F34)
      Kernel Version: Darwin 13.4.0
      Boot Volume: OS X 10.9 Mavericks
      Boot Mode: Normal
      Computer Name: iMac
      User Name: hanyazou
      Secure Virtual Memory: Enabled
      Time since boot: 7:19

bash-3.2$ brew list
cmake           glfw3           libusb          openssl         wget
git             jpeg-turbo      openni2         pkg-config
bash-3.2$ git checkout master
Previous HEAD position was a819be4... tools: Add mkcontrib.py
Switched to branch 'master'
Your branch is up-to-date with 'origin/master'.
bash-3.2$ mkdir build
bash-3.2$ cd build
bash-3.2$ cmake ..
-- The C compiler identification is AppleClang 6.0.0.6000057
-- The CXX compiler identification is AppleClang 6.0.0.6000057
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- using tinythread as threading library
-- Found PkgConfig: /usr/local/bin/pkg-config (found version "0.29")
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.20
-- Checking for modules 'libva;libva-drm'
--   Package 'libva' not found
--   Package 'libva-drm' not found
-- Could NOT find JPEG (missing:  JPEG_LIBRARY JPEG_INCLUDE_DIR)
-- Could NOT find TegraJPEG (missing:  TegraJPEG_LIBRARIES TegraJPEG_INCLUDE_DIRS TegraJPEG_L4T_OK TegraJPEG_DRIVER_OK TegraJPEG_WORKS)
-- Performing Test TURBOJPEG_WORKS
-- Performing Test TURBOJPEG_WORKS - Success
-- Found TurboJPEG: /usr/local/opt/jpeg-turbo/lib/libturbojpeg.dylib
-- Checking for module 'glfw3'
--   Found glfw3, version 3.1.2
-- Found OpenGL: /System/Library/Frameworks/OpenGL.framework
-- Looking for CL_VERSION_2_0
-- Looking for CL_VERSION_2_0 - not found
-- Looking for CL_VERSION_1_2
-- Looking for CL_VERSION_1_2 - not found
-- Looking for CL_VERSION_1_1
-- Looking for CL_VERSION_1_1 - not found
-- Looking for CL_VERSION_1_0
-- Looking for CL_VERSION_1_0 - not found
-- Found OpenCL: /System/Library/Frameworks/OpenCL.framework
CUDA_TOOLKIT_ROOT_DIR not found or specified
-- Could NOT find CUDA (missing:  CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY)
-- RPATH set to /usr/local/Cellar/libusb/1.0.20/lib
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Linking with these libraries:
 /usr/local/Cellar/libusb/1.0.20/lib/libusb-1.0.dylib
 pthread
 /System/Library/Frameworks/VideoToolbox.framework
 /System/Library/Frameworks/CoreFoundation.framework
 /System/Library/Frameworks/CoreMedia.framework
 /System/Library/Frameworks/CoreVideo.framework
 /usr/local/opt/jpeg-turbo/lib/libturbojpeg.dylib
 /usr/local/Cellar/glfw3/3.1.2/lib/libglfw3.dylib
 /System/Library/Frameworks/OpenGL.framework
 /System/Library/Frameworks/OpenCL.framework
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE)
-- Configurating examples
-- Using in-tree freenect2 target
-- Checking for module 'libopenni2'
--   Package 'libopenni2' not found
-- Found OpenNI2: /usr/local/lib/ni2
-- Feature list:
--   CUDA    no
--   CXX11    disabled
--   Examples    yes
--   FLAG_SEARCH_PATHS_FIRST    1
--   OpenCL    yes
--   OpenGL    yes
--   OpenNI2    yes
--   TegraJPEG    no
--   Threading    tinythread
--   TurboJPEG    yes
--   VAAPI    no
--   VideoToolbox    yes
-- Configuring done
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   freenect2
   freenect2-openni2

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /Users/hanyazou/github/libfreenect2/build
bash-3.2$ make
Scanning dependencies of target generate_resources_tool
[  2%] Building CXX object CMakeFiles/generate_resources_tool.dir/tools/generate_resources.cpp.o
[  5%] Linking CXX executable bin/generate_resources_tool
[  5%] Built target generate_resources_tool
[  7%] Generating resources.inc.h
Scanning dependencies of target freenect2
[ 10%] Building CXX object CMakeFiles/freenect2.dir/src/transfer_pool.cpp.o
[ 12%] Building CXX object CMakeFiles/freenect2.dir/src/event_loop.cpp.o
[ 15%] Building CXX object CMakeFiles/freenect2.dir/src/usb_control.cpp.o
[ 17%] Building CXX object CMakeFiles/freenect2.dir/src/allocator.cpp.o
[ 20%] Building CXX object CMakeFiles/freenect2.dir/src/frame_listener_impl.cpp.o
[ 23%] Building CXX object CMakeFiles/freenect2.dir/src/packet_pipeline.cpp.o
[ 25%] Building CXX object CMakeFiles/freenect2.dir/src/rgb_packet_stream_parser.cpp.o
[ 28%] Building CXX object CMakeFiles/freenect2.dir/src/rgb_packet_processor.cpp.o
[ 30%] Building CXX object CMakeFiles/freenect2.dir/src/depth_packet_stream_parser.cpp.o
[ 33%] Building CXX object CMakeFiles/freenect2.dir/src/depth_packet_processor.cpp.o
[ 35%] Building CXX object CMakeFiles/freenect2.dir/src/cpu_depth_packet_processor.cpp.o
[ 38%] Building CXX object CMakeFiles/freenect2.dir/src/resource.cpp.o
[ 41%] Building CXX object CMakeFiles/freenect2.dir/src/command_transaction.cpp.o
[ 43%] Building CXX object CMakeFiles/freenect2.dir/src/registration.cpp.o
[ 46%] Building CXX object CMakeFiles/freenect2.dir/src/logging.cpp.o
[ 48%] Building CXX object CMakeFiles/freenect2.dir/src/libfreenect2.cpp.o
[ 51%] Building CXX object CMakeFiles/freenect2.dir/src/tinythread/tinythread.cpp.o
[ 53%] Building CXX object CMakeFiles/freenect2.dir/src/vt_rgb_packet_processor.cpp.o
/Users/hanyazou/github/libfreenect2/src/vt_rgb_packet_processor.cpp:150:5: error: use of undeclared identifier
      'CMSampleBufferCreateReady'
    CMSampleBufferCreateReady(
    ^
1 error generated.
make[2]: *** [CMakeFiles/freenect2.dir/src/vt_rgb_packet_processor.cpp.o] Error 1
make[1]: *** [CMakeFiles/freenect2.dir/all] Error 2
make: *** [all] Error 2

xlz, 2016-02-19:
@fran6co bug report about VideoToolbox

xlz, 2016-02-20:
https://developer.apple.com/library/mac/releasenotes/General/APIDiffsMacOSX10_10_3/frameworks/CoreMedia.html

> Added CMSampleBufferCreateReady

fran6co, 2016-02-20:
Fixed in https://github.com/OpenKinect/libfreenect2/pull/583

closed 2016-02-20 by xlz