OpenKinect/libfreenect2 #584: Tegra XHCI stability

xlz, 2016-02-21:
```
for i in {1..100}; do
echo ================ $i
lsusb -v 2>/dev/null| | grep -i nui
DISPLAY=:0 ./bin/Protonect gl --frames 1 || break
done
```

When `i=81`, dmesg:
```
[ 1851.612674] tegra-xhci tegra-xhci: xHCI host not responding to stop endpoint command.
[ 1851.612893] tegra-xhci tegra-xhci: Assuming host is dying, halting host.
[ 1851.614590] tegra-xhci tegra-xhci: HC died; cleaning up
[ 1851.620989] tegra-xhci tegra-xhci: Timeout while waiting for configure endpoint command
[ 1851.621153] tegra-xhci tegra-xhci: Abort the command ring, but the xHCI is dead.
```

Originally reported in #539.

hanyazou, 2016-02-21:
@xlz I've tried rebooting and running my script several times. I found that my scripts could run i = 1 to 200~300 after reboot. Then, the kernel starts to kill the Protonect process by OOM killer. This means the kernel (may be the tegra-xhci) leaks some memory.  I couldn't reproduce 'tegra-xhci: xHCI host not responding to stop endpoint command.' But I can guess that the kernel and the driver couldn't use enough memory at that time. Attached dmesg log contains various memory warnings and errors.

[tegra-ubuntu-dmesg.txt](https://github.com/OpenKinect/libfreenect2/files/139506/tegra-ubuntu-dmesg.txt)

hanyazou, 2016-02-22:
JFYI:
I've purchased mini PCI express USB 3.0 card and tried to run the Protonect. But it didn't work for now.
(MiniPCI Express USB3.0 Renesas μPD720200 http://www.amazon.co.jp/gp/product/B009KKLRM6/)

$ dmesg | grep xhci
[ 3.722136] xhci_hcd 0000:01:00.0: xHCI Host Controller
[ 3.722160] xhci_hcd 0000:01:00.0: new USB bus registered, assigned bus number 1
[ 4.105962] usb usb1: Manufacturer: Linux 3.10.40-gdacac96 xhci_hcd
[ 4.106337] xHCI xhci_add_endpoint called for root hub
[ 4.106341] xHCI xhci_check_bandwidth called for root hub
[ 4.106809] xhci_hcd 0000:01:00.0: xHCI Host Controller
[ 4.106825] xhci_hcd 0000:01:00.0: new USB bus registered, assigned bus number 2
[ 4.106990] usb usb2: Manufacturer: Linux 3.10.40-gdacac96 xhci_hcd
[ 4.107332] xHCI xhci_add_endpoint called for root hub
[ 4.107335] xHCI xhci_check_bandwidth called for root hub

[ 10.486878] tegra-xhci tegra-xhci: Firmware File: tegra_xusb_firmware (134656 Bytes)
[ 10.487340] tegra-xhci tegra-xhci: Firmware DMA Memory: dma 0x800c0000 mapped 0xf0357000 (134656 Bytes)
[ 10.489331] tegra-xhci tegra-xhci: XUSB device id = 0xfa3 (T124+)
[ 10.542500] tegra-xhci tegra-xhci: Firmware timestamp: 2014-09-16 02:10:07 UTC, Falcon state 0x20
[ 10.542621] tegra-xhci tegra-xhci: Nvidia xHCI Host Controller
[ 10.542654] tegra-xhci tegra-xhci: new USB bus registered, assigned bus number 3
[ 10.543561] tegra-xhci tegra-xhci: irq 71, io mem 0x70090000
[ 10.543770] usb usb3: Manufacturer: Linux 3.10.40-gdacac96 tegra-xhci
[ 10.543787] usb usb3: SerialNumber: tegra-xhci
[ 10.544335] xHCI xhci_add_endpoint called for root hub
[ 10.544348] xHCI xhci_check_bandwidth called for root hub
[ 10.555178] tegra-xhci tegra-xhci: Nvidia xHCI Host Controller
[ 10.555213] tegra-xhci tegra-xhci: new USB bus registered, assigned bus number 4
[ 10.555464] usb usb4: Manufacturer: Linux 3.10.40-gdacac96 tegra-xhci
[ 10.555475] usb usb4: SerialNumber: tegra-xhci
[ 10.556500] xHCI xhci_add_endpoint called for root hub
[ 10.556515] xHCI xhci_check_bandwidth called for root hub

[ 46.267863] usb 1-2: new high-speed USB device number 2 using xhci_hcd
[ 46.299601] usb 1-2: New USB device found, idVendor=045e, idProduct=02d9
[ 46.299823] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 46.303627] usb 1-2: Product: NuiSensor Adaptor
[ 46.303795] usb 1-2: Manufacturer: Microsoft Corporation
[ 46.339625] hub 1-2:1.0: USB hub found
[ 46.341368] hub 1-2:1.0: 1 port detected
[ 46.886994] usb 2-2: new SuperSpeed USB device number 2 using xhci_hcd
[ 48.220287] usb 2-2: Parent hub missing LPM exit latency info. Power management will be impacted.
[ 49.875021] usb 2-2: New USB device found, idVendor=045e, idProduct=02d9
[ 49.875675] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 49.875849] usb 2-2: Product: NuiSensor Adaptor
[ 49.875991] usb 2-2: Manufacturer: Microsoft Corporation
[ 49.898454] hub 2-2:1.0: USB hub found
[ 49.898852] hub 2-2:1.0: 1 port detected
[ 50.370112] usb 2-2.1: new SuperSpeed USB device number 3 using xhci_hcd
[ 50.383952] usb 2-2.1: Parent hub missing LPM exit latency info. Power management will be impacted.
[ 50.388005] usb 2-2.1: New USB device found, idVendor=045e, idProduct=02c4
[ 50.388015] usb 2-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[ 50.388023] usb 2-2.1: Product: Xbox NUI Sensor
[ 50.388029] usb 2-2.1: Manufacturer: Microsoft
[ 50.388035] usb 2-2.1: SerialNumber: 005697142447

$ DISPLAY=:0 ./bin/Protonect gl
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
[-noviewer] [-norgb | -nodepth] [-help] [-version]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 8 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:5 with serial 005697142447
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[... 79 lines cut ...]
$ dmesg
[ 217.168976] xhci_hcd 0000:01:00.0: ERROR Transfer event TRB DMA ptr not part of current TD
[ 217.177222] xhci_hcd 0000:01:00.0: ERROR Transfer event TRB DMA ptr not part of current TD
[ 217.185498] xhci_hcd 0000:01:00.0: ERROR Transfer event TRB DMA ptr not part of current TD
[ 217.193746] xhci_hcd 0000:01:00.0: ERROR Transfer event TRB DMA ptr not part of current TD
[ 217.201992] xhci_hcd 0000:01:00.0: ERROR Transfer event TRB DMA ptr not part of current TD
[ 217.210237] xhci_hcd 0000:01:00.0: ERROR Transfer event TRB DMA ptr not part of current TD
[ 217.218482] xhci_hcd 0000:01:00.0: ERROR Transfer event TRB DMA ptr not part of current TD
[ 217.226727] xhci_hcd 0000:01:00.0: ERROR Transfer event TRB DMA ptr not part of current TD
[ 217.234972] xhci_hcd 0000:01:00.0: ERROR Transfer event TRB DMA ptr not part of current TD
[ 217.243216] xhci_hcd 0000:01:00.0: ERROR Transfer event TRB DMA ptr not part of current TD
[ 217.251602] ERROR: 256 KiB atomic DMA coherent pool is too small!
[ 217.251602] Please increase it with coherent_pool= kernel parameter!
[ 217.264039] xhci_hcd 0000:01:00.0: Ring expansion failed
[ 217.269608] usb 2-2.1: usbfs: usb_submit_urb returned -12
[ 217.269909] xhci_hcd 0000:01:00.0: Ring expansion failed
[ 217.275259] usb 2-2.1: usbfs: usb_submit_urb returned -12
[ 217.275506] xhci_hcd 0000:01:00.0: Ring expansion failed
[ 217.280870] usb 2-2.1: usbfs: usb_submit_urb returned -12
[ 217.281123] xhci_hcd 0000:01:00.0: Ring expansion failed
[ 217.286462] usb 2-2.1: usbfs: usb_submit_urb returned -12
[ 217.286706] xhci_hcd 0000:01:00.0: Ring expansion failed
[ 217.292047] usb 2-2.1: usbfs: usb_submit_urb returned -12
[ 217.292333] xhci_hcd 0000:01:00.0: Ring expansion failed
[ 217.297667] usb 2-2.1: usbfs: usb_submit_urb returned -12

xlz, 2016-02-22:
https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting#jetson-tk1-issues

hanyazou, 2016-02-22:
@xlz do you mean 'External USB 3 extension cards or USB hub may interfere with the Kinect.' ? I thought that was a good idea to isolate the problem because μPD720200 controller is known to be stable with Kinect and Jetson's PCI ex slot was standard one.

xlz, 2016-02-22:
μPD720200 controller is known to be stable with 4.2 kernel.

hanyazou, 2016-02-22:
is there any good instruction for replacing old kernel (and userland) of TK1? I'd like to install recent kernel instead of nVIDIA's official kernel.

xlz, 2016-02-22:
I have no idea. Upstream kernel has some support for Tegra but I'm afraid CUDA support will be lost.

xlz, 2016-02-22:
I ran your test too. Sometimes it got no errors at all, sometimes it started with this
```
[  924.641711] usb 2-1.1: Busted HC?  Not enough HCD resources for old configuration.
[  924.644995] usb 2-1.1: USB disconnect, device number 3
```
Then the tegra xhci started to have problems. It seems the errors only started if I kill Protonect.

I don't think there is anything we can do here.

xlz, 2016-02-22:
The guy in #581 found that https://github.com/jetsonhacks/libfreenect2 works with uPD720202 extension card, but the upstream libfreenect2 doesn't work. 

Maybe you want to try the jetsonhacks version too.

xlz, 2016-02-22:
I found you have been explicitly using gl processor. Weird, why not cuda processor? It can run without `DISPLAY=:0.0`. Scripts to max the clock https://gist.github.com/xlz/10ff1434af1ba33f5456

About your μPD720200, your dmesg tells you
```
[ 217.251602] ERROR: 256 KiB atomic DMA coherent pool is too small!
[ 217.251602] Please increase it with coherent_pool= kernel parameter!
```
Use kernel parameters `vmalloc=512M cma=64M coherent_pool=32M`? https://devtalk.nvidia.com/default/topic/770634/

xlz, 2016-02-26:
I find I have a uPD720202 card. I may be able to test it recently.

hanyazou, 2016-02-27:
>I broke my Jetson. I won't be able to test it anymore.

I'm so sorry to hear that.

>Use kernel parameters vmalloc=512M cma=64M coherent_pool=32M? https://devtalk.nvidia.com/default/topic/770634/

I tried this. But this error did not change. Protonect shows black window, no image.

>[ 217.234972] xhci_hcd 0000:01:00.0: ERROR Transfer event TRB DMA ptr not part of current TD

The tegra-xchi is better.

xlz, 2016-02-27:
> The tegra-xchi is better.

OK, but does patching libusb as in https://github.com/OpenKinect/libfreenect2/issues/581#issuecomment-187454035 make it work?

xlz, 2016-02-28:
I have a Intel separate machine, with these:
```
02:00.0 USB controller: NEC Corporation uPD720200 USB 3.0 Host Controller (rev 04)
kernel: 3.10.40-031040-generic #201405130835
```
Protonect works fine.

hanyazou, 2016-03-12:
> does patching libusb as in #581 (comment) make it work?

Wow! it works.

I was using libusb binary provided apt-get (that was libusb-1.0-0-dev:armhf 2:1.0.20-1ppa1~trusty).
PD720200 and Protonect could not show any images.

I tried depends/install_libusb.sh then. PD720200 and Protonect could display RGB image only. (-nodepth option required.)

Now I change MAX_ISO_BUFFER_LENGTH from 49152 * 128 to 49152. PD720200 and Protonect can display RGB and depth images.

Protonect still complains about packet skipping and lost.
>Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 8 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @2:4 with serial 005697142447
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9855
[Debug] [Freenect2DeviceImpl] status 0x090000: 9855
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Error] [RgbPacketStreamParser] packetsize or sequence doesn't match!
[Debug] [DepthPacketStreamParser] image data too short!
[Debug] [DepthPacketStreamParser] image data too short!
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Info] [Freenect2DeviceImpl] started
device serial: 005697142447
device firmware: 4.0.3912.0
[Debug] [DepthPacketStreamParser] not all subsequences received 992
[Debug] [DepthPacketStreamParser] not all subsequences received 76
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [RgbPacketStreamParser] skipping rgb packet!
[Debug] [DepthPacketStreamParser] skipping depth packet
[Info] [DepthPacketStreamParser] 30 packets were lost

No errors in dmesg. Dmesg shows only these lines.
>[ 2679.286062] usb 2-1.1: reset SuperSpeed USB device number 4 using xhci_hcd
[ 2679.300134] usb 2-1.1: Parent hub missing LPM exit latency info.  Power management will be impacted.

hanyazou, 2016-03-12:
@xlz Is this means we must use patched libusb for TK1 and uPD720200?

>MAX_ISO_BUFFER_LENGTH from 49152 * 128 to 49152

hanyazou, 2016-03-12:
I noticed that color of RGB image is odd on TK1. It mistakes Red for Blue and Blue for Red. Is this known issue?
[image]

hanyazou, 2016-03-12:
I tried the script on uPD720200. Whole Linux system had frozen when i=212. (no dmesg because of freeze.) Both tegra-xhci and uPD720200 could repeat 200 times and could not repeat 300 times.

>for i in {1..300}; do
echo ================ $i
DISPLAY=:0 ./bin/Protonect --frames 1 || break
done

I tired -noviewer. It stopped i=362. (system was frozen)

>for i in {1..1000}; do
echo ================ $i
DISPLAY=:0 ./bin/Protonect --frames 1 || break
done

"while dmesg; do sleep 1; done" ended these lines. 
>[ 1860.572469] usb 2-1.1: reset SuperSpeed USB device number 3 using xhci_hcd
[ 1860.587523] usb 2-1.1: Parent hub missing LPM exit latency info.  Power management will be impacted.
[ 1862.289492] usb 2-1.1: reset SuperSpeed USB device number 3 using xhci_hcd
[ 1862.302492] usb 2-1.1: Parent hub missing LPM exit latency info.  Power management will be impacted.
[ 1863.130286] mc-err: [mcerr] (vde) csr_vdebsevr: EMEM decode error on PDE or PTE entry
[ 1863.138111] mc-err: [mcerr]   status = 0x60000022; addr = 0x80185000
[ 1863.144276] mc-err: [mcerr]   secure: no, access-type: read, SMMU fault: nr-nw-s
[ 1863.267758] mc-err: Too many MC errors; throttling prints
[ 1863.972637] usb 2-1.1: reset SuperSpeed USB device number 3 using xhci_hcd
[ 1863.986564] usb 2-1.1: Parent hub missing LPM exit latency info.  Power management will be impacted.
[ 1864.956145] mc-err: [mcerr] (vde) csr_vdebsevr: EMEM decode error on PDE or PTE entry
[ 1864.963971] mc-err: [mcerr]   status = 0x60000022; addr = 0x81085000
[ 1864.970138] mc-err: [mcerr]   secure: no, access-type: read, SMMU fault: nr-nw-spacket_write_wait:

The conclusion is tegra-xhci is better. I'd like to try newer kernel later.

xlz, 2016-03-12:
> Is this means we must use patched libusb for TK1 and uPD720200?

No. This is just showing you how to make upd720200 work.

> I noticed that color of RGB image is odd on TK1. It mistakes Red for Blue and Blue for Red.

TegraJpeg outputs RGBX. The viewer expects BGRX. This is a mistake, yes, but it's not important. This is known too, it's in the Api docs.

I don't think extended testing is useful. Nvidia is not known for its great kernel support.

hanyazou, 2016-03-12:
> I don't think extended testing is useful.

I agree. We can close this thread.

xlz, 2016-03-12:
Actually nvidia's kernel support is horrible, otherwise Linus would not give them the finger.

closed 2016-03-12 by xlz