OpenKinect/libfreenect2 #587: Qt and libfreenect2 OpenGL pipeline crashed

whatisor, 2016-02-23:
I am using Libfreenect2 with OpenGL pipeline on Qt which I also use OpenGL to display 3D pointcloud.

I can run it with OpenCL pipeline but not OpenGL pipeline.

When start application, it alsways crash with error:

 BadWindow
Major code : 155 (GLX)

I think there are some conflict between Qt OpenGL  and glfw, but I don't know where.

(Protonect example can run well with OpenGL pipeline)

xlz, 2016-02-23:
Overview Description:

Check https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting if you haven't done so. 

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`
2. `uname -a`
3. `lsusb -t`
4. `lspci -nn`

Steps to Reproduce:

1. 
2. 
3. 

Actual Results:

Expected Results:

Reproducibility:

Additional Information:

whatisor, 2016-03-02:
Sorry, I do not have time to prepare this. 
Howevers, I fixed by using QGLWidget instead of QOpenGLWidget to avoid this issue.
Qt said QOpenGLWidget is a new implementation to replace QGLWidget, but it seems to be conflict with libfreenect2 (glfw I think).
Thank you

closed 2016-03-02 by xlz