OpenKinect/libfreenect2 #588: Protonect: packets lost

mmeliabit, 2016-02-23:
Overview Description:

When I run Protonect on Windows I get Info messages during all the execution like these ones:

[Info] [DepthPacketStreamParser] 34 packets were lost
[Info] [TurboJpegRgbPacketProcessor] avg. time: 57.1977ms -> ~17.4832Hz
[Info] [DepthPacketStreamParser] 30 packets were lost
[Info] [DepthPacketStreamParser] 30 packets were lost
[Info] [DepthPacketStreamParser] 24 packets were lost
[Info] [DepthPacketStreamParser] 31 packets were lost
[Info] [DepthPacketStreamParser] 14 packets were lost
[Info] [DepthPacketStreamParser] 26 packets were lost
[Info] [DepthPacketStreamParser] 25 packets were lost

Does this packet loss correspond to the normal performance? I see the Protonect depth image, but not very smoothly.

xlz, 2016-02-23:
It means the performance is terrible. You can use CUDA depth processor if you have GPU.

wired365, 2017-04-13:
I know this thread was closed a while ago but I have a follow up question regarding packet loss. I have tried running the Protonect example with various settings, -gpu=cuda with and without -noviewer and I am still getting lost packets. Below is an example of the output using -gpu=cuda -noviewer:

```
The viewer is turned off. Received 9500 frames. Ctrl-C to stop.
[Info] [DepthPacketStreamParser] 3 packets were lost
[Info] [DepthPacketStreamParser] 1 packets were lost
[Info] [TurboJpegRgbPacketProcessor] avg. time: 20.9214ms -> ~47.7981Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 9.01348ms -> ~110.945Hz
[Info] [DepthPacketStreamParser] 1 packets were lost
The viewer is turned off. Received 9600 frames. Ctrl-C to stop.
[Info] [DepthPacketStreamParser] 2 packets were lost
[Info] [DepthPacketStreamParser] 3 packets were lost
[Info] [TurboJpegRgbPacketProcessor] avg. time: 20.7446ms -> ~48.2054Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 9.03714ms -> ~110.654Hz
[Info] [DepthPacketStreamParser] 2 packets were lost
The viewer is turned off. Received 9700 frames. Ctrl-C to stop.
[Info] [DepthPacketStreamParser] 3 packets were lost
[Info] [TurboJpegRgbPacketProcessor] avg. time: 20.9905ms -> ~47.6405Hz
[Info] [DepthPacketStreamParser] 1 packets were lost
[Info] [OpenGLDepthPacketProcessor] avg. time: 9.23243ms -> ~108.314Hz
[Info] [DepthPacketStreamParser] 2 packets were lost
The viewer is turned off. Received 9800 frames. Ctrl-C to stop.
[Info] [DepthPacketStreamParser] 1 packets were lost
[Info] [DepthPacketStreamParser] 1 packets were lost
[Info] [TurboJpegRgbPacketProcessor] avg. time: 21.3172ms -> ~46.9104Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.97776ms -> ~111.386Hz
[Info] [DepthPacketStreamParser] 1 packets were lost
The viewer is turned off. Received 9900 frames. Ctrl-C to stop.
[Info] [DepthPacketStreamParser] 1 packets were lost
[Info] [DepthPacketStreamParser] 2 packets were lost
[Info] [TurboJpegRgbPacketProcessor] avg. time: 20.7948ms -> ~48.0888Hz
[Info] [DepthPacketStreamParser] 1 packets were lost
[Info] [OpenGLDepthPacketProcessor] avg. time: 8.5661ms -> ~116.739Hz
The viewer is turned off. Received 10000 frames. Ctrl-C to stop.
[Info] [TurboJpegRgbPacketProcessor] avg. time: 20.4966ms -> ~48.7886Hz
[Info] [DepthPacketStreamParser] 1 packets were lost
[Info] [OpenGLDepthPacketProcessor] avg. time: 4.47217ms -> ~223.605Hz
[Info] [DepthPacketStreamParser] 2 packets were lost
The viewer is turned off. Received 10100 frames. Ctrl-C to stop.
[Info] [DepthPacketStreamParser] 1 packets were lost
[Info] [TurboJpegRgbPacketProcessor] avg. time: 20.2215ms -> ~49.4523Hz
[Info] [OpenGLDepthPacketProcessor] avg. time: 4.44775ms -> ~224.833Hz
[Info] [DepthPacketStreamParser] 1 packets were lost
```

Why am I still getting packet loss? Isn't this the fastest way to run the example and process frames? Is this amount of packet loss normal?

xlz, 2017-04-13:
`Protonect cuda -noviewer`?

wired365, 2017-04-13:
Woops, I misunderstood the protonect help text. I was using `protonect -gpu=cuda -noviewer`. I had always wondered why it was still using the openGLDepthPacketProcessor. The weird part is when I run `protonect cuda -noviewer` I get the LIBUSB_ERROR_OTHER that I was getting a while back. When I don't use cuda it works just fine.

wired365, 2017-04-13:
I read the conversation between @xlz and @bn-dignitas regarding my original usbdk issue. My LIBUSB_ERROR_OTHER appears to be resolved by the addition of a pause as suggested in the other thread. The aside, when I do insert the pause and get protonect running with Cuda there appears to be pretty sever performance issues:
```
$ ./bin/Debug/Protonect.exe cuda -noviewer
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: C:\Users\kinect_admin\Documents\Repositories\libfreenect2\build\bin\Debug\Protonect.exe [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [CudaDepthPacketProcessorImpl] device 0: Quadro K5000 @ 705MHz Memory 4096MB
[Info] [CudaDepthPacketProcessorImpl] selected device 0
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 9 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:2 with serial 003370163847
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 3*1048576 ir: 8*64*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 003370163847
device firmware: 4.0.3916.0
[Info] [DepthPacketStreamParser] 5 packets were lost
[Info] [DepthPacketStreamParser] 8 packets were lost
[Info] [TurboJpegRgbPacketProcessor] avg. time: 19.4216ms -> ~51.4891Hz
[Info] [DepthPacketStreamParser] 17 packets were lost
[Info] [CudaDepthPacketProcessor] avg. time: 2.773ms -> ~360.621Hz
[Info] [DepthPacketStreamParser] 8 packets were lost
The viewer is turned off. Received 100 frames. Ctrl-C to stop.
[Info] [DepthPacketStreamParser] 17 packets were lost
[Info] [TurboJpegRgbPacketProcessor] avg. time: 19.0546ms -> ~52.4808Hz
[Info] [DepthPacketStreamParser] 8 packets were lost
[Info] [DepthPacketStreamParser] 8 packets were lost
[Info] [CudaDepthPacketProcessor] avg. time: 2.72927ms -> ~366.398Hz
The viewer is turned off. Received 200 frames. Ctrl-C to stop.
[Info] [TurboJpegRgbPacketProcessor] avg. time: 19.0612ms -> ~52.4626Hz
[Info] [DepthPacketStreamParser] 17 packets were lost
[Info] [DepthPacketStreamParser] 8 packets were lost
[Info] [DepthPacketStreamParser] 8 packets were lost
[Info] [CudaDepthPacketProcessor] avg. time: 2.66133ms -> ~375.753Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 19.4743ms -> ~51.3497Hz
The viewer is turned off. Received 300 frames. Ctrl-C to stop.
[Info] [DepthPacketStreamParser] 17 packets were lost
[Info] [DepthPacketStreamParser] 8 packets were lost
[Info] [DepthPacketStreamParser] 8 packets were lost
[Info] [TurboJpegRgbPacketProcessor] avg. time: 19.1219ms -> ~52.296Hz
```

xlz, 2017-04-13:
If you run it individually how much packet loss?
`Protonect cuda -noviewer -norgb`
`Protonect -noviewer -nodepth`

wired365, 2017-04-13:
`Protonect cuda -noviewer -norgb` outputs:
```
$ ./bin/Debug/Protonect.exe cuda -noviewer -norgb
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: C:\Users\kinect_admin\Documents\Repositories\libfreenect2\build\bin\Debug\Protonect.exe [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [CudaDepthPacketProcessorImpl] device 0: Quadro K5000 @ 705MHz Memory 4096MB
[Info] [CudaDepthPacketProcessorImpl] selected device 0
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:2 with serial 003370163847
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 3*1048576 ir: 8*64*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 003370163847
device firmware: 4.0.3916.0
[Info] [CudaDepthPacketProcessor] avg. time: 2.52198ms -> ~396.514Hz
The viewer is turned off. Received 100 frames. Ctrl-C to stop.
[Info] [CudaDepthPacketProcessor] avg. time: 2.424ms -> ~412.542Hz
The viewer is turned off. Received 200 frames. Ctrl-C to stop.
[Info] [CudaDepthPacketProcessor] avg. time: 2.42889ms -> ~411.711Hz
The viewer is turned off. Received 300 frames. Ctrl-C to stop.
[Info] [CudaDepthPacketProcessor] avg. time: 2.4047ms -> ~415.851Hz
The viewer is turned off. Received 400 frames. Ctrl-C to stop.
[Info] [CudaDepthPacketProcessor] avg. time: 2.39106ms -> ~418.224Hz
The viewer is turned off. Received 500 frames. Ctrl-C to stop.
[Info] [CudaDepthPacketProcessor] avg. time: 2.3913ms -> ~418.183Hz
The viewer is turned off. Received 600 frames. Ctrl-C to stop.
```

`Protonect cuda -noviewer -nodepth` outputs:
```
$ ./bin/Debug/Protonect.exe cuda -noviewer -nodepth
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: C:\Users\kinect_admin\Documents\Repositories\libfreenect2\build\bin\Debug\Protonect.exe [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [CudaDepthPacketProcessorImpl] device 0: Quadro K5000 @ 705MHz Memory 4096MB
[Info] [CudaDepthPacketProcessorImpl] selected device 0
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 10 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:2 with serial 003370163847
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 3*1048576 ir: 8*64*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 003370163847
device firmware: 4.0.3916.0
[Info] [TurboJpegRgbPacketProcessor] avg. time: 18.5471ms -> ~53.9168Hz
The viewer is turned off. Received 100 frames. Ctrl-C to stop.
[Info] [TurboJpegRgbPacketProcessor] avg. time: 18.4255ms -> ~54.2727Hz
The viewer is turned off. Received 200 frames. Ctrl-C to stop.
[Info] [TurboJpegRgbPacketProcessor] avg. time: 18.3064ms -> ~54.6256Hz
The viewer is turned off. Received 300 frames. Ctrl-C to stop.
[Info] [TurboJpegRgbPacketProcessor] avg. time: 18.3048ms -> ~54.6304Hz
The viewer is turned off. Received 400 frames. Ctrl-C to stop.
[Info] [TurboJpegRgbPacketProcessor] avg. time: 18.2788ms -> ~54.7081Hz
The viewer is turned off. Received 500 frames. Ctrl-C to stop.
[... 20 lines cut ...]
[Info] [Freenect2Impl] found valid Kinect v2 @4:2 with serial 003370163847
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 3*1048576 ir: 8*64*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 003370163847
device firmware: 4.0.3916.0
[Info] [DepthPacketStreamParser] 3 packets were lost
[Info] [DepthPacketStreamParser] 7 packets were lost
[Info] [DepthPacketStreamParser] 8 packets were lost
[Info] [TurboJpegRgbPacketProcessor] avg. time: 18.8218ms -> ~53.1299Hz
[Info] [DepthPacketStreamParser] 7 packets were lost
[Info] [CudaDepthPacketProcessor] avg. time: 3.03095ms -> ~329.929Hz
[Info] [DepthPacketStreamParser] 8 packets were lost
The viewer is turned off. Received 100 frames. Ctrl-C to stop.
[Info] [DepthPacketStreamParser] 7 packets were lost
[Info] [DepthPacketStreamParser] 8 packets were lost
[Info] [TurboJpegRgbPacketProcessor] avg. time: 18.8582ms -> ~53.0273Hz
```

It looks things will work fine when only processing either the RGB or Depth, but there are some performance impacts when processing both.

xlz, 2017-04-13:
I noticed you're running Debug build. What is the performance with Release build?

wired365, 2017-04-13:
Hmm the performance appears to be intermittent for both Debug and Release. I am now getting up to 25 packets lost per update running the same Debug version with `cuda -noviewer -norgb`. The release mode appears to be a bit more consistent but still drops packets. The various outputs are below:
```
$ ./bin/Release/Protonect.exe cuda -noviewer -norgb
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: C:\Users\kinect_admin\Documents\Repositories\libfreenect2\build\bin\Release\Protonect.exe [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [CudaDepthPacketProcessorImpl] device 0: Quadro K5000 @ 705MHz Memory 4096MB
[Info] [CudaDepthPacketProcessorImpl] selected device 0
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 9 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:2 with serial 003370163847
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 3*1048576 ir: 8*64*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 003370163847
device firmware: 4.0.3916.0
[Info] [DepthPacketStreamParser] 4 packets were lost
[Info] [DepthPacketStreamParser] 5 packets were lost
[Info] [DepthPacketStreamParser] 19 packets were lost
[Info] [CudaDepthPacketProcessor] avg. time: 2.44671ms -> ~408.712Hz
The viewer is turned off. Received 100 frames. Ctrl-C to stop.
[Info] [DepthPacketStreamParser] 11 packets were lost
[Info] [DepthPacketStreamParser] 11 packets were lost
[Info] [DepthPacketStreamParser] 6 packets were lost
[Info] [CudaDepthPacketProcessor] avg. time: 2.2665ms -> ~441.209Hz
The viewer is turned off. Received 200 frames. Ctrl-C to stop.
[Info] [DepthPacketStreamParser] 5 packets were lost
[Info] [DepthPacketStreamParser] 7 packets were lost
[Info] [DepthPacketStreamParser] 1 packets were lost
[Info] [CudaDepthPacketProcessor] avg. time: 2.30438ms -> ~433.957Hz
The viewer is turned off. Received 300 frames. Ctrl-C to stop.
[Info] [DepthPacketStreamParser] 4 packets were lost
[Info] [DepthPacketStreamParser] 4 packets were lost
[Info] [DepthPacketStreamParser] 5 packets were lost
[Info] [CudaDepthPacketProcessor] avg. time: 2.31292ms -> ~432.355Hz
The viewer is turned off. Received 400 frames. Ctrl-C to stop.
[Info] [DepthPacketStreamParser] 3 packets were lost
[Info] [DepthPacketStreamParser] 3 packets were lost
```

```
$ ./bin/Release/Protonect.exe cuda -noviewer -nodepth
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: C:\Users\kinect_admin\Documents\Repositories\libfreenect2\build\bin\Release\Protonect.exe [-gpu=<id>] [gl | cl | clkde | cuda | cudakde | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [CudaDepthPacketProcessorImpl] device 0: Quadro K5000 @ 705MHz Memory 4096MB
[Info] [CudaDepthPacketProcessorImpl] selected device 0
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 9 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:2 with serial 003370163847
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 3*1048576 ir: 8*64*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 003370163847
device firmware: 4.0.3916.0
[Info] [TurboJpegRgbPacketProcessor] avg. time: 19.9904ms -> ~50.0241Hz
[... 34 lines cut ...]
[Info] [DepthPacketStreamParser] 1 packets were lost
[Info] [CudaDepthPacketProcessor] avg. time: 2.36845ms -> ~422.217Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 20.5164ms -> ~48.7415Hz
The viewer is turned off. Received 100 frames. Ctrl-C to stop.
[Info] [DepthPacketStreamParser] 1 packets were lost
[Info] [CudaDepthPacketProcessor] avg. time: 2.26398ms -> ~441.7Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 20.1672ms -> ~49.5855Hz
The viewer is turned off. Received 200 frames. Ctrl-C to stop.
[Info] [DepthPacketStreamParser] 1 packets were lost
[Info] [CudaDepthPacketProcessor] avg. time: 2.26578ms -> ~441.348Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 20.1413ms -> ~49.6492Hz
The viewer is turned off. Received 300 frames. Ctrl-C to stop.
[Info] [DepthPacketStreamParser] 1 packets were lost
[Info] [DepthPacketStreamParser] 1 packets were lost
[Info] [CudaDepthPacketProcessor] avg. time: 2.26683ms -> ~441.144Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 20.0848ms -> ~49.7888Hz
[Info] [DepthPacketStreamParser] 2 packets were lost
The viewer is turned off. Received 400 frames. Ctrl-C to stop.
[Info] [DepthPacketStreamParser] 2 packets were lost
[Info] [CudaDepthPacketProcessor] avg. time: 2.26846ms -> ~440.828Hz
[Info] [TurboJpegRgbPacketProcessor] avg. time: 20.5686ms -> ~48.6179Hz
The viewer is turned off. Received 500 frames. Ctrl-C to stop.
```

I am not sure why performance isn't consistent from run to run. I get runs where there appears to be very few packets dropped and others where it is dropping almost 20+ packets per line. The `-nodepth` runs do appear to be much more consistent than those involving the depth processing.

xlz, 2017-04-13:
I don't know what kind of shell you're using. You can run these commands with the environment variable `LOGFILE=protonect.log` which provides more information about why packets are lost.

wired365, 2017-04-13:
Running with the command `protonect.exe cuda -noviewer -norgb` I get the following in the log file:
```
[Info] [CudaDepthPacketProcessorImpl] device 0: Quadro K5000 @ 705MHz Memory 4096MB
[Info] [CudaDepthPacketProcessorImpl] selected device 0
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 9 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:2 with serial 003370163847
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 3*1048576 ir: 8*64*33792
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Debug] [DepthPacketStreamParser] not all subsequences received 1021
[Debug] [DepthPacketStreamParser] not all subsequences received 1015
[Debug] [DepthPacketStreamParser] not all subsequences received 1007
[Info] [DepthPacketStreamParser] 3 packets were lost
[Debug] [DepthPacketStreamParser] not all subsequences received 991
[Debug] [DepthPacketStreamParser] not all subsequences received 991
[Debug] [DepthPacketStreamParser] not all subsequences received 1021
[Debug] [DepthPacketStreamParser] not all subsequences received 1015
[Debug] [DepthPacketStreamParser] not all subsequences received 991
[Debug] [DepthPacketStreamParser] not all subsequences received 767
[Debug] [DepthPacketStreamParser] not all subsequences received 511
[Debug] [DepthPacketStreamParser] not all subsequences received 1019
[Info] [DepthPacketStreamParser] 8 packets were lost
[Debug] [DepthPacketStreamParser] not all subsequences received 767
[Debug] [DepthPacketStreamParser] not all subsequences received 1007
[Debug] [DepthPacketStreamParser] not all subsequences received 1007
[Debug] [DepthPacketStreamParser] not all subsequences received 991
[Debug] [DepthPacketStreamParser] not all subsequences received 991
[Debug] [DepthPacketStreamParser] not all subsequences received 1015
[Debug] [DepthPacketStreamParser] not all subsequences received 767
[Debug] [DepthPacketStreamParser] not all subsequences received 959
[Debug] [DepthPacketStreamParser] not all subsequences received 1015
[Debug] [DepthPacketStreamParser] not all subsequences received 1021
[Debug] [DepthPacketStreamParser] not all subsequences received 959
[Debug] [DepthPacketStreamParser] not all subsequences received 895
[Debug] [DepthPacketStreamParser] not all subsequences received 955
[Debug] [DepthPacketStreamParser] not all subsequences received 1021
[Debug] [DepthPacketStreamParser] not all subsequences received 1007
[Debug] [DepthPacketStreamParser] not all subsequences received 1021
[Debug] [DepthPacketStreamParser] not all subsequences received 959
[Debug] [DepthPacketStreamParser] not all subsequences received 1015
[Debug] [DepthPacketStreamParser] not all subsequences received 959
[Debug] [DepthPacketStreamParser] not all subsequences received 1007
[Debug] [DepthPacketStreamParser] not all subsequences received 895
[Debug] [DepthPacketStreamParser] not all subsequences received 959
[Debug] [DepthPacketStreamParser] not all subsequences received 1021
[Debug] [DepthPacketStreamParser] not all subsequences received 503
[Debug] [DepthPacketStreamParser] not all subsequences received 991
[Debug] [DepthPacketStreamParser] not all subsequences received 1007
[Debug] [DepthPacketStreamParser] not all subsequences received 991
[Debug] [DepthPacketStreamParser] not all subsequences received 991
[Info] [CudaDepthPacketProcessor] avg. time: 2.44386ms -> ~409.189Hz
[Info] [DepthPacketStreamParser] 28 packets were lost
```

The `protonect.exe cuda -noviewer -nodepth` looks fine to me, no dropped packets:
```
[Info] [CudaDepthPacketProcessorImpl] device 0: Quadro K5000 @ 705MHz Memory 4096MB
[Info] [CudaDepthPacketProcessorImpl] selected device 0
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 9 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:2 with serial 003370163847
[Info] [Freenect2Impl] found 1 devices
[... 62 lines cut ...]
[Debug] [DepthPacketStreamParser] not all subsequences received 1021
[Info] [DepthPacketStreamParser] 3 packets were lost
[Debug] [DepthPacketStreamParser] not all subsequences received 991
[Info] [DepthPacketStreamParser] 1 packets were lost
[Debug] [DepthPacketStreamParser] not all subsequences received 991
[Info] [TurboJpegRgbPacketProcessor] avg. time: 20.5297ms -> ~48.71Hz
[Info] [CudaDepthPacketProcessor] avg. time: 2.27963ms -> ~438.667Hz
[Debug] [DepthPacketStreamParser] not all subsequences received 767
[Info] [DepthPacketStreamParser] 2 packets were lost
[Debug] [DepthPacketStreamParser] not all subsequences received 1015
[Info] [DepthPacketStreamParser] 1 packets were lost
[Debug] [DepthPacketStreamParser] not all subsequences received 1021
[Info] [DepthPacketStreamParser] 1 packets were lost
[Debug] [DepthPacketStreamParser] not all subsequences received 1021
[Info] [TurboJpegRgbPacketProcessor] avg. time: 20.5762ms -> ~48.5999Hz
[Debug] [DepthPacketStreamParser] not all subsequences received 1015
[Info] [DepthPacketStreamParser] 2 packets were lost
[Info] [CudaDepthPacketProcessor] avg. time: 2.28212ms -> ~438.19Hz
[Debug] [DepthPacketStreamParser] not all subsequences received 895
[Debug] [DepthPacketStreamParser] not all subsequences received 991
[Info] [DepthPacketStreamParser] 2 packets were lost
[Debug] [DepthPacketStreamParser] not all subsequences received 991
[Debug] [DepthPacketStreamParser] not all subsequences received 767
[Debug] [DepthPacketStreamParser] not all subsequences received 767
```

xlz, 2017-04-13:
`not all subsequences received` means there is something wrong in your usb stack.

Are you running on UsbDk 1.0.16 or .17?

wired365, 2017-04-13:
I'm running UsbDk 1.0.17

xlz, 2017-04-13:
What are the results with 1.0.16? The other bug report you referenced also was concluded with the suggestion of using 1.0.16 instead of adding manual delay.

wired365, 2017-04-14:
Yeah, I tried rolling back to usbDk 1.0.16 and rebuilding everything and I I still end up with the same error message, "not all subsequences were received". Is there any kind of deeper logging or error checking that I could do to determine where in the stack things are going wrong?

xlz, 2017-04-20:
I'm testing it.

xlz, 2017-04-21:
Are you using Windows 10?

I tested on Windows. It seems to be something deep in the USB driver causing real packet losses. My USB device is 8086:a12f Sunrise Point-H USB 3.0 xHCI Controller.

I tried various things (usbdk vs libusbk, libusb versions, vs2015 vs vs2017) but none worked. Too bad.

The only other things possible would be testing it under Linux (hardware problem?), or under Windows 8.1 (driver problem?).

wired365, 2017-04-21:
Yeah I am using Windows 10 and I have tried a handful of different USB 3.0 cards, thinking it was a hardware issue, as well as different versions of usbDk with no change in the result. I've also tested on different windows 10 computers and get the same issue. So it looks like my results match yours.

I can try to get my hands on a Linux machine, but for right now I am stuck with Windows 10 as the platform where my work will be deployed. I'm going to try and work with the packet loss for now to see how severe the performance impact is to my program.

xlz, 2017-04-21:
Your results indicate it's a Windows 10 driver/kernel issue. Bad luck that we can do nothing about it.

taseenb, 2017-05-06:
It's not a Windows 10 issue. I can reproduce this on macOs 1.12.4 (Sierra) with:

`./Protonect cl`

And it does not seem a performance issue anyway, since the error appears in both Macs i tested (a fast MacBook Pro with a Nvidia discrete gpu and a much slower MacBook Air), with and without CL activated (CUDA does not work on my system).

xlz, 2017-05-06:
No, this issue is for Windows 10 because it's confirmed the cause of the problem reported is in Windows 10's USB driver or other kernel part, possibly also in libusb, and least likely in libfreenect2. The result is a performance issue.

If Mac has kernel problems that's a different source of problems. Unfortunately we are equally unable to do anything about it because its kernel is not open source.

taseenb, 2017-05-06:
Strange issue. I used a java wrapper of the native Windows SDK (Kinect2 for Processing) and it didn't seem to show any packet loss, but I could be wrong as I don't know how could i test it properly on Windows.