OpenKinect/libfreenect2 #589: Cannot build protonect

user3e93rf23, 2016-02-23:
I try to make the directory but cannot get the examples to build. I have glfw3 in mind but their examples (e.g. ./particles) work fine.  

Linking CXX executable ../bin/Protonect
../lib/libfreenect2.so.0.2.0: undefined reference to `XRRSetCrtcGamma'
../lib/libfreenect2.so.0.2.0: undefined reference to `XRRGetOutputInfo'
../lib/libfreenect2.so.0.2.0: undefined reference to `XcursorImageLoadCursor'
../lib/libfreenect2.so.0.2.0: undefined reference to `XRRGetScreenResourcesCurrent'
../lib/libfreenect2.so.0.2.0: undefined reference to `XRRAllocGamma'
../lib/libfreenect2.so.0.2.0: undefined reference to `XRRQueryVersion'
../lib/libfreenect2.so.0.2.0: undefined reference to `XRRFreeOutputInfo'
../lib/libfreenect2.so.0.2.0: undefined reference to `XRRFreeScreenResources'
../lib/libfreenect2.so.0.2.0: undefined reference to `XRRGetOutputPrimary'
../lib/libfreenect2.so.0.2.0: undefined reference to `XRRFreeGamma'
../lib/libfreenect2.so.0.2.0: undefined reference to `XineramaQueryScreens'
../lib/libfreenect2.so.0.2.0: undefined reference to `XRRQueryExtension'
../lib/libfreenect2.so.0.2.0: undefined reference to `XineramaIsActive'
../lib/libfreenect2.so.0.2.0: undefined reference to `XineramaQueryExtension'
../lib/libfreenect2.so.0.2.0: undefined reference to `XRRGetScreenResources'
../lib/libfreenect2.so.0.2.0: undefined reference to `XRRGetCrtcGamma'
../lib/libfreenect2.so.0.2.0: undefined reference to `XRRFreeCrtcInfo'
../lib/libfreenect2.so.0.2.0: undefined reference to `XRRUpdateConfiguration'
../lib/libfreenect2.so.0.2.0: undefined reference to `XRRGetCrtcInfo'
../lib/libfreenect2.so.0.2.0: undefined reference to `XcursorImageCreate'
../lib/libfreenect2.so.0.2.0: undefined reference to `XRRSetCrtcConfig'
../lib/libfreenect2.so.0.2.0: undefined reference to `XcursorImageDestroy'
../lib/libfreenect2.so.0.2.0: undefined reference to `XRRSelectInput'
../lib/libfreenect2.so.0.2.0: undefined reference to `XRRGetCrtcGammaSize'
collect2: error: ld returned 1 exit status
make[2]: *** [bin/Protonect] Error 1
make[1]: *** [examples/CMakeFiles/Protonect.dir/all] Error 2
make: *** [all] Error 2

user3e93rf23, 2016-02-23:
`baum@baum-ThinkPad-L520:~/catkin_ws/src_3party/libfreenect2/build$ cmake -DENABLE_CXX11=ON -DCMAKE_INSTALL_PREFIX=$HOME/freenect2 ..
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Performing Test LIBFREENECT2_THREADING_STDLIB
-- Performing Test LIBFREENECT2_THREADING_STDLIB - Success
-- using stdlib as threading library
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- checking for module 'libusb-1.0>=1.0.20'
--   found libusb-1.0, version 1.0.20
-- checking for modules 'libva;libva-drm'
--   package 'libva' not found
--   package 'libva-drm' not found
-- Found JPEG: /usr/lib/x86_64-linux-gnu/libjpeg.so  
-- Could NOT find TegraJPEG (missing:  TegraJPEG_LIBRARIES TegraJPEG_INCLUDE_DIRS TegraJPEG_L4T_OK TegraJPEG_DRIVER_OK TegraJPEG_WORKS) 
-- Performing Test TURBOJPEG_WORKS
-- Performing Test TURBOJPEG_WORKS - Success
-- Found TurboJPEG: /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0  
-- checking for module 'glfw3'
--   found glfw3, version 3.2.0
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/x86_64-linux-gnu/libX11.so
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Looking for CL_VERSION_2_0
-- Looking for CL_VERSION_2_0 - not found
-- Looking for CL_VERSION_1_2
-- Looking for CL_VERSION_1_2 - found
-- Found OpenCL: /usr/lib/x86_64-linux-gnu/libOpenCL.so (found version "1.2") 
-- Performing Test OpenCL_C_WORKS
-- Performing Test OpenCL_C_WORKS - Success
-- Performing Test OpenCL_CXX_WORKS
-- Performing Test OpenCL_CXX_WORKS - Success
CUDA_TOOLKIT_ROOT_DIR not found or specified
-- Could NOT find CUDA (missing:  CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY) 
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Linking with these libraries: 
 /usr/lib/x86_64-linux-gnu/libusb-1.0.so
 /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0
 /usr/local/lib/libglfw3.a
 /usr/lib/x86_64-linux-gnu/libGL.so
 /usr/lib/x86_64-linux-gnu/libOpenCL.so
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- Configurating examples
-- Using in-tree freenect2 target
-- checking for module 'libopenni2'
--   package 'libopenni2' not found
-- Could NOT find OpenNI2 (missing:  OpenNI2_LIBRARY_DIR OpenNI2_INCLUDE_DIRS) 
-- Feature list:
--   CUDA    no
--   CXX11    yes
--   Examples    yes
--   OpenCL    yes
--   OpenGL    yes
--   OpenNI2    no
--   TegraJPEG    no
--   Threading    std::thread
--   TurboJPEG    yes
--   VAAPI    no
--   VideoToolbox    no (Apple only)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/baum/catkin_ws/src_3party/libfreenect2/build
`

user3e93rf23, 2016-02-23:
`baum@baum-ThinkPad-L520:~/catkin_ws/src_3party/libfreenect2/build$ cmake -DENABLE_CXX11=ON -DCMAKE_INSTALL_PREFIX=$HOME/freenect2 ..
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Performing Test LIBFREENECT2_THREADING_STDLIB
-- Performing Test LIBFREENECT2_THREADING_STDLIB - Success
-- using stdlib as threading library
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- checking for module 'libusb-1.0>=1.0.20'
--   found libusb-1.0, version 1.0.20
-- checking for modules 'libva;libva-drm'
--   package 'libva' not found
--   package 'libva-drm' not found
-- Found JPEG: /usr/lib/x86_64-linux-gnu/libjpeg.so  
-- Could NOT find TegraJPEG (missing:  TegraJPEG_LIBRARIES TegraJPEG_INCLUDE_DIRS TegraJPEG_L4T_OK TegraJPEG_DRIVER_OK TegraJPEG_WORKS) 
-- Performing Test TURBOJPEG_WORKS
-- Performing Test TURBOJPEG_WORKS - Success
-- Found TurboJPEG: /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0  
-- checking for module 'glfw3'
--   found glfw3, version 3.2.0
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/x86_64-linux-gnu/libX11.so
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Looking for CL_VERSION_2_0
-- Looking for CL_VERSION_2_0 - not found
-- Looking for CL_VERSION_1_2
-- Looking for CL_VERSION_1_2 - found
-- Found OpenCL: /usr/lib/x86_64-linux-gnu/libOpenCL.so (found version "1.2") 
-- Performing Test OpenCL_C_WORKS
-- Performing Test OpenCL_C_WORKS - Success
-- Performing Test OpenCL_CXX_WORKS
-- Performing Test OpenCL_CXX_WORKS - Success
CUDA_TOOLKIT_ROOT_DIR not found or specified
-- Could NOT find CUDA (missing:  CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY) 
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Linking with these libraries: 
 /usr/lib/x86_64-linux-gnu/libusb-1.0.so
 /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0
 /usr/local/lib/libglfw3.a
 /usr/lib/x86_64-linux-gnu/libGL.so
 /usr/lib/x86_64-linux-gnu/libOpenCL.so
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- Configurating examples
-- Using in-tree freenect2 target
-- checking for module 'libopenni2'
--   package 'libopenni2' not found
-- Could NOT find OpenNI2 (missing:  OpenNI2_LIBRARY_DIR OpenNI2_INCLUDE_DIRS) 
-- Feature list:
--   CUDA    no
--   CXX11    yes
--   Examples    yes
--   OpenCL    yes
--   OpenGL    yes
--   OpenNI2    no
--   TegraJPEG    no
--   Threading    std::thread
--   TurboJPEG    yes
--   VAAPI    no
--   VideoToolbox    no (Apple only)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/baum/catkin_ws/src_3party/libfreenect2/build
...`

user3e93rf23, 2016-02-23:
> baum@baum-ThinkPad-L520:~/catkin_ws/src_3party/libfreenect2/build$ cmake -DENABLE_CXX11=ON -DCMAKE_INSTALL_PREFIX=$HOME/freenect2 ..
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Performing Test LIBFREENECT2_THREADING_STDLIB
-- Performing Test LIBFREENECT2_THREADING_STDLIB - Success
-- using stdlib as threading library
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- checking for module 'libusb-1.0>=1.0.20'
--   found libusb-1.0, version 1.0.20
-- checking for modules 'libva;libva-drm'
--   package 'libva' not found
--   package 'libva-drm' not found
-- Found JPEG: /usr/lib/x86_64-linux-gnu/libjpeg.so  
-- Could NOT find TegraJPEG (missing:  TegraJPEG_LIBRARIES TegraJPEG_INCLUDE_DIRS TegraJPEG_L4T_OK TegraJPEG_DRIVER_OK TegraJPEG_WORKS) 
-- Performing Test TURBOJPEG_WORKS
-- Performing Test TURBOJPEG_WORKS - Success
-- Found TurboJPEG: /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0  
-- checking for module 'glfw3'
--   found glfw3, version 3.2.0
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/x86_64-linux-gnu/libX11.so;/usr/lib/x86_64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/x86_64-linux-gnu/libX11.so
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Looking for CL_VERSION_2_0
-- Looking for CL_VERSION_2_0 - not found
-- Looking for CL_VERSION_1_2
-- Looking for CL_VERSION_1_2 - found
-- Found OpenCL: /usr/lib/x86_64-linux-gnu/libOpenCL.so (found version "1.2") 
-- Performing Test OpenCL_C_WORKS
-- Performing Test OpenCL_C_WORKS - Success
-- Performing Test OpenCL_CXX_WORKS
-- Performing Test OpenCL_CXX_WORKS - Success
CUDA_TOOLKIT_ROOT_DIR not found or specified
-- Could NOT find CUDA (missing:  CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY) 
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Linking with these libraries: 
 /usr/lib/x86_64-linux-gnu/libusb-1.0.so
 /usr/lib/x86_64-linux-gnu/libturbojpeg.so.0
 /usr/local/lib/libglfw3.a
 /usr/lib/x86_64-linux-gnu/libGL.so
 /usr/lib/x86_64-linux-gnu/libOpenCL.so
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- Configurating examples
-- Using in-tree freenect2 target
-- checking for module 'libopenni2'
--   package 'libopenni2' not found
-- Could NOT find OpenNI2 (missing:  OpenNI2_LIBRARY_DIR OpenNI2_INCLUDE_DIRS) 
-- Feature list:
--   CUDA    no
--   CXX11    yes
--   Examples    yes
--   OpenCL    yes
--   OpenGL    yes
--   OpenNI2    no
--   TegraJPEG    no
--   Threading    std::thread
--   TurboJPEG    yes
--   VAAPI    no
--   VideoToolbox    no (Apple only)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/baum/catkin_ws/src_3party/libfreenect2/build

user3e93rf23, 2016-02-23:
[image]

xlz, 2016-02-23:
What is `/usr/local/lib/libglfw3.a`?

> (Ubuntu 14.04 only) sudo dpkg -i debs/libglfw3*deb; sudo apt-get install -f; sudo apt-get install libgl1-mesa-dri-lts-vivid (If the last command conflicts with other packages, don't do it.)

Did you follow the instructions?

user3e93rf23, 2016-02-23:
Thats what I cannot figure out, what libglfw3 is. 
Yes I followed the instructions, and additionally installed glfw like [so](http://www.glfw.org/docs/latest/compile.html#compile_generate)

user3e93rf23, 2016-02-23:
I set-up a fresh system and try to reproduce the error. If you have a fast hint I would be happy, otherwise I try it on the new system

xlz, 2016-02-23:
> additionally installed glfw like so

It is NOT okay to do this. The instructions didn't ask you to do this. Remove files generated by this step and try again.

user3e93rf23, 2016-02-23:
Thank you , I will try and be back

user3e93rf23, 2016-02-24:
Yes it does work fine if glfw is not installed seperately too, which I needed for other compilations. I switched notebooks and the fresh one works fine with your readme. Thank you.

closed 2016-02-24 by user3e93rf23

Gitwangshuo, 2018-07-21:
thanks a lot

eshafeeqe, 2019-02-21:
I come across the same issue and resolved as below. 

First searched available libglfw using command below
`locate libglfw`

Got following output:
`/usr/lib/x86_64-linux-gnu/libglfw.so
/usr/lib/x86_64-linux-gnu/libglfw.so.3
/usr/lib/x86_64-linux-gnu/libglfw.so.3.1
/usr/local/lib/libglfw3.a
/usr/share/doc/libglfw3
/usr/share/doc/libglfw3-dev
/usr/share/doc/libglfw3/changelog.Debian.gz
/usr/share/doc/libglfw3/copyright
/usr/share/doc/libglfw3-dev/changelog.Debian.gz
/usr/share/doc/libglfw3-dev/copyright
/var/lib/dpkg/info/libglfw3-dev:amd64.list
/var/lib/dpkg/info/libglfw3-dev:amd64.md5sums
/var/lib/dpkg/info/libglfw3:amd64.list
/var/lib/dpkg/info/libglfw3:amd64.md5sums
/var/lib/dpkg/info/libglfw3:amd64.shlibs
/var/lib/dpkg/info/libglfw3:amd64.symbols
/var/lib/dpkg/info/libglfw3:amd64.triggers`

took /usr/lib/x86_64-linux-gnu/libglfw.so.3.1 and edit the path for GLFW3 library path manually using ccmake 

Then generated cmake. 
 And it started working fine now.