OpenKinect/libfreenect2 #590: libfreenect2 builds but doesn't run Protonect

robozo, 2016-02-24:
Overview Description:

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`
2. `uname -a`
3. `lsusb -t`
4. `lspci -nn`

Steps to Reproduce:

1. 
2. 
3. 

Actual Results:

Expected Results:

Reproducibility:

Additional Information:

xlz, 2016-02-24:
@fran6co This time we have 10.8.

https://developer.apple.com/library/mac/releasenotes/General/APIDiffsMacOSX10_9/VideoToolbox.html
```
Added kVTVideoDecoderSpecification_EnableHardwareAcceleratedVideoDecoder
Added kVTVideoDecoderSpecification_RequireHardwareAcceleratedVideoDecoder
```
What happens if you use kVTVideoDecoderSpecification_RequireHardwareAcceleratedVideoDecoder?

fran6co, 2016-02-24:
I think I can remove the hardware accelerated stuff because it's not actually accelerated in Mac... It is for IOS but it should be using it by default. I'll create the PR and test it for 10.8

fran6co, 2016-02-24:
Just tried in 10.8 and works no problem.

closed 2016-02-24 by xlz

robozo, 2016-02-27:
Thanks!  Sorry about 10.8, is stable for me for a lot of old code.
It could have been worse, could have been classic mac 9.1!

fran6co, 2016-02-27:
The patch was trivial, as long Apple doesn't add JPEG hardware acceleration by setting some flags we are going to be ok.