OpenKinect/libfreenect2 #591: OS X Captain [DepthPacketStreamParser] subpacket too large

EnvnHash, 2016-02-24:
I got an issue on my MacBook Pro 8,2. I had libfreenect2 running on OS X 10.10 without any issues. Upgrading to OS X 10.11 and building again from source, I get continuously " [DepthPacketStreamParser] subpacket too large". No Viewer, no image.

Tried to run with cpu, gl and cl. Same issue. Running Protonect with -nodepth option, the RGB image is shown and the app runs smoothly without problems.

I´m using libusb 1.0.20 from macports. Also tried libusb from git - same issue.

My Kinect v2 runs via a StarTech Thunderbolt / USB 3.0 adapter (http://www.startech.com/de/Karten-Adapter/Festplattencontroller/SATA-Karten/Thunderbolt-eSATA-USB-3-Adapter~TB2USB3ES)
So far had no issues with this...
Graphics card is AMD Radeon HD 6490M 256 MB

`  +-o Root Hub Simulation Simulation@00000000  <class AppleUSBRootHubDevice, id 0x100006c1c, registered, matched, active, busy 0 (1 ms), retain 8>
    | {
    |   "iManufacturer" = 0
    |   "bNumConfigurations" = 1
    |   "idProduct" = 32773
    |   "bMaxPacketSize0" = 8
    |   "Built-In" = Yes
    |   "iProduct" = 0
    |   "USB Product Name" = "Root Hub Simulation Simulation"
    |   "iSerialNumber" = 0
    |   "bDeviceClass" = 9
    |   "IOUserClientClass" = "IOUSBDeviceUserClientV2"
    |   "locationID" = 0
    |   "bDeviceSubClass" = 255
    |   "bcdUSB" = 256
    |   "non-removable" = "yes"
    |   "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
    |   "bDeviceProtocol" = 0
    |   "USB Vendor Name" = "Apple Inc."
    |   "idVendor" = 1452
    |   "IOGeneralInterest" = "IOCommand is not serializable"
    |   "IOClassNameOverride" = "IOUSBRootHubDevice"
    | }
    | 
    +-o NuiSensor Adaptor      @00300000  <class AppleUSBDevice, id 0x100006ef4, registered, matched, active, busy 0 (4 ms), retain 14>
    |   {
    |     "sessionID" = 132639939487740
    |     "iManufacturer" = 1
    |     "bNumConfigurations" = 1
    |     "idProduct" = 729
    |     "bcdDevice" = 112
    |     "Bus Power Available" = 500
    |     "USB Address" = 1
    |     "bMaxPacketSize0" = 64
    |     "iProduct" = 2
    |     "iSerialNumber" = 0
    |     "bDeviceClass" = 9
    |     "Built-In" = No
    |     "locationID" = 3145728
    |     "bDeviceSubClass" = 0
    |     "bcdUSB" = 528
    |     "USB Product Name" = "NuiSensor Adaptor      "
    |     "PortNum" = 3
    |     "non-removable" = "no"
    |     "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
    |     "bDeviceProtocol" = 1
    |     "IOUserClientClass" = "IOUSBDeviceUserClientV2"
    |     "IOPowerManagement" = {"DevicePowerState"=0,"CurrentPowerState"=3,"CapabilityFlags"=65536,"MaxPowerState"=4,"DriverPowerState"=3}
    |     "Device Speed" = 2
    |     "USB Vendor Name" = "Microsoft Corporation  "
    |     "idVendor" = 1118
    |     "IOGeneralInterest" = "IOCommand is not serializable"
    |     "IOClassNameOverride" = "IOUSBDevice"
    |   }
    |   
    +-o NuiSensor Adaptor      @00100000  <class AppleUSBDevice, id 0x100006f07, registered, matched, active, busy 0 (3 ms), retain 15>
      | {
      |   "sessionID" = 132640241507728
      |   "iManufacturer" = 1
      |   "bNumConfigurations" = 1
[... 1227 lines cut ...]
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] subpacket too large
[ 1.044258] [0000340b] libusb: debug [libusb_submit_transfer] transfer 0x7f882a02cec8
[ 1.044288] [0000340b] libusb: debug [ep_to_pipeRef] converting ep address 0x84 to pipeRef and interface
[ 1.044294] [0000340b] libusb: debug [ep_to_pipeRef] pipe 1 on interface 1 matches
[ 1.044764] [0000340b] libusb: debug [libusb_get_next_timeout] next timeout in 0.952308s
[ 1.044773] [0000340b] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 1.044778] [0000340b] libusb: debug [handle_events] poll() 1 fds with timeout in 100ms
[ 1.058325] [00002507] libusb: debug [darwin_async_io_callback] an async io operation has completed
[ 1.058430] [0000340b] libusb: debug [handle_events] poll() returned 1
[ 1.058453] [0000340b] libusb: debug [handle_events] caught a fish on the event pipe
[ 1.058465] [0000340b] libusb: debug [darwin_handle_transfer_completion] handling isoc completion with kernel status 0
[ 1.058482] [0000340b] libusb: debug [usbi_handle_transfer_completion] transfer 0x7f88290226c8 has callback 0x10840a4c0
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] subpacket too large
.....`

any ideas?

best regards

xlz, 2016-02-24:
Add a line here https://github.com/OpenKinect/libfreenect2/blob/master/src/depth_packet_stream_parser.cpp#L72
```cpp
LOG_DEBUG << "length " << in_length;
```
And post what you get.

Anyway it's a change in your OS that breaks it, I'm not sure there is much we can do.

EnvnHash, 2016-02-24:
Thanks for the quick reply!!!

I added the line, now i get:

[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 14 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @0:3 with serial 016293740947
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 1024
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 016293740947
device firmware: 4.0.3912.0
[Debug] [DepthPacketStreamParser] length 62464
[Debug] [DepthPacketStreamParser] length 63488
[Debug] [DepthPacketStreamParser] length 64512
[Debug] [DepthPacketStreamParser] length 0
[Debug] [DepthPacketStreamParser] length 62464
[Debug] [DepthPacketStreamParser] length 63488
[Debug] [DepthPacketStreamParser] length 64512
[Debug] [DepthPacketStreamParser] length 0
[Debug] [DepthPacketStreamParser] length 62464
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 26928
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] length 28976
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] length 31024
[Debug] [DepthPacketStreamParser] length 62464
[Debug] [DepthPacketStreamParser] length 63488
[Debug] [DepthPacketStreamParser] length 64512
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[... 300 lines cut ...]
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 31024
[Debug] [DepthPacketStreamParser] length 62464
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 26928
[Debug] [DepthPacketStreamParser] length 64512
[Debug] [DepthPacketStreamParser] length 0
[Debug] [DepthPacketStreamParser] length 62464
[Debug] [DepthPacketStreamParser] length 63488
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] length 33792
[Debug] [DepthPacketStreamParser] subpacket too large
[Debug] [DepthPacketStreamParser] length 33792

xlz, 2016-02-24:
Can you remove that line, and add another line here https://github.com/OpenKinect/libfreenect2/blob/master/src/libfreenect2.cpp#L657
```
LOG_DEBUG << "max_iso_packet_size " << max_iso_packet_size;
```
And tell us what is your max_iso_packet_size?

EnvnHash, 2016-02-24:
[Debug] [Freenect2DeviceImpl] max_iso_packet_size 33792

xlz, 2016-02-24:
It's a USB driver bug. There is nothing we can do here.

Each buffer has size of 33792. It's writing size of something like 62464 bytes into each buffer.

EnvnHash, 2016-02-24:
Do I get it right, that you mean it´s related to my thunderbolt / USB3.0 adapter? ...or in other words, trying other versions of libusb probably won´t help here, right?

xlz, 2016-02-24:
Probably not. Maybe you can report a bug with libusb, but this seems to kernel problem.

EnvnHash, 2016-02-24:
This was Os X 10.11.3... So I´ll try on Debian now... Thanks so much!!!

closed 2016-02-24 by xlz

RyanGordon, 2016-02-25:
Do you think it's worthy of listing this as a known issue?

xlz, 2016-02-25:
I added this https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting#thunderbolt--usb-30-adaptor

But this is not entirely consistent. It did work on 10.10.

Also, it seems USB hubs are also troublesome but there is no conclusion or definite report on what exactly doesn't work.

RyanGordon, 2016-02-25:
Ah perfect - Yes, it is too bad those cause instability especially for systems like the Tegra TK1

poqudrof, 2016-08-01:
Hello, 

I just read this thread and it seems that I also have this problem (not exactly), and a very bad solution. 

I also have a `iso_packet_size   33792`, but  libfreenect2 was working with a previous release... So I checked it out and it is still working. I tried to figure out the differences between the two versions 
but they are quite huge.  

Here is the version working for me: 
https://github.com/jfrey-xx/libfreenect2/tree/master

So, it might be a hardware problem, or a new bug... 

Cheers, 

Jeremy.