OpenKinect/libfreenect2 #593: windows: poll fails because of too many pollfds with two Kinects in one context

xlz, 2016-02-25:
Reported by @SteveSmithStyku https://github.com/OpenKinect/libfreenect2/issues/97#issuecomment-188553614

Two instances of Protonect.exe 030260350347 and Protonect.exe 181369735247 work.

Accessing two Kinects in one program doesn't work, as in
```
[ 2.017330] [00000f18] libusb: debug [handle_events] poll fds modified, reallocating
[ 2.017330] [00000f18] libusb: debug [handle_events] poll() 82 fds with timeout in 983ms
[ 2.017330] [00000f18] libusb: debug [handle_events] poll() returned -1
[ 2.017330] [00000f18] libusb: error [handle_events] poll failed -1 err=5
[ 2.017330] [00000f18] libusb: error [sync_transfer_wait_for_completion] libusb_handle_events failed: LIBUSB_ERROR_IO, cancelling transfer and retrying
```

libusb uses `WaitForMultipleObjects()`, which requires a maximum number of fds being MAXIMUM_WAIT_OBJECTS (64). But libusb doesn't limit the number of fds per context.

xlz, 2016-02-25:
@SteveSmithStyku

Can you try this:
* Change https://github.com/OpenKinect/libfreenect2/blob/master/src/libfreenect2.cpp#L54  `NUM_XFERS` to 24, and
* Change https://github.com/OpenKinect/libfreenect2/blob/master/src/libfreenect2.cpp#L664 to `rgb_transfer_pool_.allocate(8, 0x4000);`

?

SteveSmithStyku, 2016-02-25:
@xlz I get a different error:

[Error] [usb::TransferPool] too few idle transfers!

one for each device

xlz, 2016-02-25:
I forgot here: https://github.com/OpenKinect/libfreenect2/blob/master/src/libfreenect2.cpp#L756

Change `rgb_transfer_pool_.submit(20)` to `rgb_transfer_pool_.submit(8)`

SteveSmithStyku, 2016-02-25:
@xlz Success! I am able to see streams from both sensors now. Thank you!

Framerate is still very low, however. I'll try using non-default packet pipelines and see if that improves the performance of the example. Other suggestions welcome.

xlz, 2016-02-25:
You can compare the performance with the benchmark here https://github.com/OpenKinect/libfreenect2/wiki/Performance

SteveSmithStyku, 2016-02-25:
@xlz My "timing loop" in my test code isn't special, and it's synchronous as well, so that probably has most to do with it. With OpenCL I'm getting fast reported depth processing times (about 2.5 ms) but slow reported rgb processing times (34 ms). Is rgb processing CPU based? If so using a faster CPU should help with that.

I will look into this further later today, but at a glance it appears my issue has been resolved.

xlz, 2016-02-25:
The "peformance" I referred to is the framerate information printed at runtime.

TurboJPEG is CPU based.

xlz, 2016-02-25:
@christiankerl I'm thinking of adjusting the transfer pool. Here the user has problem with 20 rgb transfers and 60 ir transfers. Is there a particular reason for the particular numbers of transfers. Is it fine to reduce it to something like 3 rgb transfers and 8 ir transfers?

christiankerl, 2016-02-27:
if I remember correctly the official driver submitted so many transfers according to some usb trace, but I don't know exactly. I guess the idea is to just have enough pending transfers to pick up all the data. If there are no packet losses the lower numbers are fine.

SteveSmithStyku, 2016-03-02:
Just wanted to note that on a high end machine, I was able to get 3 kinects working with the reduced numbers suggested by @xlz (8 ir transfers per sensor) without any frame drops, which suits my current use case fine. So long as the total number of transfers are 64 or fewer it appears it will work; any device that brings the total transfer count above 64 will show the error I experienced.

Perhaps these parameters can be set directly through library calls in the short term. In the long term, maybe it's worth implementing a method of initializing several devices at once, and setting these parameters under the hood. Either way I have no problem with just setting different hardcodes for my own needs.

Thanks again for your help!

xlz, 2016-03-07:
https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting#multiple-kinects

I may add a saner default later.

closed 2016-03-07 by xlz