OpenKinect/libfreenect2 #599: stop stream all images on macos10.11.3

hyrofumi, 2016-02-28:
I use libfreenect2 on ofxMultiKinectV2.
I compiled my program.
2-3minute is no problem, but after I have some error.

---------------------
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[TransferPool::submit] failed to submit transfer
[ error ] ofThread: - name: Thread 1 - Unable to completely waitForThread.
[Freenect2DeviceImpl] stopping...
[Freenect2DeviceImpl] disabling usb transfer submission...
[Freenect2DeviceImpl] canceling usb transfers...
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
[TransferPool::submit] transfer submission disabled!
---------------------

let me know way of solving.

xlz, 2016-02-28:
Overview Description:

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`
2. `uname -a`
3. `lsusb -t`
4. `lspci -nn`

Steps to Reproduce:

1. 
2. 
3. 

Actual Results:

Expected Results:

Reproducibility:

Additional Information:

xlz, 2016-03-18:
Close as INVALID.

closed 2016-03-18 by xlz