OpenKinect/libfreenect2 #600: Ubuntu 15.04 installs libusb 1.0.19 instead of 1.0.20

alexkiritz, 2016-02-29:
Overview Description:
When following the instructions to install on Ubuntu 15.04, the 'sudo apt-get install libusb-1.0-0-dev' command installs libusb-1.0.19 instead of the required later version  >= 1.0.20.

xlz, 2016-02-29:
Try the 14.04 instructions?

alexkiritz, 2016-02-29:
The 14.04 instructions are 14.04 specific. I got an error when trying on 15.04.

xlz, 2016-02-29:
What error?

xlz, 2016-02-29:
According to this, http://www.ubuntu.com/info/release-end-of-life, 15.04 is already end of life.

xlz, 2016-02-29:
Look, this is a fresh Ubuntu 15.04 installation:
```
(vivid)xlz@thinkpadx240:/tmp/libfreenect2/depends$ sudo dpkg -i debs/libusb*deb
Selecting previously unselected package libusb-1.0-0-dev:amd64.
(Reading database ... 21034 files and directories currently installed.)
Preparing to unpack .../libusb-1.0-0-dev_1.0.20-1_amd64.deb ...
Unpacking libusb-1.0-0-dev:amd64 (2:1.0.20-1) ...
Selecting previously unselected package libusb-1.0-0:amd64.
Preparing to unpack .../libusb-1.0-0_1.0.20-1_amd64.deb ...
Unpacking libusb-1.0-0:amd64 (2:1.0.20-1) ...
Setting up libusb-1.0-0:amd64 (2:1.0.20-1) ...
Setting up libusb-1.0-0-dev:amd64 (2:1.0.20-1) ...
Processing triggers for libc-bin (2.21-0ubuntu4) ...
(vivid)xlz@thinkpadx240:/tmp/libfreenect2/depends$ cat /etc/issue.net 
Ubuntu 15.04
```

Close as INVALID.

closed 2016-02-29 by xlz