OpenKinect/libfreenect2 #603: Bug (?) in libfreenect2/OpenNI2 enumerateDevices() -- too many?

smokhov, 2016-03-07:
- Windows 8.1
- libfreenect2 with the OpenNI2 driver contributed by @hanyazou from the main repo
- 2x Kinects

In the modified `UserViewer` code sample from NiTE samples I am trying to enumerate devices using the following code snippet:

```
		openni::OpenNI::enumerateDevices(&devices);

		m_iDeviceCount = devices.getSize();

		printf("Device count: %d\n", m_iDeviceCount);

		for (int i = 0; i < devices.getSize(); ++i)
		{
			printf("Device \"%s\" already connected\n", devices[i].getUri());
		}
```

I get:

```
...
Device count: 6
Device "freenect2://0" already connected
Device "freenect2://0?depth-size=640x480" already connected
Device "freenect2://0?depth-size=512x424" already connected
Device "freenect2://1" already connected
Device "freenect2://1?depth-size=640x480" already connected
Device "freenect2://1?depth-size=512x424" already connected
...
```

Is it expected to get 3 "devices" per Kinect or is it bug? Intuitively I'd expect enumerate only two devices, `freenect2://0` and `freenect2://1`. I was at first fooled by this. `open()` appears to successfully open the first 3 of the 6, but fails with the last 3 (in the same process), but this a separate issue I will comment on elsewhere.

It appears the `?depth-size` "devices" are added in `openni2/DeviceDriver.cpp`, in `initialize()`?

```
      for (int i = 0; i < freenect2.enumerateDevices(); i++)
      {
        std::string uri = devid_to_uri(i);
        const char* modes_c[] = {
          "",
          "?depth-size=640x480",
          "?depth-size=512x424",
        };
        std::vector<std::string> modes(modes_c, modes_c + 3);

        WriteMessage("Found device " + uri);

        for (unsigned i = 0; i < modes.size(); i++) {
          register_uri(uri + modes[i]);
        }
```

hanyazou, 2016-03-07:
This is not a bug.

smokhov, 2016-03-07:
@hanyazou OK, so this a feature that is expected to be, correct? :) If this is expected, then I guess needs to be mentioned somewhere that when using `openni::OpenNI::enumerateDevices()` as opposed to `freenect2.enumerateDevices()`, they do not the same semantics. (Though this is not the impression I am getting from the OpenNI2 docs). Please feel free to resolve this, if this is not an issue or consider a documentation bug? :)

hanyazou, 2016-03-12:
@smokhov Could you please point the description which tells the API semantics?

smokhov, 2016-03-12:
- @hanyazou, [here](http://com.occipital.openni.s3.amazonaws.com/OpenNI_Programmers_Guide.pdf), e.g., 

> The function OpenNI::enumerateDevices() returns a list of all available devices connected to the system.
...
> If multiple sensors are attached to the system, then you should first call
OpenNI::enumerateDevices() to obtain a list of all active devices.  Then, find the desired device
in the list, and obtain its URI by calling DeviceInfo:getUri().  Use the output from this function as
the calling parameter for Device:open() in order to open that specific device.

I guess it is a bit vague though intuitively one would assume a device is a physical Kinect object in this case rather than each stream source with a given resolution. My intuition may be wrong of course for OpenNI2.

chengzijiazu, 2016-06-12:
Hi smokhov ,
I am Songbin,I am new to libfreenect2(also I am new to use multi-kinect2),could you give me a code about this libfreenect, just for study.Thanks! yesongbin66@gmail.com

smokhov, 2016-08-23:
@chengzijiazu -- you'd have to talk to CAS RCSC for that code. I described our methodology in the poster you've found. However, depends on what you need to do, if you don't need skeletal data from multiple Kinects, `Protonect` can be easily extended to initialize and work with multiple Kinects for their IR, depth, RGB, etc. data by simply duplicating the calls and instances of the devices.

xlz, 2016-12-05:
NOTABUG

closed 2016-12-05 by xlz