OpenKinect/libfreenect2 #604: make error when building the actual protonect executable

Severn-Vergil, 2016-03-07:
Overview Description:

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`
9c80ccd usb: Issue reboot command on Mac OS X

2. `uname -a`
Linux gqy-desktop 3.16.0-62-generic #83~14.04.1-Ubuntu SMP Fri Feb 26 22:52:48 UTC 2016 i686 i686 i686 GNU/Linux

3. `lsusb -t`
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/9p, 480M
    |__ Port 2: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 4: Dev 5, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 4: Dev 5, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 3: Dev 3, If 0, Class=Vendor Specific Class, Driver=r8188eu, 480M
    |__ Port 5: Dev 4, If 0, Class=Human Interface Device, Driver=, 12M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/3p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M

4. `lspci -nn`
00:00.0 Host bridge [0600]: Intel Corporation Haswell-ULT DRAM Controller [8086:0a04] (rev 0b)
00:02.0 VGA compatible controller [0300]: Intel Corporation Haswell-ULT Integrated Graphics Controller [8086:0a16] (rev 0b)
00:03.0 Audio device [0403]: Intel Corporation Haswell-ULT HD Audio Controller [8086:0a0c] (rev 0b)
00:14.0 USB controller [0c03]: Intel Corporation 8 Series USB xHCI HC [8086:9c31] (rev 04)
00:16.0 Communication controller [0780]: Intel Corporation 8 Series HECI #0 [8086:9c3a] (rev 04)
00:16.3 Serial controller [0700]: Intel Corporation 8 Series HECI KT [8086:9c3d] (rev 04)
00:19.0 Ethernet controller [0200]: Intel Corporation Ethernet Connection I218-LM [8086:155a] (rev 04)
00:1b.0 Audio device [0403]: Intel Corporation 8 Series HD Audio Controller [8086:9c20] (rev 04)
00:1c.0 PCI bridge [0604]: Intel Corporation 8 Series PCI Express Root Port 1 [8086:9c10] (rev e4)
00:1c.3 PCI bridge [0604]: Intel Corporation 8 Series PCI Express Root Port 4 [8086:9c16] (rev e4)
00:1d.0 USB controller [0c03]: Intel Corporation 8 Series USB EHCI #1 [8086:9c26] (rev 04)
00:1f.0 ISA bridge [0601]: Intel Corporation 8 Series LPC Controller [8086:9c43] (rev 04)
00:1f.2 SATA controller [0106]: Intel Corporation 8 Series SATA Controller 1 [AHCI mode] [8086:9c03] (rev 04)
00:1f.3 SMBus [0c05]: Intel Corporation 8 Series SMBus Controller [8086:9c22] (rev 04)
02:00.0 Ethernet controller [0200]: Intel Corporation I210 Gigabit Network Connection [8086:1533] (rev 03)

Actual Results:
make
[  6%] Built target generate_resources_tool
[  6%] Building CXX object CMakeFiles/freenect2.dir/src/transfer_pool.cpp.o
/home/gqy/RGBD_ws/libfreenect2/src/transfer_pool.cpp: In member function ‘void libfreenect2::usb::TransferPool::cancel()’:
/home/gqy/RGBD_ws/libfreenect2/src/transfer_pool.cpp:138: error: ‘sleep_for’ is not a member of ‘libfreenect2::this_thread’
/home/gqy/RGBD_ws/libfreenect2/src/transfer_pool.cpp:145: error: ‘sleep_for’ is not a member of ‘libfreenect2::this_thread’
make[2]: *** [CMakeFiles/freenect2.dir/src/transfer_pool.cpp.o] error 1
make[1]: *** [CMakeFiles/freenect2.dir/all] error 2
make: *** [all] error 2

Additional Information:

gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/i686-linux-gnu/4.8/lto-wrapper
Target: i686-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 4.8.4-2ubuntu1~14.04.1' --with-bugurl=file:///usr/share/doc/gcc-4.8/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.8 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.8 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libmudflap --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.8-i386/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.8-i386 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.8-i386 --with-arch-directory=i386 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-targets=all --enable-multiarch --disable-werror --with-arch-32=i686 --with-multilib-list=m32,m64,mx32 --with-tune=generic --enable-checking=release --build=i686-linux-gnu --host=i686-linux-gnu --target=i686-linux-gnu
Thread model: posix
gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04.1) 

cmake .. -DENABLE_CXX11=ON -DCMAKE_INSTALL_PREFIX=$HOME/freenect2
-- using stdlib as threading library
-- checking for modules 'libva;libva-drm'
--   package 'libva' not found
--   package 'libva-drm' not found
-- Could NOT find TegraJPEG (missing:  TegraJPEG_LIBRARIES TegraJPEG_INCLUDE_DIRS TegraJPEG_L4T_OK TegraJPEG_DRIVER_OK TegraJPEG_WORKS) 
CUDA_TOOLKIT_ROOT_DIR not found or specified
-- Could NOT find CUDA (missing:  CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY) 
-- Linking with these libraries: 
 /usr/lib/i386-linux-gnu/libusb-1.0.so
 /usr/lib/i386-linux-gnu/libturbojpeg.so.0
 /usr/lib/i386-linux-gnu/libglfw.so
 /usr/lib/i386-linux-gnu/libGL.so
 /usr/lib/i386-linux-gnu/libOpenCL.so
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- Configurating examples
-- Using in-tree freenect2 target
-- Feature list:
--   CUDA    no
--   CXX11    c++0x
--   Examples    yes
--   OpenCL    yes
--   OpenGL    yes
--   OpenNI2    yes
--   TegraJPEG    no
--   Threading    std::thread
--   TurboJPEG    yes
--   VAAPI    no
--   VideoToolbox    no (Apple only)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/gqy/RGBD_ws/libfreenect2/build

dpkg -l|grep libusb
ii  libgusb2:i386                                         0.1.6-5                                             i386         GLib wrapper around libusb1
ii  libusb-0.1-4:i386                                     2:0.1.12-23.3ubuntu1                                i386         userspace USB programming library
ii  libusb-1.0-0:i386                                     2:1.0.20-1                                          i386         userspace USB programming library
ii  libusb-1.0-0-dev:i386                                 2:1.0.20-1                                          i386         userspace USB programming library development files
ii  libusb-1.0-doc                                        2:1.0.17-1ubuntu2                                   all          documentation for userspace USB programming
ii  libusb-dev                                            2:0.1.12-23.3ubuntu1                                i386         userspace USB programming library development files
ii  libusbmuxd2                                           1.0.8-2ubuntu1                                      i386         USB multiplexor daemon for iPhone and iPod Touch devices - library

Severn-Vergil, 2016-03-07:
Thank you in advance! Your advice would helps me a lot.

xlz, 2016-03-07:
Please attach `build/CMakeFiles/CMakeOutput.log`.

Severn-Vergil, 2016-03-08:
[](url)

Severn-Vergil, 2016-03-09:
[CMakeOutput.log.txt](https://github.com/OpenKinect/libfreenect2/files/164524/CMakeOutput.log.txt)

xlz, 2016-03-09:
It should detect CXX11 support but it instead detected CXX0X.  I don't know why.

Anyway the direct problem is this http://stackoverflow.com/questions/4438084/stdthis-threadsleep-for-and-gcc

You can disable it with `-DENABLE_CXX=OFF`, or use workarounds in the above link.

xlz, 2016-03-23:
Maybe I'll try 32bit userspace for Linux and Windows sometime.