OpenKinect/libfreenect2 #605: Problem for the 3rd Kinect V2 connecting in Windows 10+UsbDk

jb4652, 2016-03-07:
Windows 10, UsbDk, fine for 2 Kinect V2 connecting via USB 3 port on motherboard (not PCI-E card). While connecting the 3rd one, error info repeated:
Driver file operation error. GetOverlappedResult failed (The system cannot find the file specified.  Error code = 2)
and that cause the other connected two interrupted.

>Protonect.exe 500769142442 cuda -noviewer -norgb
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: Protonect.exe [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [CudaDepthPacketProcessorImpl] device 0: GeForce GTX 980 @ 1215MHz Memory 4096MB
[Info] [CudaDepthPacketProcessorImpl] selected device 0
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 15 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @103:2 with serial 500769142442
Driver file operation error. DeviceIoControl failed (Cannot create a file when that file already exists.  Error code = 183)
[Error] [Freenect2Impl] failed to open Kinect v2: @104:2 LIBUSB_ERROR_OTHER Other error
Driver file operation error. DeviceIoControl failed (Cannot create a file when that file already exists.  Error code = 183)
[Error] [Freenect2Impl] failed to open Kinect v2: @105:2 LIBUSB_ERROR_OTHER Other error
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 500769142442
device firmware: 4.0.3911.0
Driver file operation error. GetOverlappedResult failed (The system cannot find the file specified.  Error code = 2)
Driver file operation error. GetOverlappedResult failed (The system cannot find the file specified.  Error code = 2)
Driver file operation error. GetOverlappedResult failed (The system cannot find the file specified.  Error code = 2)

xlz, 2016-03-07:
In this case you can use the libusbk backend by setting `LIBUSB_USE_USBDK` to `OFF` in CMake.

I'll look into it.

xlz, 2016-03-07:
OK, I think this actually has nothing to do usbdk. You start to have errors with the third Kinects. There is a limit on file descriptors that can be polled at the same time. Maybe it's the same issue as #593.

Can you try setting environment variables (assuming powershell):
```
$env:LIBFREENECT2_RGB_TRANSFER_SIZE=1048576
$env:LIBFREENECT2_RGB_TRANSFERS=3
#env:LIBFREENECT2_IR_TRANSFERS=8
```
?

jb4652, 2016-03-08:
With default params, two (RGB+D) + one (RGB only) is OK. For depth processing, cpu, cl, gl, cuda have all been tested, no luck.
By changing the params, it seems that the "avg. time" varies (even with packets lost), but error
>Driver file operation error. GetOverlappedResult failed (The system cannot find the file specified.  Error code = 2)

remains.

xlz, 2016-03-08:
Do 3 Kinects work with libusbk backend?

What is the log with environment variable `LIBUSB_DEBUG=3`?

jb4652, 2016-03-08:
libusbk has not been tested yet.

With UsbDk, if using default params, by setting LIBUSB_DEBUG=3, I will get repeated
>libusb: error [handle_events] poll failed -1 err=5

But the rgb and depth stream still seems OK for 2 Kinects.

If setting
>LIBFREENECT2_IR_PACKETS=128
>LIBFREENECT2_IR_TRANSFERS=4 

while keeping LIBFREENECT2_RGB_TRANSFER_SIZE & LIBFREENECT2_RGB_TRANSFERS as default, error
>libusb: error [handle_events] poll failed -1 err=5

disappeared. However, while opening the 3rd Kinect, I get repeated
>libusb: error [windows_transfer_callback] detected I/O error 1627: [1627] Function failed during execution.

xlz, 2016-03-08:
`poll failed -1 err=5` is exactly the same issue as #593, and the workaround atm is to use environment variables LIBFREENECT2_IR_PACKETS etc.

Can you post a little bit more log for this:

> libusb: error [windows_transfer_callback] detected I/O error 1627: [1627] Function failed during execution.

? I need the context of this line.