OpenKinect/libfreenect2 #606: Beignet-opencl-icd no supported GPU found

cpaniagua1, 2016-03-08:
Hello,

I'm working with Intel Iris 6100 (Brodwell GT3) graphics and Ubuntu 14.04. I followed the installation instructions for Linux and everything was fine but when I run ./Proctonect cl  I have problems with OpenCL. 

I get the following error:
beignet-opencl-icd no supported GPU found

I read at the end of this thread [https://github.com/OpenKinect/libfreenect2/issues/109]() that there were problems with Beignet and Brodwell. 

Can somebody help me with this? 
Thanks in advance.

xlz, 2016-03-08:
We'll create a ppa for beignet 1.1.1

https://github.com/OpenKinect/libfreenect2/issues/558#issuecomment-190405220

closed 2016-03-08 by xlz

cpaniagua1, 2016-03-08:
Would you mind telling me when it will be create?
Is there any way to install beignet 1.1.1 without the ppa?

xlz, 2016-03-08:
> For the record, I built a ported beignet 1.1.1 to Ubuntu 14.04 with the beignet sources (1.1.1) from Ubuntu 16.04, changed debian/control to remove build dependencies on clang, llvm, installed clang-3.6 libclang-3.6-dev, and apt-src build beignet-dev built some beignet 1.1.1 debs.

floe, 2016-03-10:
Try this PPA: https://launchpad.net/~floe/+archive/ubuntu/beignet (only AMD64 at the moment).

cpaniagua1, 2016-03-10:
Thank your very much!
It solved my problem.