OpenKinect/libfreenect2 #607: Stream mulitple Kinect

siriusye, 2016-03-09:
Sorry this is not really a bug, but i could not find other places to get help. I successfully connect 2 kinect v2 on one computer using libfreenect2, They can be opened with openDevice. However, I notice that they cannot stream data simultaneously. The problem seems on the pipline step. Only one gpu pipline can be allocated to one device. One device with gpu piplien, the other with cpu pipline will not work. My computer only have a single graphics card, dose this mean only one stream can be processed? 

Thanks very much!

xlz, 2016-03-09:
You should create one packet pipeline for each device.

siriusye, 2016-03-10:
I did new a instance of packet pipeline for each device. Could I get a confirmation about:
If everything is right, the driver can stream 2 kinect v2 data on a single graphic card high end computer?

xlz, 2016-03-10:
yes

siriusye, 2016-03-10:
it is working now. I did render two images to the same framebuffer..Thanks very much mate!

closed 2016-03-10 by xlz