OpenKinect/libfreenect2 #609: OS X USB Access Denied

johnmastri, 2016-03-11:
LIBUSB_DEBUG=3 ./bin/Protonect
Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
libusb: warning [darwin_cache_device_descriptor] could not retrieve device descriptor 05ac:8510: device not responding (e00002ed). skipping device
libusb: warning [darwin_cache_device_descriptor] could not retrieve device descriptor 05ac:0262: pipe is stalled (e000404f). skipping device
libusb: warning [darwin_cache_device_descriptor] could not retrieve device descriptor 05ac:8286: pipe is stalled (e000404f). skipping device
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 16 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @20:26 with serial 123624340447
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
libusb: info [darwin_claim_interface] no interface found; setting configuration: 1
libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
[Error] [protocol::UsbControl] failed to claim interface with IrInterfaceId(=1)! LIBUSB_ERROR_ACCESS Access denied (insufficient permissions). Try debugging with environment variable: export LIBUSB_DEBUG=3 .
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
libusb: warning [darwin_release_interface] USBInterfaceClose: no connection to an IOService
libusb: warning [darwin_close] USBDeviceClose: no connection to an IOService
[Info] [Freenect2DeviceImpl] closed
[Error] [Freenect2Impl] failed to open Kinect v2: @20:26
failure opening device!
Johns-MacBook-Pro:libfreenect2 johnmastri$ 

(Port 2 of Hub at 0x15000000)
utun_ctl_connect: creating interface utun0
USBF:    11. 40    The IOUSBFamily was not able to enumerate a device.
utun_start: ifnet_disable_output returned error 12
USBF:    11.787    The IOUSBFamily is having trouble enumerating a USB device that has been plugged in.  It will keep retrying.  (Port 2 of Hub at 0x15000000)
USBF:    12. 87    The IOUSBFamily was not able to enumerate a device.
USBF:    12.669    The IOUSBFamily is having trouble enumerating a USB device that has been plugged in.  It will keep retrying.  (Port 2 of Hub at 0x15000000)
USBF:    12.669    The IOUSBFamily gave up enumerating a USB device after 10 retries.  (Port 2 of Hub at 0x15000000)
USBF:    12.669    The IOUSBFamily was not able to enumerate a device.
AppleKeyStore:Sending lock change 0 for handle -501
IO80211AWDLPeerManager::setAwdlPeerTrafficRegistration Clearing AirPlay peer 00:00:00:00:00:00
Google Chrome (map: 0xffffff801b8ccf00) triggered DYLD shared region unnest for map: 0xffffff801b8ccf00, region 0x7fff8b200000->0x7fff8b400000. While not abnormal for debuggers, this increases system memory footprint until the target exits.
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny file-read-metadata /Library
Sandbox: appleeventsd(54) deny mach-lookup com.apple.ocspd
Sandbox: appleeventsd(54) deny mach-lookup com.apple.ocspd
Sandbox: appleeventsd(54) deny mach-lookup com.apple.ocspd
[... 101 lines cut ...]
USB Sound assertion in AppleUSBAudioDevice at line 674
USB Sound assertion in AppleUSBAudioDevice at line 293
process Dropbox[937] thread 8712 caught burning CPU! It used more than 50% CPU (Actual recent usage: 72%) over 180 seconds. thread lifetime cpu usage 90.083201 seconds, (80.444686 user, 9.638515 system) ledger info: balance: 90016884385 credit: 90016884385 debit: 0 limit: 90000000000 (50%) period: 180000000000 time since last refill (ns): 124532650028 
CODE SIGNING: cs_invalid_page(0x103506000): p=1024[spindump] final status 0x2000000, allowing (remove VALID) page
USBF:    208.192    The IOUSBFamily is having trouble enumerating a USB device that has been plugged in.  It will keep retrying.  (Port 1 of Hub at 0x14200000)
USBF:    208.493    The IOUSBFamily was not able to enumerate a device.
USBF:    209. 27    The IOUSBFamily is having trouble enumerating a USB device that has been plugged in.  It will keep retrying.  (Port 1 of Hub at 0x14200000)
USBF:    209.328    The IOUSBFamily was not able to enumerate a device.
USBF:    209.859    The IOUSBFamily is having trouble enumerating a USB device that has been plugged in.  It will keep retrying.  (Port 1 of Hub at 0x14200000)
USBF:    210.160    The IOUSBFamily was not able to enumerate a device.
USBF:    210.692    The IOUSBFamily is having trouble enumerating a USB device that has been plugged in.  It will keep retrying.  (Port 1 of Hub at 0x14200000)
USBF:    211.195    The IOUSBFamily was not able to enumerate a device.
Google Chrome He (map: 0xffffff802ae21000) triggered DYLD shared region unnest for map: 0xffffff802ae21000, region 0x7fff8b200000->0x7fff8b400000. While not abnormal for debuggers, this increases system memory footprint until the target exits.
Google Chrome He (map: 0xffffff801d8a6b40) triggered DYLD shared region unnest for map: 0xffffff801d8a6b40, region 0x7fff8b200000->0x7fff8b400000. While not abnormal for debuggers, this increases system memory footprint until the target exits.
Google Chrome He (map: 0xffffff801d8a6b40) triggered DYLD shared region unnest for map: 0xffffff801d8a6b40, region 0x7fff8b200000->0x7fff8b400000. While not abnormal for debuggers, this increases system memory footprint until the target exits.
Google Chrome He (map: 0xffffff801d8a6c30) triggered DYLD shared region unnest for map: 0xffffff801d8a6c30, region 0x7fff8b200000->0x7fff8b400000. While not abnormal for debuggers, this increases system memory footprint until the target exits.
Google Chrome He (map: 0xffffff801d8a6780) triggered DYLD shared region unnest for map: 0xffffff801d8a6780, region 0x7fff8b200000->0x7fff8b400000. While not abnormal for debuggers, this increases system memory footprint until the target exits.
Google Chrome He (map: 0xffffff802ae21000) triggered DYLD shared region unnest for map: 0xffffff802ae21000, region 0x7fff8b200000->0x7fff8b400000. While not abnormal for debuggers, this increases system memory footprint until the target exits.
Google Chrome He (map: 0xffffff80218ac870) triggered DYLD shared region unnest for map: 0xffffff80218ac870, region 0x7fff8b200000->0x7fff8b400000. While not abnormal for debuggers, this increases system memory footprint until the target exits.
Google Chrome He (map: 0xffffff801d8a6780) triggered DYLD shared region unnest for map: 0xffffff801d8a6780, region 0x7fff8b200000->0x7fff8b400000. While not abnormal for debuggers, this increases system memory footprint until the target exits.
Google Chrome He (map: 0xffffff801894f000) triggered DYLD shared region unnest for map: 0xffffff801894f000, region 0x7fff8b200000->0x7fff8b400000. While not abnormal for debuggers, this increases system memory footprint until the target exits.
[image]

￼￼
[image]

johnmastri, 2016-03-11:
Device is showing up on both buses - and XBOX NUI Sensor is under high speed - tried the unplug fix several times:

[image]

xlz, 2016-03-11:
https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting#verify-kinect-is-indeed-connected-to-a-usb-3-port

johnmastri, 2016-03-11:
Hey @xlz - the device shows up in both hi-speed and superspeed - but the Xbox NUI sensor only appears under Hi-Speed (see above screen capture).  This is after I followed the troubleshooting guide of plugging the kinect in without power then adding the power to it.

xlz, 2016-03-11:
See the exact words:

> verify "Xbox NUI Sensor" is under "USB 3.0 SuperSpeed Bus" not "High-Speed Bus"

"NuiSensor Adaptor" is not "the device". It's the power brick thingy.

If this is not verified, you don't need to run Protonect at all, because it will just not work. I don't know enough about Mac to offer any other suggestions.

xlz, 2016-12-05:
Probably NOTOURBUG.

closed 2016-12-05 by xlz