OpenKinect/libfreenect2 #610: Running libfreenect2 headless

JustSpiffy, 2016-03-11:
I am trying to use libfreenect2 with kinect2_bridge (ros). With a monitor attached everything works great. When I move the machine (with tthe kinect and libfreenect2) on to a platform without my monitor, I ssh in, and it no longer works.

It doesn't seem like I'm using any windowing, and I don't think I'm enabling OpenGL. Does libfreenect2 requires an active frame buffer (aka an X windowing session) to work? 

I did try ssh -X but that didn't seem to help.

Attached is my error log:

libGL error: No matching fbConfigs or visuals found
libGL error: failed to log driver: swrast
[Error] [OpenGLDepthPacketProcessorImpl] GLFW error 65543 GLX: Failed to create context: BadMatch (invalid parameter attributes)
[Error] [OpenGLDepthPacketProcessor] Failed to create opengl window

xlz, 2016-03-11:
The OpenGL processor needs an X window session. You may use `DISPLAY=:0.0`, or use OpenCL or CUDA processors which don't require X.

floe, 2016-03-11:
The opengl depth packet processor does indeed need an X11 environment. Use the opencl DPP instead (try `Protonect -cl`).

closed 2016-03-15 by xlz