OpenKinect/libfreenect2 #613: status 0x090000: 1024 does not change

hanyazou, 2016-03-12:
Overview Description:
The Protonect wait for the changes of the status 0x090000: 1024 forever.

Version, Platform, and Hardware Bug Found:

1. `git log -1 --oneline`
>9c80ccd usb: Issue reboot command on Mac OS X

2. `uname -a`
>Darwin iMac.local 15.3.0 Darwin Kernel Version 15.3.0: Thu Dec 10 18:40:58 PST 2015; root:xnu-3248.30.4~1/RELEASE_X86_64 x86_64

3. `lsusb -t`
> Bus 000.Dev 001: USB 2.0 Bus,
  Bus 000.Dev 001: USB 2.0 Bus,
  Bus 000.Dev 001: USB 3.0 Bus,
    |__ Bus 020.Dev 003: NuiSensor Adaptor, 480M
    |__ Bus 020.Dev 006: iPad, 480M
    |__ Bus 020.Dev 005: NuiSensor Adaptor, 5G
        |__ Bus 020.Dev 024: Xbox NUI Sensor, 5G
    |__ Bus 026.Dev 001: Hub, 480M
        |__ Bus 026.Dev 002: FaceTime HD Camera (Built-in), 480M
    |__ Bus 029.Dev 001: Hub, 480M
        |__ Bus 029.Dev 002: Hub, 480M
            |__ Bus 029.Dev 003: BRCM20702 Hub, 12M
                |__ Bus 029.Dev 005: Bluetooth USB Host Controller, 12M

4. `lspci -nn`
>bash: lspci: command not found

Steps to Reproduce:

1. Just execute the Protonect repeatedly
> for i in {0..10000}; do ./bin/Protonect -frames 1; done

2. It stops 
>Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
	[-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) Core(TM) i7-3770 CPU @ 3.40GHz (CPU)[Intel]
[Info] [OpenCLDepthPacketProcessorImpl]   1: GeForce GTX 675MX (GPU)[NVIDIA]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: GeForce GTX 675MX (GPU)[NVIDIA]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @20:17 with serial 005697142447
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 1024

Actual Results:
> [Debug] [Freenect2DeviceImpl] status 0x090000: 1024

Expected Results:
>Version: 0.2.0
Environment variables: LOGFILE=<protonect.log>
Usage: ./bin/Protonect [-gpu=<id>] [gl | cl | cuda | cpu] [<device serial>]
        [-noviewer] [-norgb | -nodepth] [-help] [-version]
        [-frames <number of frames to process>]
To pause and unpause: pkill -USR1 Protonect
[Info] [OpenCLDepthPacketProcessorImpl]  devices:
[Info] [OpenCLDepthPacketProcessorImpl]   0: Intel(R) Core(TM) i7-3770 CPU @ 3.40GHz^@ (CPU)[Intel^@]
[Info] [OpenCLDepthPacketProcessorImpl]   1: GeForce GTX 675MX^@ (GPU)[NVIDIA^@]
[Info] [OpenCLDepthPacketProcessorImpl] selected device: GeForce GTX 675MX^@ (GPU)[NVIDIA^@]
[Info] [OpenCLDepthPacketProcessorImpl] building OpenCL program...
[Info] [OpenCLDepthPacketProcessorImpl] OpenCL program built successfully
[Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 13 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @20:24 with serial 005697142447
[Info] [Freenect2Impl] found 1 devices
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] opened
[Info] [Freenect2DeviceImpl] starting...
[Debug] [Freenect2DeviceImpl] status 0x090000: 1024
[Debug] [Freenect2DeviceImpl] status 0x090000: 9729
[Debug] [Freenect2DeviceImpl] status 0x090000: 9731
[Info] [Freenect2DeviceImpl] submitting rgb transfers...
[Info] [Freenect2DeviceImpl] submitting depth transfers...
[Info] [Freenect2DeviceImpl] started
device serial: 005697142447
device firmware: 4.0.3912.0
[Debug] [DepthPacketStreamParser] not all subsequences received 0
[Debug] [DepthPacketStreamParser] not all subsequences received 1008
[Debug] [DepthPacketStreamParser] skipping depth packet
[Debug] [DepthPacketStreamParser] skipping depth packet
[Info] [Freenect2DeviceImpl] stopping...
[Info] [Freenect2DeviceImpl] canceling rgb transfers...
[Info] [Freenect2DeviceImpl] canceling depth transfers...
[Info] [Freenect2DeviceImpl] stopped
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] releasing usb interfaces...
[Info] [Freenect2DeviceImpl] deallocating usb transfer pools...
[Info] [Freenect2DeviceImpl] closing usb device...
[Info] [Freenect2DeviceImpl] closed
[Info] [Freenect2DeviceImpl] closing...
[Info] [Freenect2DeviceImpl] already closed, doing nothing

Reproducibility:
Very rare. But once this problem occurred this occurs repeatedly.

Additional Information:
We can ignore this status checking. This patch resolve the problem.

Refer https://github.com/OpenKinect/libfreenect2/pull/597#issuecomment-190018314, https://github.com/OpenKinect/libfreenect2/issues/539#issuecomment-189577849 and #612.

>diff --git a/src/libfreenect2.cpp b/src/libfreenect2.cpp
index 194efe7..e5962f4 100644
--- a/src/libfreenect2.cpp
+++ b/src/libfreenect2.cpp
@@ -713,7 +713,8 @@ bool Freenect2DeviceImpl::startStreams(bool enable_rgb, bool enable_depth)
   if (!command_tx_.execute(SetModeEnabledWith0x00640064Command(nextCommandSeq()), result)) return false;
   if (!command_tx_.execute(SetModeDisabledCommand(nextCommandSeq()), result)) return false;

>\-  for (uint32_t status = 0, last = 0; (status & 1) == 0; last = status)
\+  int timeout = 50; // about 5 seconds (100ms x 50)
\+  for (uint32_t status = 0, last = 0; (status & 1) == 0 && 0 < timeout; last = status, timeout--)
   {
     if (!command_tx_.execute(ReadStatus0x090000Command(nextCommandSeq()), result)) return false;
     status = Status0x090000Response(result).toNumber();
@@ -722,6 +723,9 @@ bool Freenect2DeviceImpl::startStreams(bool enable_rgb, bool enable_depth)
     if ((status & 1) == 0)
       this_thread::sleep_for(chrono::milliseconds(100));
   }
\+  if (timeout == 0) {
\+    LOG_DEBUG << "status 0x090000: timeout";
\+  }

>   if (!command_tx_.execute(InitStreamsCommand(nextCommandSeq()), result)) return false;

xlz, 2016-03-12:
I think you had a PR for this which was rejected. I think the straightforward fix is add a limit to the for loop, say 100. This means about 10 seconds timeout after which the status check is ignored.

xlz, 2016-03-12:
It looks like your diff is showing something similar. You can create a PR with the diff. Simple change is easier ti accept.

closed 2016-03-14 by xlz