OpenKinect/libfreenect2 #615: Multiple Kinect: Not enough bandwidth

chihyaoma, 2016-03-15:
Overview Description: 

First of all, I would like to thank you guys for all the hard works and your fast replies on reported issues. I am trying to use two Kinect V2 sensors in one program, but with no luck. I have basically read through all the issues reported before related to using multiple Kinect sensors. Yet, I still have one question: **can we connect two Kinect sensors using built-in USB 3.0 ports on the motherboard?** 

I understand that this question depends on how much bandwidth is supported on your motherboard, but can we have a number for the bandwidth is required corresponding to the number of Kinect devices? 

Here are some specs for my PC running Ubuntu 14.04 LTS:
CPU: Intel® Core™ i7-4790K CPU @ 4.00GHz × 8
Motherboard: ASUS Z97-A
Graphics: GeForce GTX 970

Basically, my goal is to have a single program displaying color and depth streams from two Kinect sensors. However, no matter what I tried, using `LIBUSB_DEBUG=3` the `dmesg` will give the result below: 
> Not enough bandwidth for new device state

Below are what I have already tried: 

1. Two instances with Protonect exmaple: 
When using the example program with commands: ./Protonect 007618650247 and ./Protonect 114231241047. Note that `007618650247` and `114231241047` are the serial number of my Kinect Sensors. 

2. Disable depth stream to lower bandwidth requirement
I have tried to lower the bandwidth requirement by disabling depth with `-nodepth` for both instances above. 

3. Modified grub file like suggested in #500 #97 #7 
`GRUB_CMDLINE_LINUX_DEFAULT="quiet splash usbcore.usbfs_memory_mb=64"`

4. Try increasing USBFS buffer size to 64, 128, 256, 512...
I changed the number in ` /sys/module/usbcore/parameters/usbfs_memory_mb` to 64, 128, 256, 512, and so on, with no luck. 

4. Modified libfreenect2.cpp followed by #593 
Change https://github.com/OpenKinect/libfreenect2/blob/master/src/libfreenect2.cpp#L54 NUM_XFERS to 24
Change https://github.com/OpenKinect/libfreenect2/blob/master/src/libfreenect2.cpp#L664 to rgb_transfer_pool_.allocate(8, 0x4000);
Change https://github.com/OpenKinect/libfreenect2/blob/master/src/libfreenect2.cpp#L756
rgb_transfer_pool_.submit(20) to rgb_transfer_pool_.submit(8)

I understand that the problem seems to be that I don't have enough bandwidth on my motherboard. However, I have tried different methods to lower the requirement for the bandwidth. I expect it will work after reducing the bandwidth requirement. 

If all of the above modifications turns out to be correct, and I still can't make it work. I will purchase a PCI-E USB 3.0 controller. I just want to make sure if I am doing everything correctly before I purchase the controller. Again, thank you guys for all your hard works and helps!

xlz, 2016-03-15:
All these methods do not change the bandwidth requirements.

Only change to your hardware can increase the bandwidth. Each Kinect requires about 3Gbps on the PCIe bus. I don't know specifics about your motherboard but this is related to the motherboard/chipset configuration.

chihyaoma, 2016-03-15:
Yes, you are right. I expected to disable the depth steam will lower the bandwidth requirement, but it doesn't. I will purchase several USB 3.0 controllers to see if the problem can be solved. 

Thank you for the quick reply!

closed 2016-03-15 by xlz

gypsytearss, 2016-03-22:
Currently experiencing similar issues myself... @chihyaoma Did the new USB controllers fix it?

chihyaoma, 2016-03-22:
@gypsytearss Yes! I tried the USB 3.0 controller today, and the bandwidth problem was fixed. 

By the way, since there is one post discussing which USB 3.0 controllers have been proved to work, I am using **Anker® Uspeed USB 3.0 PCI-E Express Card** from Amazon. 
[http://www.amazon.com/Anker%C2%AE-Express-Connector-Desktops-Chipset/dp/B005ARQV6U](url)

gypsytearss, 2016-03-29:
@chihyaoma Awesome thanks for the reply and especially for the suggestion! We just ordered some and will test and reply with whether it worked for us soon

xlz, 2016-08-09:
@Severn-Vergil Sorry, I actually do not know if there is a way to check PCIe bandwidth despite having looked into this.