OpenKinect/libfreenect2 #617: failed to submit transfer:LIBUSB_ERROR_IO Input/Output Error

wangtingwei1993, 2016-03-15:
Hi,
   I am attempting to install libfreenect2 on jetson-tx1 with tegra 23.2.I install the package follow the instruction and succeed.But when I run the example(Protonect),I meet two problems.
1、[Error] [usb::TransferPool] failed to submit transfer:LIBUSB_ERROR_IO Input/Output Error.
2、[Error] [RgbPackketStreamParser] packetsize or sequence doesn't match!
Any suggestion will be appreciated!

xlz, 2016-03-15:
https://github.com/OpenKinect/libfreenect2/wiki/Troubleshooting#jetson-tx1-issues

closed 2016-03-15 by xlz

cforntu, 2016-06-21:
After I did what they said, downloading the xusb_sil_rel_fw-test.zip, the error is still there.
  [Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): unknown libva error
[Info] [Freenect2DeviceImpl] opening...
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
[Error] [protocol::UsbControl] failed to claim interface with ControlAndRgbInterfaceId(=0)! LIBUSB_ERROR_BUSY Resource busy. Try debugging with environment variable: export LIBUSB_DEBUG=3 .
after I run with LIBUSB_DEBUG=3, the output is 
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
[Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): unknown libva error
[Info] [Freenect2DeviceImpl] opening...
libusb: error [submit_iso_transfer] submiturb failed error -1 errno=16
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=16
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=16
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=16
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=16
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=16
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=16
[Error] [usb::TransferPool] failed to submit transfer: LIBUSB_ERROR_IO Input/Output Error

 Could you help me about it

ratman, 2016-09-09:
I had exactly the same problem a day ago, althrough with an older version of libfreenect2. I don't know which version you try to use but make sure that you install the libusb library from the dependends folder and then link against that one (set the path in cmake).

prarobo, 2017-01-21:
I am seeing this error when connecting to a usb 3.1 port. Everything works when I use a usb 3.0 port. I am short of usb 3.0 root hubs, so I was wondering if I can use the usb 3.1 root hub I have on my motherboard. Any ideas or ways to get things working with usb 3.1?

xlz, 2017-01-22:
@prarobo Likely kernel driver issue.

jiapei100, 2024-07-09:
When I connect my **Kinect 2** with **Jetson AGX Xavier**, I met the same issue...